blueman-project/blueman #382: AttributeError: 'NoneType' object has no attribute 'get_window'

latot, 2015-11-02:
Hi ppl, well here a little thing, probably a deprecated feature, to reproduce it open blueman-manager and scan, then you will see the message, log:

```
_on_property_changed (/usr/lib64/python2.7/site-packages/blueman/bluez/PropertiesBase.py:30)
/org/bluez/hci0 Discovering 0 
Traceback (most recent call last):
  File "/usr/bin/blueman-manager", line 172, in prop_changed
    prog.finalize()
  File "/usr/lib64/python2.7/site-packages/blueman/gui/manager/ManagerProgressbar.py", line 112, in finalize
    Gdk.Window.set_cursor(self.window.get_window(), None)
AttributeError: 'NoneType' object has no attribute 'get_window'
```

i'm using gentoo 64 with 9999 ebuild.

Thx cya!

infirit, 2015-11-02:
I changed how the windows/dialogs are contructed and did not update the progressbar for this change. I'll push a change a bit later that will fix this.

infirit, 2015-11-02:
The problem should be fixed with https://github.com/blueman-project/blueman/commit/bae8267b55e671c70a6a9f3fedffb0a98a911f06.

latot, 2015-11-02:
yay, thx.

closed 2015-11-02 by latot