blueman-project/blueman #383: dbus.exceptions.DBusException -- multiples windows

latot, 2015-11-02:
Hi again, mm, first the log:

```
Traceback (most recent call last):
  File "/usr/lib64/python2.7/site-packages/blueman/gui/manager/ManagerMenu.py", line 92, in <lambda>
    item.connect('activate', lambda *args: self.blueman.Applet.open_plugin_dialog())
  File "/usr/lib64/python2.7/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib64/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
```

basically this happend when you open first blueman-manager and then open the plugin windows, the close the main windows and there is the error, probably this can be fixed closing the secondary windows with the main.

Distr: Gentoo 64 (9999 ebuild)
BlueZ: 5.35

Thx all.

infirit, 2015-11-02:
Confirmed, but the problem is that blueman-manager is blocked and you can just wait ~60 seconds and it will show up. Not sure why this is, can you try ``net-wireless/blueman-2.0-r1`` and check if you see the same?

latot, 2015-11-02:
testing... testing...
okis, this happend in the 2.0-r1
practically the problem is closing the main windows don't close the secondaries.
tecnically the error message, in fact if we wait a little after click close the main windows it close and we get the message.

cschramm, 2015-11-03:
I guess the issue is that the ManagerMenu waits for a reply from the applet's `open_plugin_dialog` method as the result is used as the result of the lambda although it is actually dropped and that method blocks on `dialog.run()` until the dialog is closed. Either the lambda or the `open_plugin_dialog` method should not block to avoid this.

infirit, 2015-11-03:
Merging the details from #384 as this is the same problem.

>Hi, rethinking this issue, basically blueman-manager don't load the info of the plugins, and it was right, if i open the plugins from the blueman-applet seems works.
i'm checking other things for now..

latot, 2015-11-03:
uu, hi, i like this message, opening the pluging's windows
```
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
```

infirit, 2015-11-03:
@cschramm, would executing the function over dbus help? I see it is exposed by org.blueman.Applet (I am very bad with dbus so don't ask me how :smile:)

cschramm, 2015-11-03:
It's already called via dbus (there's no other way for the manager to talk to the applet), but up to now the call was synchronous. Fixed that. Please test.

closed 2015-11-03 by cschramm

latot, 2015-11-03:
yay now works, the only thing, when i open the plugins from the applet now works load the info, but now i can't reproduce it, well, little weird, anyway.
Thx.

commit 44e74e11 (2015-11-03): Do not block manager with plugin dialog

Fixes #383
commit 2c1c7999 (2015-12-10): Do not block manager with plugin dialog

Backported from 44e74e11b2bf9290463a5f798a619f68d7faf401 in master

Fixes #383