blueman-project/blueman #384: Little things plugins GUI

latot, 2015-11-02:
Hi, again, well here a little things i found in the gui of the plugins windows:

1) For some reason the info of all the plugins are missed, i use the version 2.0 in gentoo and there works fine.

2) When we click configuration button of a plugin it hides the info of the plugin permanently, and the options of the configuration, ex.
Select a marked plugin (bold letters), there we can see the info of the plugin, click in config, now we can see the options, click again config, now the options and the info are hidded we only can see the name of the plugin and the button, then click again the config button, it doesn't show the options. This happend with all config buttons, if you click the config button of a unmarked plugin, you hide all with one click xD

Here the log of the 2):

```
Traceback (most recent call last):
  File "/usr/lib64/python2.7/site-packages/blueman/gui/manager/ManagerMenu.py", line 92, in <lambda>
    item.connect('activate', lambda *args: self.blueman.Applet.open_plugin_dialog())
  File "/usr/lib64/python2.7/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib64/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
```

Distr: Gentoo 64 (9999 ebuild)
BlueZ: 5.35

Thx all and for yours work.

infirit, 2015-11-02:
I can't reproduce this but I think this is related to the previous issue you opened. It looks like you are starting the plugin dialog from blueman-manager. Does the same happen when right clicking on the status icon (blueman-applet)?

latot, 2015-11-02:
i'll be back in some hours, then i can answare.

latot, 2015-11-02:
Hi, rethinking this issue, basically blueman-manager don't load the info of the plugins, and it was right, if i open the plugins from the blueman-applet seems works.

infirit, 2015-11-03:
Same problem as reported in #383, closiong.

closed 2015-11-03 by infirit