blueman-project/blueman #387: Constants.py makes assumptions on directory structure

ngoonee, 2015-11-07:
Lines 31-35 of blueman/Constants.py have

if os.path.exists("../apps") and os.path.exists("../data"):
	BIN_DIR = "./"
	ICON_PATH = "../data/icons"
	PIXMAP_PATH = "../data/icons/pixmaps"
	UI_PATH = "../data/ui"

Obviously, this would fail if blueman-manager (or anyone else importing Functions or Constants) is run in a directory which happens to have those two directory names in it's parent directory (as my ~ does). Is there any way to make this test a bit more specific?

cschramm, 2015-11-07:
I think this test is only relevant when running blueman from the sources as if it's an installed version where that actual structure was used, the constants should already be set to the correct paths by `configure`. So I think it's the exact same case for which the apps use

````python
#support running uninstalled
_dirname = os.path.abspath(os.path.join(os.path.dirname(__file__), ".."))
if os.path.exists(os.path.join(_dirname, "CHANGELOG.md")):
````

It should fit in `Constants.py(.in)` as well. This would also be based on the file's path instead of the working directory, which makes more sense to me (I would not expect that effect when running an installed version of blueman from a source tree, but I would expect it when running a source version from some other directory).

infirit, 2015-11-07:
Checking for ``CHANGELOG.md`` in __addition__ to the two dirs sound like a good plan. In addition as I am sure someone will have that file in their homedir and we end up with the same problem :smile:.

cschramm, 2015-11-07:
Yeah, we can expand the list to everything in the repository that's not getting installed... :wink:

infirit, 2015-11-07:
So, why not use an environment variable instead, ``BLUEMAN_RUN_UNUNSTALLED``?

cschramm, 2015-11-10:
Agree, here's a PR: #389

commit 61ab35b5 (2015-11-10): Introduce BLUEMAN_SOURCE variable

Replaces the different file systems based detection mechanisms for the source tree

Fixes #387
commit f7ce3b8c (2015-11-26): Introduce BLUEMAN_SOURCE variable

Replaces the different file systems based detection mechanisms for the source tree

Fixes #387
closed 2015-11-26 by cschramm