blueman-project/blueman #392: Can't receive files from my android phone

nifker, 2015-11-20:
That is my console output: http://pastebin.com/4zwXmZRv

cschramm, 2015-11-21:
Unfortunately obex-data-server is broken with BlueZ 5 in most distributions. Starting with 1.99.alpha2 blueman will use obexd instead, so just upgrade to a recent version.

nifker, 2015-11-25:
Still not working retried it with version 2.0.1
http://pastebin.com/AzUQEkvP

cschramm, 2015-11-26:
There's a small bug in 2.0.1 causing the exceptions. I just fixed it in https://github.com/blueman-project/blueman/tree/2-0-stable. However, I do not think it's really related to the file transfer.

nifker, 2015-11-27:
I compiled and installed the git version but now it can't find the module "blueman" -  how do I install this module?

cschramm, 2015-11-28:
Unfortunately I don't know your system, but I'd guess you used the default prefix /usr/local and its contents are not considered by your python installation. Supplying `--prefix=/usr` when configuring should help.

nifker, 2015-11-28:
So now I used the git version and can't receive them as well: http://pastebin.com/Ly2itjxP

infirit, 2015-11-28:
I see the output from blueman-manager but the interesting info comes from blueman-applet. Please kill ``blueman-applet``, run it from a terminal and paste the output :smile:

nifker, 2015-11-28:
http://pastebin.com/FS362UuK

infirit, 2015-11-28:
>ERROR:dbus.proxies:Introspect error on org.bluez.obex:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.Spawn.ChildExited: Process org.bluez.obex exited with status 1

This indicates to me that obexd (or whatever is listening) is failing somehow, is there anything interesting in the system logs like errors, segfaults etc?

nifker, 2015-11-28:
Which system logs?

infirit, 2015-11-28:
For example ``/var/log/messages`` or if you are on a systemd ``journalctl``.

nifker, 2015-11-29:
http://pastebin.com/wDAe8t9J

cschramm, 2015-11-29:
What system is that? What version of obexd do you have? 5.36?

nifker, 2015-11-29:
Manjaro 64bit - where can I see my obexd version?

cschramm, 2015-11-29:
I take my question back as obexd seems to be included in Manjaro's bluez package, so it cannot differ.

Well, it's probably some issue with the bluez package. What do you get if you try to run `/usr/lib/bluetooth/obexd`?

nifker, 2015-12-02:
nothing get no output

yars068, 2015-12-03:
In bluez5, obexd have same version as bluez package. But you can try `/usr/lib/bluetooth/obexd -n`, this command would provide information about it's version.

nifker, 2015-12-03:
obex daemon 5.36
I got bluez from the manjaro repos and I have configured blueman without any options

infirit, 2015-12-03:
I just tried manjaro 15.09 in a virtualbox vm and everything works fine there. This is with an out of the box installation with default version of blueman. And because of gpg silliness I can't update so this is as far as I can go.

rworkman, 2015-12-07:
Do you also have obex-data-server installed? If so, remove it and reinstall bluez. Both obex-data-server and obexd (included with bluez5 rather than as a standalone package) claim the same bus name, so whichever was installed last will be invoked, and obex-data-server isn't going to work. 
As a (non-helpful) data point, my android phone (a Moto X 2nd Gen) and blueman-2.0.1 (with bluez-5.36) communicate just fine with sending/receiving files. This could be phone-dependent too - make sure the phone is able to send files to other places; my old phone was just broken in this regard, and I thought that blueman was the culprit for a while...

nifker, 2015-12-07:
obex-data-server wasnt installed and I have a Samsung Galaxy S3

aphirst, 2016-01-11:
I'm having a similar problem as well. Arch Linux x86_64, Motorola Droid 4. I can send to the phone but never receive from it. I'll edit this post / make another post after I do further diagnosis, but what I can see is that blank files in .cache/obexd seem to get made and then the phone reports a transfer error.

cschramm, 2016-04-18:
How's the status, @nifker? Did any bluez, blueman, or whatever updates change anything in the meantime?

What we have: blueman fails to invoke the org.bluez.obex service, but running `/usr/lib/bluetooth/obexd` manually works fine. BlueZ is 5.36. It works for @infirit (same version of BlueZ?).

Please try. `dbus-send --session --dest=org.bluez.obex --print-reply / org.freedesktop.DBus.Introspectable.Introspect`. If that gives you a similar error, I'd just suggest to file a bug report with the Manjaro package. It's probably not a generic packaging bug, but something on your system, especially as it works for @infirit, but I don't know how to further assist with that and it's clearly not a blueman issue then.

closed 2016-05-17 by cschramm