blueman-project/blueman #393: Starting Blueman results in error

retired1, 2015-11-24:
Blueman: v2.0.1
Bluez5: v5.30
Distribution: Vector Linux Vlocity 7.2
Desktop Environment: XFCE

Attempting to start Blueman results in the following error: 

http://pastebin.com/Efdt4gAG

Blueman was compiled from git clone with --prefix=/usr passed to .configure

I get the feeling we missed something on this one, but we're at a loss to determine where to start looking.

yars068, 2015-11-24:
I see that you probably don't have libappindicator installed yet. To avoid this, you can rebuild blueman with passing `--disable-appindicator` to `configure`. Alternatively, just install libappindicator3 package (but that may be not a trivial operation).
Also, check that you have adwaita theme installed correctly.

retired1, 2015-11-24:
We had a bum pyobject build. After redoing it, we're able to get blueman to run.

closed 2015-11-24 by retired1