blueman-project/blueman #396: Two icons in the dash board

MS893, 2015-11-26:
In the dash board, when adding the greffon (indicator), I have two icons for bluetooth, 1 for switching ON or OFF the bluetooth and the devices, 1 for general menu managing bluetooth.

Is it possible to group these two icons in only one ?

Thanks.

cschramm, 2015-11-26:
What dashboard? Are you talking about tray icons? I doubt that they are both blueman's.

MS893, 2015-11-26:
Yes tray icons... (in the bar on top screen or bottom)

I will sens a screen capture when back home (sunday).

One icon with BT configuration menu (with enable/disable option) and one with switches to change between ON and OFF and list of paires devices in which you have also switches to connecter device.

yars068, 2015-11-27:
Which distribution you are running on?

MS893, 2015-11-27:
I tell you when back home.

This problem is for long time and several xubuntu updates.

MS893, 2015-11-27:
xubuntu...

yars068, 2015-11-28:
Possible it is, because blueman uses libappindicator3, which is part of ubuntu-based distributions. Xfce have a system tray, which have a similar functionality as libappindicator. I use Slackware with Xfce and that does *not* use libappindicator anymore, so I have no problem with that.

MS893, 2015-11-29:
Here are the screen captures...

[image]
[image]
[image]
[image]

infirit, 2015-11-29:
The first one is __not__ blueman and from the looks of it it most likely is gnome-bluetooth. Do you have gnome-bluetooth installed and is it running?

MS893, 2015-11-29:
I don't know.
Here is the blueman-applet informations :
[image]
[image]

infirit, 2015-11-29:
I just checked my vm and the first one is definitely gnome-bluetooth. You will need to prevent gnome-bluetooth from loading or remove gnome-bluetooth with the ubuntu package manager of choice.

closed 2015-11-29 by infirit

MS893, 2015-11-29:
I've removed gnome-bluetooth (gnome-bluetooth 3.16.1-1ubuntu2), but I still have 2 icons in the top bar.

Anyway, if I cannot remove one (or in fact merge the 2 icons), I give up I have this for a long time, and I'm not dead :-)

Thanks for trying to help

infirit, 2015-11-29:
>but I still have 2 icons in the top bar

Just removing will not stop the still running process. Run the following command to kill it.
``
kill `pgrep gnome-bluetooth`
``

MS893, 2015-11-29:
Yes I know, I have rebooted my computer after uninstalling gnome-bluetooth and there is no more process.

The 2 icons are not due to gnome_bluetooth, but the xfce4-indicator-plugin which display 2 icons. The problem is not from Blueman...

I don't know which alternative I have for the greffon (indicator plugin)...

Anyway, it's not a big problem, just not logical to have 2 icons for the same function. Thanks

cschramm, 2015-11-29:
The gnome-bluetooth version shipped with wily should not have bluetooth-applet anymore, so basically it makes sense that removing that package does not have the expected effect. The applet is now provided by gnome-shell. I guess it uses the package libgnome-bluetooth13 so you can you check if it's installed and if removing it (is possible) and removes the other icon?

MS893, 2015-11-29:
Yes libgnome-bluetooth13 is installed, but how can I uninstall it ?

MS893, 2015-11-29:
I have done this : sudo apt-get remove libgnome-bluetooth13 to remove it, and now I have 1 icon, but "no indicator" displayed also !!!
[image]

cschramm, 2015-11-29:
I guess that removed the only libindicator icon from xfce4-indicator-plugin so that it now displays a placeholder. You can get rid of it by removing it from the panel or uninstalling it completely.

MS893, 2015-11-29:
Thanks, that's all good now !

I have reinstall the libgnome-bluetooth13, than I removed the indicator greffon from the tab, and I added other elements and now I have only one bluetooth icon in the notification element.

So my problem is solved, thank you very much for your patience !

[image]