blueman-project/blueman #397: Network Access point (NAP) exception

msajadi832, 2015-11-26:
when turn on nap service from local services, it show an exception window with this lines:
```
org.freedesktop.DBus.Python.IOError: Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python2.7/dist-packages/blueman/main/DbusService.py", line 36, in wrapper
    return method(*args[1:], **kwargs)
  File "/usr/lib/python2.7/dist-packages/blueman/plugins/mechanism/Network.py", line 57, in EnableNetwork
    nc.apply_settings()
  File "/usr/lib/python2.7/dist-packages/blueman/main/NetConf.py", line 304, in apply_settings
    self.dhcp_handler.do_apply()
  File "/usr/lib/python2.7/dist-packages/blueman/main/NetConf.py", line 138, in do_apply
    self.do_remove()
  File "/usr/lib/python2.7/dist-packages/blueman/main/NetConf.py", line 167, in do_remove
    dhcp_config, existing_subnet = self._read_dhcp_config()
  File "/usr/lib/python2.7/dist-packages/blueman/main/NetConf.py", line 75, in _read_dhcp_config
    f = open(DHCP_CONFIG_FILE, "r")
IOError: [Errno 2] No such file or directory: '/etc/dhcp3/dhcpd.conf'
```

I want to share my internet with bluetooth for my phone

cschramm, 2015-11-26:
You need a DHCP server for that to work. dhcpd and dnsmasq are supported.

infirit, 2015-11-26:
So, should the user have created a dhcpd.conf file or should blueman take care of that?

If blueman is the one, we end up calling ``do_remove`` which tries to read that file which does not exist. So should we check for ``dhcpcd.conf``?

cschramm, 2015-11-26:
When dhcpd is installed the configuration file should exist.

I'm not sure what happens here as I have very little data, but I guess either blueman was configured to use dhcpd which was removed afterward or the configuration file path is different and was not set on configure.

infirit, 2015-11-26:
>When dhcpd is installed the configuration file should exist.

It doesn't for me but then I am a Gentoo user :wink:. Ok, then the assumption is there should be a conf file, I'll check a couple vm's to see if this holds true for other distro's.

cschramm, 2015-11-27:
Debian has `isc-dhcp-server` which ships `/usr/sbin/dhcpd` and `/etc/dhcp/dhcpd.conf` (you need to set `--with-dhcp-config` accordingly).

msajadi832, 2015-11-27:
Excuse me for bad English!

but `isc-dhcp-server` is installed and then i try install blueman and select dhcp server for nap service.
`/usr/sbin/dhcpd` and `/etc/dhcp/dhcpd.conf` are exists. but `/etc/dhcp3/dhcpd.conf` not exist.

also for first time `dnsmasq` work correct but when after restart ubuntu, selected service for nap automatically changed to dhcp server!

cschramm, 2015-11-27:
Aha, so you're trying to use `isc-dhcp-server` on Ubuntu.

I see two issues here:

A packaging bug as the package should be configured with `--with-dhcp-config /etc/dhcp/dhcpd.conf` to work with `isc-dhcp-server`. I'm actually the one accountable for that. I'll fix the Debian package.

Second, it in fact looks like the DHCP server option is not stored in the configuration. @infirit, could you have a look at that?

infirit, 2015-11-27:
@msajadi832, If no state (dhcp server options) has been saved in ``/var/lib/blueman`` it will default to dnsmasq. So the question is, does it __always__ revert back to dnsmasq or just once when you installed dnsmasq?

@cschramm, I think we should be a bit more clear and tell the user a server binary was found but we failed to find it's config file.

Also, we should probably think about storing the dhcpd config file locations in gsettings (a key for each server supported). Then users can specify their own file and distro's do overrides if the default is incorrect. Opinion?

cschramm, 2015-11-27:
Yeah, that would be better at least as there could theoretically be a case where a distribution ships different dhcpd(3)s with different configuration paths. The blueman maintainer would have no chance to support both then.

infirit, 2015-11-27:
I looked at putting his stuff in gsettings before and remember now why I did not pursue this. blueman-mechanism runs as root, and will fail as gsettings for it may not be available at all or not matching the user's. The same applies to the stuff stored in ``/var/lib/blueman/``. :cry:. So ignore this for now :smile:

msajadi832, 2015-11-27:
Thanks for answering
I'm sorry. I forget an important information. I'm using Ubuntu 15.10. 
@infirit, no after restart ubuntu, it revert back to `dhcpd`. 
a new problem. now check it, for first time after restart, I choose `dnsmasq` and apply. but when open `local services` window, it reverted to `dhcpd` and when I choose `dnsmasq` and apply, it show an exception window with this lines:
```
org.freedesktop.DBus.Python.TypeError: Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python2.7/dist-packages/blueman/main/DbusService.py", line 36, in wrapper
    return method(*args[1:], **kwargs)
  File "/usr/lib/python2.7/dist-packages/blueman/plugins/mechanism/Network.py", line 57, in EnableNetwork
    nc.apply_settings()
  File "/usr/lib/python2.7/dist-packages/blueman/main/NetConf.py", line 304, in apply_settings
    self.dhcp_handler.do_apply()
  File "/usr/lib/python2.7/dist-packages/blueman/main/NetConf.py", line 35, in do_apply
    self.do_remove()
  File "/usr/lib/python2.7/dist-packages/blueman/main/NetConf.py", line 65, in do_remove
    os.kill(self.pid, signal.SIGTERM)
TypeError: an integer is required
```

infirit, 2015-11-27:
@msajadi832 hmm, can you try the below patch and see how things work out for you. This is not meant as a permanent fix just a workaround for now :smile:.

```diff
diff --git a/blueman/main/NetConf.py b/blueman/main/NetConf.py
index b73aee0..e2f740f 100644
--- a/blueman/main/NetConf.py
+++ b/blueman/main/NetConf.py
@@ -61,7 +61,7 @@ class DnsMasqHandler(object):
 
 
     def do_remove(self):
-        if self.netconf.locked("dhcp"):
+        if self.netconf.locked("dhcp") and self.pid:
             os.kill(self.pid, signal.SIGTERM)
             self.netconf.unlock("dhcp")
 
@@ -169,7 +169,7 @@ class DhcpdHandler(object):
         f.write(dhcp_config)
         f.close()
 
-        if self.netconf.locked("dhcp"):
+        if self.netconf.locked("dhcp") and self.pid:
             os.kill(self.pid, signal.SIGTERM)
             self.netconf.unlock("dhcp")
```

msajadi832, 2015-11-27:
with your changes only exception changed to this:
```
org.freedesktop.DBus.Python.Exception: Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python2.7/dist-packages/blueman/main/DbusService.py", line 36, in wrapper
    return method(*args[1:], **kwargs)
  File "/usr/lib/python2.7/dist-packages/blueman/plugins/mechanism/Network.py", line 57, in EnableNetwork
    nc.apply_settings()
  File "/usr/lib/python2.7/dist-packages/blueman/main/NetConf.py", line 304, in apply_settings
    self.dhcp_handler.do_apply()
  File "/usr/lib/python2.7/dist-packages/blueman/main/NetConf.py", line 60, in do_apply
    raise Exception("dnsmasq failed to start. Check the system log for errors")
Exception: dnsmasq failed to start. Check the system log for errors
```

but I find these:
1- after start ubuntu (not log out and login), if I want to use nap ( means my phone try to connect to nap of my ubuntu ), phone connect but don't have IP. then when try to change to `dnsmasq` top exception showed.
2- after start Ubuntu(not log out and log in), I should choose `dnsmasq` to nap work successfully and if I want to open `local services` window again, it revert to `dhcpd` and by change to `dnsmasq`,  top exception showed.

infirit, 2015-11-27:
There are two things:
First, ``/var/run/blueman-dhcp`` is not properly unlinked by NetConf for some reason which causes ``self.netconf.locked("dhcp")`` to be true and we try to kill a non running process (pid == None). calling os.unlink on the file as root work fine so this looks like a proper bug we need to fix.

Second, we now are starting a process but it failed to start, please check the logs (journalctrl I believe now on 15.10) and it should have ~10 entries about dnsmasq/dhcpd failing to start.

msajadi832, 2015-11-27:
journalctl (journalctrl not exist  :smile:):
```
Nov 27 20:38:34 mortezaPC gnome-session[2248]: Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
Nov 27 20:38:41 mortezaPC dbus[720]: [system] Activating service name='org.blueman.Mechanism' (using servicehelper)
Nov 27 20:38:41 mortezaPC org.blueman.Mechanism[720]: Failed to connect to Mir: Failed to connect to server socket: No such file or directory
Nov 27 20:38:41 mortezaPC org.blueman.Mechanism[720]: Unable to init server: Could not connect: Connection refused
Nov 27 20:38:41 mortezaPC org.blueman.Mechanism[720]: Failed to connect to Mir: Failed to connect to server socket: No such file or directory
Nov 27 20:38:41 mortezaPC org.blueman.Mechanism[720]: Unable to init server: Could not connect: Connection refused
Nov 27 20:38:41 mortezaPC blueman-mechanism[10363]: Starting blueman-mechanism
Nov 27 20:38:41 mortezaPC dbus[720]: [system] Successfully activated service 'org.blueman.Mechanism'
Nov 27 20:38:41 mortezaPC org.blueman.Mechanism[720]: (blueman-mechanism:10363): Gtk-CRITICAL **: gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed
Nov 27 20:38:41 mortezaPC blueman-mechanism[10363]: loading Rfcomm
Nov 27 20:38:41 mortezaPC blueman-mechanism[10363]: loading Ppp
Nov 27 20:38:41 mortezaPC blueman-mechanism[10363]: loading Network
Nov 27 20:38:41 mortezaPC blueman-mechanism[10363]: loading RfKill
Nov 27 20:38:41 mortezaPC avahi-daemon[754]: Withdrawing address record for 10.204.222.1 on pan1.
Nov 27 20:38:41 mortezaPC avahi-daemon[754]: Leaving mDNS multicast group on interface pan1.IPv4 with address 10.204.222.1.
Nov 27 20:38:41 mortezaPC avahi-daemon[754]: Interface pan1.IPv4 no longer relevant for mDNS.
Nov 27 20:38:41 mortezaPC avahi-daemon[754]: Joining mDNS multicast group on interface pan1.IPv4 with address 10.218.34.1.
Nov 27 20:38:41 mortezaPC avahi-daemon[754]: New relevant interface pan1.IPv4 for mDNS.
Nov 27 20:38:41 mortezaPC avahi-daemon[754]: Registering new address record for 10.218.34.1 on pan1.IPv4.
Nov 27 20:38:41 mortezaPC avahi-daemon[754]: Withdrawing address record for 10.218.34.1 on pan1.
Nov 27 20:38:41 mortezaPC avahi-daemon[754]: Leaving mDNS multicast group on interface pan1.IPv4 with address 10.218.34.1.
Nov 27 20:38:41 mortezaPC avahi-daemon[754]: Interface pan1.IPv4 no longer relevant for mDNS.
Nov 27 20:38:41 mortezaPC avahi-daemon[754]: Joining mDNS multicast group on interface pan1.IPv4 with address 10.218.34.1.
Nov 27 20:38:41 mortezaPC avahi-daemon[754]: New relevant interface pan1.IPv4 for mDNS.
Nov 27 20:38:41 mortezaPC avahi-daemon[754]: Registering new address record for 10.218.34.1 on pan1.IPv4.
Nov 27 20:38:41 mortezaPC blueman-mechanism[10363]: Return code 0
Nov 27 20:38:41 mortezaPC blueman-mechanism[10363]: Return code 0
Nov 27 20:38:41 mortezaPC blueman-mechanism[10363]: Return code 0
Nov 27 20:38:41 mortezaPC blueman-mechanism[10363]: Return code 0
Nov 27 20:38:41 mortezaPC org.blueman.Mechanism[720]: dnsmasq: failed to create listening socket for fe80::f89b:1aff:feed:4f18%pan1: Address already in use
Nov 27 20:38:41 mortezaPC dnsmasq[10372]: failed to create listening socket for fe80::f89b:1aff:feed:4f18%pan1: Address already in use
Nov 27 20:38:41 mortezaPC dnsmasq[10372]: FAILED to start up
Nov 27 20:38:41 mortezaPC gnome-session[2248]: Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
Nov 27 20:38:45 mortezaPC gnome-session[2248]: blueman-services version 2.0 starting
Nov 27 20:38:45 mortezaPC gnome-session[2248]: wlp3s0
Nov 27 20:38:45 mortezaPC gnome-session[2248]: kvnet
Nov 27 20:38:45 mortezaPC gnome-session[2248]: illegal IP address string passed to inet_aton
```

I want manually stop and start `dnsmasq` service. but after restart this error shows:
```
Job for dnsmasq.service failed because the control process exited with error code. See "systemctl status dnsmasq.service" and "journalctl -xe" for details.
```

and result of "journalctl -xe":
```
Nov 27 20:52:34 mortezaPC systemd[1]: dnsmasq.service: Failed with result 'exit-code'.
Nov 27 20:52:34 mortezaPC sudo[10770]: pam_unix(sudo:session): session closed for user root
Nov 27 20:52:34 mortezaPC polkitd(authority=local)[892]: Unregistered Authentication Agent for unix-process:10771:287547 (syst
Nov 27 20:54:45 mortezaPC sudo[10833]:  morteza : TTY=pts/2 ; PWD=/home/morteza ; USER=root ; COMMAND=/usr/sbin/service dnsmas
Nov 27 20:54:45 mortezaPC sudo[10833]: pam_unix(sudo:session): session opened for user root by morteza(uid=0)
Nov 27 20:54:46 mortezaPC polkitd(authority=local)[892]: Registered Authentication Agent for unix-process:10834:300674 (system
Nov 27 20:54:46 mortezaPC systemd[1]: Starting dnsmasq - A lightweight DHCP and caching DNS server...
-- Subject: Unit dnsmasq.service has begun start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit dnsmasq.service has begun starting up.
Nov 27 20:54:46 mortezaPC dnsmasq[10862]: dnsmasq: syntax check OK.
Nov 27 20:54:46 mortezaPC dnsmasq[10865]: failed to create listening socket for fe80::f89b:1aff:feed:4f18%pan1: Address alread
Nov 27 20:54:46 mortezaPC dnsmasq[10865]: FAILED to start up
Nov 27 20:54:46 mortezaPC systemd[1]: dnsmasq.service: Control process exited, code=exited status=2
Nov 27 20:54:46 mortezaPC dnsmasq[10865]: dnsmasq: failed to create listening socket for fe80::f89b:1aff:feed:4f18%pan1: Addre
[... cut ...]
Nov 27 20:54:46 mortezaPC systemd[1]: Failed to start dnsmasq - A lightweight DHCP and caching DNS server.
Nov 27 20:54:46 mortezaPC systemd[1]: dnsmasq.service: Unit entered failed state.
Nov 27 20:54:46 mortezaPC systemd[1]: dnsmasq.service: Failed with result 'exit-code'.
```

I  purge `isc-dhcp-server` package. But as before I should after start Ubuntu choose `dnsmasq` to work successfully (Only `dnsmas` choice exist and I can't click Apply button. Then I remove last part of `IP Address` and type again it to apply button shows and then I click it)

Thanks for reply

infirit, 2015-11-28:
Aaah, catch all exception handling strikes again :sob:.

@msajadi832, the problem with blueman forgetting which dhcp handler and IP were set is now fixed with e69f2f846f4d6079d81f369042f16d1659a7f968. I also pushed it to the stable branch 33bebbe3c2b89761055412dd8893dfee36096b9b.

This leaves, https://github.com/blueman-project/blueman/issues/397#issuecomment-160154347

@cschramm, do you know why are we adding these lock files for dhcp on top of the pid?

msajadi832, 2015-11-28:
Thanks very much. Can I close this issue?

cschramm, 2015-11-28:
> @cschramm, do you know why are we adding these lock files for dhcp on top of the pid?

You mean why we do not use the `pid` attribute as a lock condition? `blueman-mechanism` is only run for a limited time by dbus, so state needs to be persisted.

infirit, 2015-11-28:
The problem is that if the lock file exist **and** the user modified the [ip address](https://github.com/blueman-project/blueman/blob/master/blueman/main/NetConf.py#L35) we try to kill the dhcphandler with ``do_remove`` that calls ``os.kill(self.pid)``. But self.pid at that stage is None and we run into https://github.com/blueman-project/blueman/issues/397#issuecomment-160154347. And the only place we unlock is in ``do_remove``...

How should this work across reboots, the dhcphandler will probably not be running but the lock file will still be there.

infirit, 2015-11-28:
>Can I close this issue
@msajadi832 there is one open problem so I like to keep this one open until that is sorted.

cschramm, 2015-11-29:
The best solution would probably be to store the PID in the lockfile to a) have it persisted and b) be able to check if it's actually still running or not. We have such a logic in `Functions.check_single_instance`.

infirit, 2015-11-29:
I'll close this one and open a new issue with the problems found with NetConf.

closed 2015-11-29 by infirit