blueman-project/blueman #400: How to install the latest version?

kashanirod, 2015-11-29:
Hi there, 

I know this is a newbie question, but I have been really having trouble installing blueman version 2.0.1. I am having issues with blueman identifying the isc-dhcp-server I have set up. I understand this is due to blueman looking in a different directory to where the dhcp server is installed and that this has been fixed by user infirit last year on issue #109.

I currently have the pi set up as a router by setting up a hostapd, and using the isc-dhcp client. I would like to create a pan network using blueman, using the dhcp settings I have configured for the router. My goal is to create a wifi access point as well as a bluetooth having the dhcp server issue ip's. As far the as the bluetooth connection, I only need one. 

I have tried using git clone https://github.com/blueman-project/blueman.git, using ./configure, but this is where the problems start. It prompts me with "no such file or directory." Then I tried autogen.sh by using sudo sh autogen.sh, and received a "autogen.sh: 23: autogen.sh: aclocal: not found. Then I tried using autoconf configure.ac, and was prompted with a series of configure.ac: errors referring to things like AM_MAINTAINER_MODE, AC_PROG_LIBTOOL etc, about 10 in total. Then I tried a autoreconf -i. After this I am able to run the ./configure command, which gives me a few more errors at the end, on line 3726 AM_GLIB_GNU_GETTEXT not found, line 3727 syntax error near unexpected token '0.35.0' and same line 'IT_PRO_INTLTOOL(0.35.0)' I try make and it says no make file found. 

What am I doing wrong? I have searched the issues area, and no body else seems to have any issues installing the newest version. 

Thanks for your help!

cschramm, 2015-11-29:
Some Ubuntu I guess.

`configure` is not part of the git repository. It's generated by `autogen.sh` / autoconf. It does not make sense to me to run `autogen.sh` using sudo. autoconf cannot be used directly; see `autogen.sh` for the workflow. 

You seem to be missing automake, intltool, libtool, libglib2.0-dev, and probably a lot more that will bite you at a later point. Please pay attention to [the dependencies](https://github.com/blueman-project/blueman/blob/master/Dependencies.md) mentioned in [the installation section of README](https://github.com/blueman-project/blueman/blob/master/README.md#installing).

You can also try `sudo aptitude build-dep blueman` (depending of the Ubuntu version).

kashanirod, 2015-11-29:
Hi there,

Thank you very much for your response. I have since installed the packages,
and  now being prompted by the configure file that I do not have bluez
installed however I do. Do I need to edit a file to the correct path or do
I need to place bluez in the correct location? If so what is that location?

Thanks again!

cschramm, 2015-11-29:
Sound like you're missing libbluetooth-dev, but it should have been installed by `aptitude build-dep blueman`...

I can only guess the error you experience and the system you use, but what you're asked for is probably `/usr/lib/x86_64-linux-gnu/pkgconfig/bluez.pc`.

kashanirod, 2015-11-29:
Hi cschramm, 

Thank you. I am now able to run ./configure, make and make install. Now I am running sudo blueman-manager, since sudo blueman doesn't seem to work. Is this the correct approach? When I run sudo blueman-manager, I get a Bluez daemon not running and cannot continue. I reinstalled bluez 5.31, which seemed to have corrected the issue. Basic hciconfig and hcitool scan work properly. 

When I run sudo blueman-manager, I am able to scan for device, however I can't open preferences to make the PI always discover-able. The terminal says:
 ** (blueman-adapters:8632): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
blueman-adapters version 2.0 starting
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.Spawn.ChildExited: Process org.blueman.Applet exited with status 1
Traceback (most recent call last):
  File "/usr/local/bin/blueman-adapters", line 234, in <module>
    blueman_adapters = BluemanAdapters(adapter_name)
  File "/usr/local/bin/blueman-adapters", line 62, in __init__
    check_bluetooth_status(_("Bluetooth needs to be turned on for the adapter manager to work"), lambda: exit())
  File "/usr/local/lib/python2.7/dist-packages/blueman/Functions.py", line 93, in check_bluetooth_status
    if "PowerManager" in applet.QueryPlugins():
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.Spawn.ChildExited: Process org.blueman.Applet exited with status 1

I noticed, however, in the local services page, blueman did indeed recognize the dhcpd3 client. Having issues trying to enable it. IT provides, what I think to be a random IP address, which differs from the one I have set up in my dhcp. It shows a 10.60.26.1 IP and the apply button is available to click, but when I change it, it grays out, and stays grayed out even if I put the old IP back. My goal is to use the IP address I set up in the DHCP to provide an IP address the bluetooth device connecting to the PI. 

Thanks for your all your help so far!

infirit, 2015-11-29:
I think there is confusion on how blueman nap works at the moment. It basically starts it's own version of dnsmasq/dhcpd, enables ip forwarding and add some iptable rules. So AFAIK it is not possible right now for it to use another dhcp service.

kashanirod, 2015-11-29:
Currently I am using the Pi as a router, it has an IP range of 192.168.42.10 192.168.42.50. I have another PI set up as an IP camera which connects to the Access Point Pi. When I use my laptop or mobile device to connect to the Pi Access Point, I am able to enter the IP address of the Pi with the camera, and view it in a web browser. Will something like this be possible via bluetooth? Could I adhere the Access Point to use the the same settings the blueman will? The install of blueman is on the same PI as the access Point.

Thanks

infirit, 2015-11-29:
So, here is how it works. When the agent (blueman) advertises it can handle NAP Bluez will setup a network interface pan1. We then have to make a dhcp server listen on that specific local interface pan1 so that the client (phone) can request and receive an ip address and other related info like dns etc. We also setup iptables to handle the routing from the pan1 interface.

I don't see how you can make it use the routers dhcp server.

infirit, 2015-11-29:
And there was a bug that prevented the ip settings in blueman-services from being remembered, this is fixed with https://github.com/blueman-project/blueman/commit/33bebbe3c2b89761055412dd8893dfee36096b9b.

kashanirod, 2015-11-29:
Thanks for your response infrit. 

I'm trying to create a network between both interfaces, so that devices connected via wifi, could then communicate to devices connected via bluetooth via IP addresses. Could I set up the router to use the dhcp settings provided by the Bluez Nap they communicate with each other via IP?

My goal is to have two cameras accessible via wifi and bluetooth. My plan of attack would be to create a wifi access point on one PI, and have two more Pis with cameras attached connect to the PI via wifi to that access point, unionizing them on 1 network. I would like these PI's accessible via IP over bluetooth so that devices connected to the access point PI with blueman installed on it could communicate with the camera PI's. Do you think this approach is wrong?

Perhaps I could localize all the cameras on 1 Pi instead and have the cameras accessible? 

I need to be able to stream the camera over bluetooth to mobile. 

Thank you

kashanirod, 2015-11-29:
I just installed a fresh copy of jessie and blueman, so I think i'm all up to date in that regard.

infirit, 2015-11-29:
Never tried such a setup. But I can ping the ip of the phone from the machine the phone is connected to. I can also login from the phone with ssh to other machines on the network. So perhaps you can add a route for a specific subnet on the router that points to the machine with bluetooth devices connected.

I would also recommend you use dnsmasq for this part, it is the best tested and requires no "fiddling" with config files. The iptable rules are setup [here](https://github.com/blueman-project/blueman/blob/2-0-stable/blueman/main/NetConf.py#L294)

kashanirod, 2015-11-30:
Thank you very much for your input and help. Ill try to use dnsmaq, and see how it goes, and then create the router with dnsmasq as well with the same IP scheme. 

As far as setting things up with DNSMASQ, you mentioned it pretty much ready to go without having to change any config files?

Also did you happen to see the errors I was getting for blueman in regards to changing visibility settings? In the terminal I am getting dbus errors. I tried reinstalling but with no luck.

Thanks!

cschramm, 2015-11-30:
It's basically not a good idea to run blueman using sudo. Maybe that's where the dbus issues come from.

infirit, 2015-11-30:
>you mentioned it pretty much ready to go without having to change any config files

Pretty much yes. But to be clear, dnsmasq is only used by blueman. It should not impact whatever service is already running.

kashanirod, 2015-11-30:
Hey guys,

So this is what I have done on a fresh install on jessie:

sudo apt-get install apt-transport-https

add following to /etc/apt/sources.list

deb https://cschramm.eu/blueman/debian/ blueman main
deb-src https://cschramm.eu/blueman/debian/ blueman main

run command:

sudo update
sudo upgrade

sudo aptitude build-dep blueman

install libudev libudev-dev libical libical-dev libreadline:

download new dbus:

wget http://dbus.freedesktop.org/release/dbus/dbus-1.10.4.tar.gz
tar xvf dbus-1.10.4.tar.gz
cd dbus-1.10.4
./configure
make
sudo make install

update bluez with:

sudo apt-get install libglib2.0-dev libdbus-1-dev libudev-dev libical-dev libreadline-dev
wget https://www.kernel.org/pub/linux/bluetooth/bluez-5.36.tar.xz
tar xvf bluez-5.36.tar.gz
cd bluez-5.36
./configure --prefix=/usr --mandir=/usr/share/man --sysconfdir=/etc --localstatedir=/var --disable-systemd --enable-experimental --enable-maintainer-mode
make
sudo make install

downloaded latest build of blueman from:
git clone https://github.com/blueman-project/blueman

cd blueman

sh autogen.sh

run ./configure

then sudo make
then sudo make install

sudo hciconfig hci0 up

blueman-manager

Now, I'm still having issues with the program. It seems functionality is missing. I can pair with a device, but cannot send files or make any connections. No tray icon appears and the program does not show in the "preferences" menu tab. Things like preferences and device setup don't open.  Terminal gives me a Couldn't find foreign struct converter for cairo.context. 

Did I install blueman incorrectly? 
Thanks!

yars068, 2015-11-30:
Please note that using `configure && make && make install` is discouraged, because your package management tool cannot track the files that was installed this way, and you should always manage that packages manually. To avoid that, you need to build the package using appropriate package building tools (for Debian it is `dpkg-buildpackage`), and then install the package as usual (using `apt-get` or `apt` for newest systems) instead.

kashanirod, 2015-11-30:
Ill keep give that a shot. Thank you for your response yars068.

So I installed blueman form the raspberry pi repository and it seemed to have filled out the missing parts of the application, IE now it functions normally as far as the tray icon, being installed in the preference tab, and all the windows of the program open up. 

Still haven't figured out how to make a connection work properly, and haven't set up a PAN yet,

but ultimately Blueman 2.0.1 was installed. 

My issue now is creating a working connection and creating a functional PAN with blueman. If I run into issues here, should i continue in this thread or create a new issue?

Thanks for everyone's help.

closed 2015-12-10 by cschramm

browser-bug, 2020-11-04:
> Please note that using `configure && make && make install` is discouraged, because your package management tool cannot track the files that was installed this way, and you should always manage that [...]
Pretty old but still.. Could you give some insights on how to manage that? Or even link some general guide. I've tried to look more into package building from source but couldn't find much.