blueman-project/blueman #405: error conecting a wiimote

BielBdeLuna, 2015-12-09:
I'm using a Ubuntu 15.10

It shouldn't ask for a pair code as I can't input a pair code from the device, yet it still asks me to input a pair code

the device lights up when I put it in "discovery mode" in it's case pressing the "1" and "2" buttons at the same time, or even another way, pressing the "sync" button behind the device

after writing a false code (consisting of 6 numbers) I get an ubuntu error screen pointing to the "blueman-manager" and the device keeps being unpaired afterwards

why does it ask for a pair code when it shouldn't?

and starting the bluetooth manager from the console i see that the ubuntu error message translates as the following error message in the console:
```
_________
on_property_changed (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:153)
adapter propery changed Discovering 0 
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 607, in msg_reply_handler
    *message.get_args_list()))
  File "/usr/lib/python2.7/dist-packages/blueman/bluez/Device.py", line 33, in err
    raise exception
blueman.bluez.errors.DBusAuthenticationFailedError: Authentication Failed
_________
on_device_property_changed (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:161)
list: device_prop_ch Connected 0 /org/bluez/hci0/dev_40_D2_8A_5D_7D_41 () {} 
```

cschramm, 2015-12-09:
According to http://wiibrew.org/wiki/Wiimote#Bluetooth_Pairing the wiimote uses its own or the hosts address as PIN depending on how you put it into discovery mode. Unfortunately you cannot enter that binary data in a text field, however according to https://wiki.archlinux.org/index.php/XWiimote#BlueZ_does_not_include_the_wiimote_plugin BlueZ ships a plugin for that. I haven't found a reference if this plugin is enabled in Ubuntu 15.10, but obviously it's not.

infirit, 2015-12-09:
Pairs properly for for me with 15.10 in a vm with my RVL-CNT-01 wiimote. But it does not connect  immediatly and I have to redo the connection. But it pairs without issue.

infirit, 2015-12-09:
Ok, poked at this some more. When using one/two buttons both bluetoothctl and blueman will ask for a pin, which will fail for the reason @cschramm mentioned. However, it works fine when using the sync button.

So, it sounds to me this needs to be addressed by BlueZ and it's plugin. The pin code that has to be used differs between the sync and the one/two buttons so my guestimate is that the plugin only handles the sync button.

cschramm, 2015-12-09:
That's it and it's on purpose: http://git.kernel.org/cgit/bluetooth/bluez.git/tree/plugins/wiimote.c?id=2d65ef1d039511aaca1d4f69b3d9cdd89f811f9c#n50

closed 2015-12-09 by cschramm

BielBdeLuna, 2015-12-09:
OK it works! :)