blueman-project/blueman #407: Cannot add keyboard or mouse through graphical interface

Carl393, 2015-12-10:
it's really annoying how you cannot pair devices though the simple graphic interface because it asks for a pairing pin...  how in the world  am I supposed to put that in?!?! 

The only way I could pair both my keyboard and mouse was to refer to this website and do it manually through the shell entering all of these obscure commands

this process should be easier..

Carl393, 2015-12-10:
https://wiki.archlinux.org/index.php/Bluetooth_keyboard

cschramm, 2015-12-10:
I did not get your issue, sorry.

Carl393, 2015-12-10:
hello, i'm on debian jessie 8.2, you don't have the menu on "setup new bluetooth device" after pairing, you can input a number like 0000 ? but then you cannot type this in on the device (such as a mouse) so it does not pair.

cschramm, 2015-12-10:
If you're asked for a PIN the device asks for it. It's usually preset. Read the device's documentation to find out how the process is supposed to work.

cschramm, 2015-12-10:
0000 and 1234 are always good guesses in that situation. ;)

cschramm, 2015-12-10:
So you set a `KeyboardOnly` agent in `bluetoothctl`, initiate pairing, are shown a PIN in the terminal and enter it on the new keyboard?

In blueman you initiate the pairing and are asked for a PIN? Did you try entering something (with a connected keyboard) and the repeating that sequence on the new keyboard?

Carl393, 2015-12-10:
Yes sir! KeyboardOnly command in bluetoothctl like the archlinux website says! it then assigns me a random PIN when I type pair A1:0E (etc...) when I type connect A1:0E (Etc) afterwards it requests the PIN and I can type this PIN in from my bluetooth keyboard... however I cannot do this from the graphical interface (applet)

cschramm, 2015-12-10:
I guess the only difference is that bluetoothctl chooses a random key while blueman asks you to set one. Did you try that?

Carl393, 2015-12-10:
I believe the issue is... the PIN is randomly generated you are correct! however the blueman asks you to set one.. I tried both 0000 and 1234 and neither worked...

Why is it that the bluetoothctl works when it is randomly generated? and on the applet there is nowhere to input this information?

Carl393, 2015-12-10:
in fact you can ask blueman to generate a PIN and it does but afterwards you cannot input it anyhere (there is no input box)

Many other people on forums are running into this problem. Either that or I simply am not doing something right..

cschramm, 2015-12-10:
I think you have to e.g. enter something with a connected keyboard when blueman asks you for a PIN. After submitting, type the same sequence with the new keyboard. I'll check the bluetoothctl code to see in what cases it chooses a random key.

cschramm, 2015-12-10:
Cannot see any such random choosing in the code.

A quick test you can do: Use a KeyboardDisplay agent in bluetoothctl and see how that changes the process. It's the same capability the blueman agent uses.

The more involving way to get behind it: Stop the bluetooth service (make sure blueman, bluetoothctl and similar are not running; they would restart it automatically) and start bluetoothd manually adding the -d option. You will get debugging messages in the syslog (daemon.log) then and we can compare the events when using blueman and when using bluetoothctl.

Carl393, 2015-12-10:
you are correct I enter with a connected keyboard.. 0000.... then it says "please enter the PIN on your device" but there is no input window anywhere to input 0000.... so I find this confusing....

Carl393, 2015-12-10:
I tried just not typing bluetoothctl on my already connected bluetooth keyboard.... and then the connection failed suddenly even though blueman applet says it is connected

this is very disappointing.

cschramm, 2015-12-10:
> then it says "please enter the PIN on your device" but there is no input window anywhere to input 0000.... so I find this confusing....

Aha! Yes, blueman does not (yet) indicate the progress of your typing. But does it work or not?

Carl393, 2015-12-10:
no it does not.. it automatically says FAILED...

Can you see how this is confusing for people who want to use graphical interface? how is the user even supposed to know if the applet is accepting the 0000 or other PIN that is being typed

In the Shell when i type pair A0:23 etc....the agent gives me the PIN code...then I type 'trust' A0:23 
which works... then i type 'connect A0:23 etc' but it never asks for the PIN code why is that?

Carl393, 2015-12-10:
I find this confusing as well... am I doing something wrong?
[image]

Carl393, 2015-12-10:
I have tried the instructions here
https://wiki.archlinux.org/index.php/Bluetooth_keyboard

exactly and the device does not register... it works.. then I either get authentication error... or when I type quit... the Agent does not register.

Carl393, 2015-12-10:
I can get the keyboard to work but no matter what I do... I can not get the device to pair...
as you can see I can get the keyboard to be trusted but I cannot pair.. so I have to keep adding the keyboard again every time I boot up. is this a bug
[image]

infirit, 2015-12-10:
Are you using separate bluetooth dongle or the one that came with the keyboard and mouse? If so and when following the wiki instructions for bluetoothctl, it looks like you skipped the step that sets the dongle pairable, ``[bluetooth]# pairable on``.

When I un-break my machine I'll have a look how blueman behaves.

Carl393, 2015-12-10:
The dongle is already paired to my bluetooth mouse is this the problem ? (cannot be paired to two devices at the same time) my dongle did not come with either device it is one I purchased ages ago for use with my hp printer. I tried what you suggested pairable: on and doesn't work.

Carl393, 2015-12-13:
Hello Please advise, I have set up my bluetooth keyboard after typing connect there is no prompt that asks you to enter the PIN and then hit enter... Can you fix this for the command line view?
I just tried typing in the PIN assigned without me asking for it... when I hit enter it worked!

It would be much easier if the terminal requests that the PIN be entered after typing "Connect 20:15:09:29:01:1F' and that you hit enter... as this isn't obvious.

On a further note. When I type the PIN in blueman applet.. and hit enter.. it does not work at all..
Could you duplicate the same prompts for the graphical interface? as the pairing process only works from the terminal.

Thank you for your help!

Whoever is in charge of the programing of blueman can you introduce these

jamesjohn911, 2016-03-31:
when you try to pair and it gives you the pin you have to type the pin into your bluetooth keyboard your trying to pair then hit enter on same keyboard then it should work... im having similar problem with raspi3 bluetooth same issue wont pair my bluetooth keyboard or my bluetooth headset but will pair my galaxy s5 mobile (and use it for networking) and hp pavilion dv7 laptop. i also tryed to pair bt keyboard with terminal commands but still comes up not paired in the end sooo frustrating Aaargh!!!

Carl393, 2016-04-21:
Hi JamesWould you like to correspond about raspi3 I just bought one!                                                                                                                                                                                                                                                                                                                                        Carl                                                                                                                                                                                                                From: James MoretonSent: Thursday, March 31, 2016 1:23 PMTo: blueman-project/bluemanReply To: blueman-project/bluemanCc: Carl393Subject: Re: [blueman-project/blueman] Cannot add keyboard or mouse through graphical interface (#407)when you try to pair and it gives you the pin you have to type the pin into your bluetooth keyboard your trying to pair then hit enter on same keyboard then it should work... im having similar problem with raspi3 bluetooth same issue wont pair my bluetooth keyboard or my bluetooth headset but will pair my galaxy s5 mobile (and use it for networking) and hp pavilion dv7 laptop. i also tryed to pair bt keyboard with terminal commands but still comes up not paired in the end sooo frustrating Aaargh!!!

jamesjohn911, 2016-04-22:
Hi Carl mate yeah sure i'll correspond with you on RasPi3 its way better
than raspi2 bit of a pain to set up the bluetooth drivers and pair certain
devices that dont wanna pair for some stupid reason ...

do you own a cirrus audio card for your raspberry pi..??

regards
James Moreton

Carl393, 2016-04-23:
Don't really want to bump a dead thread
Maybe just send me your email./

Carl

infirit, 2016-08-21:
Old one, but I read that some keyboards just can not pair and we should instead mark them as trusted. Does marking the device as trusted make it keep it's connection?

Carl393, 2016-08-21:
Yes                                                                                                                                                                                                                From: Sander SweersSent: Sunday, August 21, 2016 10:55 AMTo: blueman-project/bluemanReply To: blueman-project/bluemanCc: Carl393; AuthorSubject: Re: [blueman-project/blueman] Cannot add keyboard or mouse through graphical interface (#407)Old one, but I read that some keyboards just can not pair and we should instead mark them as trusted. Does marking the device as trusted make it keep it's connection?

closed 2016-10-05 by cschramm

dyyzhmm, 2017-01-11:
So, what's the situation now? I has been working on this bug for a week.