blueman-project/blueman #413: Adapter will randomly loose connection?

CanadianJeff, 2015-12-15:
```
root@xubuntu:/tmp# lsusb | grep "Bluetooth"
Bus 007 Device 043: ID 0a5c:2148 Broadcom Corp. BCM92046DG-CL1ROM Bluetooth 2.1 Adapter
Bus 007 Device 040: ID 0a5c:4500 Broadcom Corp. BCM2046B1 USB 2.0 Hub (part of BCM2046 Bluetooth)

root@xubuntu:/tmp# systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2015-12-15 14:17:33 PST; 7s ago
     Docs: man:bluetoothd(8)
 Main PID: 3468 (bluetoothd)
   Status: "Running"
    Tasks: 1 (limit: 512)
   CGroup: /system.slice/bluetooth.service
           └─3468 /usr/lib/bluetooth/bluetoothd --compat

Dec 15 14:17:33 xubuntu bluetoothd[3468]: Not enough free handles to register service
Dec 15 14:17:33 xubuntu bluetoothd[3468]: Current Time Service could not be registered
Dec 15 14:17:33 xubuntu bluetoothd[3468]: gatt-time-server: Input/output error (5)
Dec 15 14:17:33 xubuntu bluetoothd[3468]: Not enough free handles to register service
Dec 15 14:17:33 xubuntu bluetoothd[3468]: Not enough free handles to register service
Dec 15 14:17:33 xubuntu bluetoothd[3468]: Sap driver initialization failed.
Dec 15 14:17:33 xubuntu bluetoothd[3468]: sap-server: Operation not permitted (1)
Dec 15 14:17:33 xubuntu bluetoothd[3468]: input-hid profile probe failed for CC:9E:00:60:1E:06
Dec 15 14:17:33 xubuntu bluetoothd[3468]: Endpoint registered: sender=:1.52 path=/MediaEndpoi...rce
Dec 15 14:17:33 xubuntu bluetoothd[3468]: Endpoint registered: sender=:1.52 path=/MediaEndpoi...ink
Hint: Some lines were ellipsized, use -l to show in full.
```

cschramm, 2015-12-16:
I cannot spot anything blueman related there. You're experiencing an issue with your system's bluetooth stack.

The issues lead me to this thread: https://bbs.archlinux.org/viewtopic.php?id=195886 Although they suspect a problem with the Arch Linux package, people come to very different solutions and conclusions for their issues: Reset the HCI, remove `~/.config/pulse` (does not make much sense to me), broken dongle. In fact somebody there posts information about the same BCM2046, so that device seems to not work reliably. It may be solved with a Kernel or BlueZ update.

closed 2015-12-16 by cschramm

yars068, 2015-12-16:
Possible that chip needs a firmware.