blueman-project/blueman #414: Can't pair with non-keyboard keyboard devices

ivazqueznet, 2015-12-17:
Devices such as the Tzumi SmartLens/MusicShutter advertise themselves as keyboards, so when an attempt to pair is made you need to enter a PIN in order to do so. However, it is impossible to enter the PIN on the device. The following image illustrates why:

[image]

Devices such as these require either one of a small set of fixed PINs or the manual entry of a specific PIN, which is why blueman does not work with them.

Or if some accommodation could be made for their non-alphabetic entry methods, that would also work.

cschramm, 2015-12-17:
Unfortunately you forgot to add data (especially blueman-applet output of the pairing).

I cannot spot anything matching in [the gnome-bluetooth PIN code database](https://github.com/GNOME/gnome-bluetooth/blob/master/lib/pin-code-database.xml) from what I know.

ivazqueznet, 2015-12-17:
The terminal output from blueman-applet is a big mash of debug output (probably since I'm running HEAD), but I believe that the important part is the address, 20:14:11:24:00:12.

Your comment makes me believe that there's a central place all fixed PINs should be entered, but editing the file you named on my system (/usr/share/gnome-bluetooth/pin-code-database.xml) made no difference.

I did manage to get it working by running bluetoothctl and connecting in there (at which point it asked me to type in the PIN), but it still shows as unpaired in blueman-manager.

cschramm, 2015-12-17:
> The terminal output from blueman-applet is a big mash of debug output (probably since I'm running HEAD), but I believe that the important part is the address, 20:14:11:24:00:12.

The important part is basically everything from where you start the discovery. Apart from the address it would especially be very interesting what agent methods are called (RequestPinCode, RequestPasskey, DisplayPinCode, DisplayPasskey, ...).

> Your comment makes me believe that there's a central place all fixed PINs should be entered, but editing the file you named on my system (/usr/share/gnome-bluetooth/pin-code-database.xml) made no difference.

blueman does not have / use a PIN database (yet). gnome-bluetooth's one was just for reference. If the device would have a fixed PIN, it could probably be found in there, but it does not look like it from the address you posted.

> I did manage to get it working by running bluetoothctl and connecting in there (at which point it asked me to type in the PIN), but it still shows as unpaired in blueman-manager.

Connecting is not pairing. Connecting with a PIN is nothing known to me yet.

What ist "the PIN"? In your initial post you stated that blueman asks you for a PIN as well, but you cannot enter it. I supposed that's because you do not know it.

If the Paired flag switches to Yes in bluetoothctl and blueman does not get that change, we have a bug. We need the blueman-applet output to debug it ... and the output from bluetoothctl in this case.

infirit, 2016-01-13:
Gentle reminder for @ivazqueznet :smile:

closed 2016-01-17 by cschramm