blueman-project/blueman #415: PIN code databse

cschramm, 2015-12-17:
As built-in PIN codes come up over and over again, I'll implement a PIN code database. I will steal [gnome-bluetooth's](https://github.com/GNOME/gnome-bluetooth/blob/master/lib/pin-code-database.xml) as a starting point.

rworkman, 2015-12-17:
Assuming I understand this, it seems like something that should probably be added to the udev hwdb and be easily query'able.

cschramm, 2015-12-17:
Sounds reasonable. bluedevil maintains basically the same file (they do not use the vendor attribute): https://quickgit.kde.org/?p=bluedevil.git&a=blob&f=src%2Fwizard%2Fpin-code-database.xml

rworkman, 2015-12-17:
Hrm... maybe the hwdata project maintained by RH is a better option: https://git.fedorahosted.org/git/hwdata.git
Are you going to use the same format, or maybe a better question is, is the bluedevil file's format the same as gnome-bluetooth's format? If they're in two different formats, I doubt they would embrace any effort to push this in to hwdata, and thus it's just as well to go ahead and do it here. I just hate the idea of trying to keep it in sync with changes that happen to gnome-bluetooth and bluedevil. Everyone would win by keeping a single copy updated somewhere neutral, maybe even as part of bluez itself.

rworkman, 2015-12-17:
Nevermind - I see the answers to the format issue by looking at the files :-)
Sooo... how about petitioning bluez upstream to include this information at e.g. /usr/share/bluetooth/pin-code-database.xml ?

infirit, 2015-12-17:
There is a big downside of having this in a project you do not control. Changes may take forever to trickle down to the distro's and having changes accepted may not be so easy. And that is after all agreed on the format with lot's of discussion between the projects, bikeshedding, format stability etc.. It probably is not worth the effort.

However what we can do is contribute back to gnome if we add new devices.

rworkman, 2015-12-17:
It looks like the format and even the file itself is pretty much identical, which makes me wonder how often things are added/changed/etcetera, and so maybe some of that's not an issue. Valid points nonetheless.

cschramm, 2015-12-18:
Yeah, it looks pretty stable. Anyway, feel free to reach out to the others for a common database. gnome-bluetooth has [its own mailing list](http://lists.usefulinc.com/mailman/listinfo/gnome-bluetooth). I think for bluedevil you'd have to use [the generic kde-devil mailing list](https://mail.kde.org/mailman/listinfo/kde-devel) (or IRC). hwdata seems to be maintained basically by a single person: Michal Minar miminar@redhat.com.

Back on topic: I intended to just grab the data and not use the XML format, but of course it would make synchronization with the other projects easier.

infirit, 2015-12-18:
>not use the XML format

We can but the data has to be stored in __a__ format anyway. Python comes with a pretty good parser so constructing the queries would be pretty easy. What other format were you thinking of, python dict?

cschramm, 2015-12-18:
Yes, I thought about a python dict, but I guess XML is in fact the easiest. I did a quick draft this morning: https://github.com/blueman-project/blueman/commit/12f2e28bba01652d15208befe166f8d02d30235f

infirit, 2016-01-09:
I have been playing with elementtree for something else and found out we can not use xpath as it does not support ``not``, ``and`` etc. This made me think of the pin db and figured we'll have to implement this in a for loop then.

@cschramm I know you self assigned this so I don't want to get in your way more than I already have.

infirit, 2016-01-19:
Was looking at the main.Device Class property and ran into [``DeviceClass.py``](https://github.com/blueman-project/blueman/blob/master/blueman/DeviceClass.py#L172) which has code to translates the class to names. So while a good learning experience for me it was in the end never needed :smile:

infirit, 2016-08-05:
This just needs some testing on devices in the list, I personally do not have any devices in the list (mice/keyboard do not generally trigger requestpin under BlueZ5) but I may have a BT keyboard which may qualify.

Commit: https://github.com/infirit/blueman/commit/f59e1ac68a5d66f1488f1784b4a7c64ded79782e

infirit, 2016-08-21:
This landed, closing.

closed 2016-08-21 by infirit