blueman-project/blueman #417: blueman-applet startup failure

spinosae, 2015-12-18:
I have been having problem with blueman since a while ago (possibly since 27th Nov). My system:
```bash
$uname -a
Linux fedora 4.2.7-300.fc23.x86_64 #1 SMP Wed Dec 9 22:28:30 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
$ dnf list installed | grep blue
NetworkManager-bluetooth.x86_64       1:1.0.6-8.fc23            @updates        
blueman.x86_64                        2.1-0.3.git7a2e20e.fc23   @fedora         
bluez.x86_64                          5.36-1.fc23               @System         
bluez-cups.x86_64                     5.36-1.fc23               @System         
bluez-hid2hci.x86_64                  5.36-1.fc23               @System         
bluez-libs.x86_64                     5.36-1.fc23               @System         
bluez-libs-devel.x86_64               5.36-1.fc23               @System         
gnome-bluetooth.x86_64                1:3.18.1-1.fc23           @updates        
gnome-bluetooth-libs.x86_64           1:3.18.1-1.fc23           @updates        
gnome-bluetooth-libs-devel.x86_64     1:3.18.1-1.fc23           @updates        
pulseaudio-module-bluetooth.x86_64    7.1-1.fc23                @updates  
```
Symptom: After system boot, bluetooth-applet appeared to be loaded (listed in processes) but no icon showed up in the system tray. If I start bluetooth-manager right away, it would took much longer to load the GUI and the manager wouldn't function properly. However if I kill bluetooth-applet (pkill) first, then applet and manager can start and work properly (icon will show up in tray). Here is the boot log:
```bash
$ journalctl --boot=-7 | grep blueman
Dec 18 19:31:47 fedora dbus[983]: [system] Activating service name='org.blueman.Mechanism' (using servicehelper)
Dec 18 19:31:47 fedora org.blueman.Mechanism[983]: Unable to init server: Could not connect: Connection refused
Dec 18 19:31:47 fedora org.blueman.Mechanism[983]: Unable to init server: Could not connect: Connection refused
Dec 18 19:31:47 fedora blueman-mechanism[3775]: Starting blueman-mechanism
Dec 18 19:31:47 fedora dbus[983]: [system] Successfully activated service 'org.blueman.Mechanism'
Dec 18 19:31:47 fedora org.blueman.Mechanism[983]: (blueman-mechanism:3775): Gtk-CRITICAL **: gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed
Dec 18 19:31:47 fedora blueman-mechanism[3775]: loading RfKill
Dec 18 19:31:47 fedora blueman-mechanism[3775]: loading Ppp
Dec 18 19:31:47 fedora blueman-mechanism[3775]: loading Network
Dec 18 19:31:47 fedora blueman-mechanism[3775]: loading Rfcomm
Dec 18 19:31:47 fedora blueman.desktop[3340]: (blueman-applet:3340): Gdk-CRITICAL **: gdk_window_thaw_toplevel_updates: assertion 'window->update_and_descendants_freeze_count > 0' failed
Dec 18 19:31:50 fedora blueman.desktop[4876]: blueman-applet version 2.0 starting
Dec 18 19:31:50 fedora blueman.desktop[4876]: There is an instance already running
Dec 18 19:32:17 fedora blueman-mechanism[3775]: Exiting
```
The suspicious line is ``Dec 18 19:31:50 fedora blueman.desktop[4876]: There is an instance already running``. It seemed that an instance of blueman-applet was runing before /etc/xdg/autostart/blueman.desktop was loaded. This line wasn't in the log before Nov 27. I can't figure out what changed at that point of time. Now my workaround was to kill blueman-applet first and start again in the .desktop file. However I wonder whether something broke in blueman. Could you look into it? Tell me if you need more log.

spinosae, 2015-12-18:
Info from blueman-report:
```
blueman: 2.0
BlueZ: 5.36
Distribution: 
Desktop: GNOME
```

cschramm, 2015-12-18:
The package you're using is some snapshot from the development tree. I you could try some older package versions like 2.1-0.2, 2.1-0.1, 2.0, we could possibly isolate an issue if it's related to the code.

That aside, to debug your problem you can try starting blueman-manager from a terminal and collect the output. Maybe it gives us a clue where and why it's slowed down. It basically sounds like the applet process is unresponsive, but that should lead to an error message in manager and not work at all... You can also check your ~/.xsession-errors. blueman-applet may leave something there.

spinosae, 2015-12-19:
Here is the log for running blueman-manager right after boot
```
$ blueman-manager 
_________
Load (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:60)
['PulseAudioProfile', 'Services'] 
_________
__load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.manager.Services.Services'> 
_________
__load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.manager.PulseAudioProfile.PulseAudioProfile'> 
_________
pa_context_event (/usr/lib/python3.4/site-packages/blueman/main/PulseAudioUtils.py:342)
1 
blueman-manager version 2.0 starting
_________
pa_context_event (/usr/lib/python3.4/site-packages/blueman/main/PulseAudioUtils.py:342)
2 
_________
on_bluez_name_owner_changed (/usr/bin/blueman-manager:94)
org.bluez owner changed to  :1.2 
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
_________
SetAdapter (/usr/lib/python3.4/site-packages/blueman/gui/DeviceList.py:233)
 
_________
SetAdapter (/usr/lib/python3.4/site-packages/blueman/gui/DeviceList.py:233)
No such adapter 
_________
SetAdapter (/usr/lib/python3.4/site-packages/blueman/gui/DeviceList.py:233)
None 
_________
on_adapter_changed (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerToolbar.py:75)
toolbar adapter /org/bluez/hci0 
_________
__init__ (/usr/lib/python3.4/site-packages/blueman/main/Device.py:23)
caching initial properties 
_________
add_device (/usr/lib/python3.4/site-packages/blueman/gui/DeviceList.py:276)
adding new device 
_________
do_cache (/usr/lib/python3.4/site-packages/blueman/gui/DeviceList.py:418)
Caching new device B4:52:7D:ED:F4:B8 
_________
row_update_event (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceList.py:252)
row update event Trusted 0 
_________
row_update_event (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceList.py:252)
row update event Paired 1 
_________
Generate (/usr/lib/python3.4/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:172)
SBH50 
_________
pa_context_event (/usr/lib/python3.4/site-packages/blueman/main/PulseAudioUtils.py:342)
5 
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
```
Corresponding log in journalctl:
```
Dec 19 09:13:07 fedora /usr/libexec/gdm-x-session[4111]: Failed to activate service 'org.blueman.Applet': timed out
Dec 19 09:19:37 fedora /usr/libexec/gdm-x-session[4111]: Activating service name='org.blueman.Applet'
Dec 19 09:19:37 fedora org.blueman.Applet[4170]: blueman-applet version 2.0 starting
Dec 19 09:19:37 fedora org.blueman.Applet[4170]: There is an instance already running
```
However ``sudo blueman-manager`` works

spinosae, 2015-12-19:
By the way, 2.1-0.3.git7a2e20e.fc23 was listed as the newest stable build in [blueman Fedora repo](https://apps.fedoraproject.org/packages/blueman/overview/). Nonetheless, I tried installing version 2.0.11, no luck. Maybe this is something related to Fedora kernel update?

cschramm, 2015-12-19:
So as expected the applet service does not respond. I suppose you kill the process with SIGTERM, so it's not the whole process that's unresponsive. Did you start blueman-manager before or after 09:13:07?

> However `sudo blueman-manager` works

That's not a surprise as the single-instance checking is per-user depending on the user's cache dir (`/root/.cache`) then. Another blueman-applet process will be started. Check your processes to make sure that's correct.

As the calls would not take that long if the dbus messages would be blocked, the only thing I can think of is that the applet fails to register its dbus service when invoked by your session manager. I'm pretty sure this should lead to some message(s) in ~/.xsessions-errors. Did you check it already?

cschramm, 2015-12-19:
@leigh123linux This is not a generic issue in current FC23, is it?

spinosae, 2015-12-19:
>Did you start blueman-manager before or after 09:13:07?
Checked again, that line (removed) was from a previous session. Blueman-manager was started at 09:19:37.
> I'm pretty sure this should lead to some message(s) in ~/.xsessions-errors. Did you check it already?
Just checked, nothing informative
```
$ journalctl -b -u session-1.scope | grep blueman
Dec 19 22:09:05 fedora blueman.desktop[4954]: blueman-applet version 2.0 starting
Dec 19 22:09:05 fedora blueman.desktop[4954]: There is an instance already running
$ journalctl -b -u session-c1.scope | grep blueman
Dec 19 22:08:49 fedora blueman.desktop[3364]: (blueman-applet:3364): Gdk-CRITICAL **: gdk_window_thaw_toplevel_updates: assertion 'window->update_and_descendants_freeze_count > 0' failed
```
Or should I look for another keyword in the log? Sorry I'm fresh to dbus, session manager etc. and thank for your kind support on this issue.

infirit, 2015-12-20:
Just dud a fresh F23 xfce spin install in a vm and I see no issues with blueman.

spinosae, 2015-12-21:
@infirit I was using gnome. Can you post your ``journalctl -b | grep blueman``?

infirit, 2015-12-21:
Why are you using blueman under gnome? I would suggest either to use XFCE/MATE/LXDE or switch to gnome-bluetooth. Below is how it looks like under xfce (under gnome it looks the same).

Also, blueman-manager ui under gnome with a fresh fedora 23 workstation netinstall is pretty broken. Would be nice if the maintainer could have a look and confirm if (s)he is willing to support this.

``
$ journalctl -b | grep blueman
dec 21 11:50:56 localhost.localdomain dbus[722]: [system] Activating service name='org.blueman.Mechanism' (using servicehelper)
dec 21 11:50:56 localhost.localdomain org.blueman.Mechanism[722]: Unable to init server: Could not connect: Connection refused
dec 21 11:50:56 localhost.localdomain org.blueman.Mechanism[722]: Unable to init server: Could not connect: Connection refused
dec 21 11:50:56 localhost.localdomain blueman-mechanism[1832]: Starting blueman-mechanism
dec 21 11:50:56 localhost.localdomain dbus[722]: [system] Successfully activated service 'org.blueman.Mechanism'
dec 21 11:50:56 localhost.localdomain org.blueman.Mechanism[722]: (blueman-mechanism:1832): Gtk-CRITICAL **: gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed
dec 21 11:50:56 localhost.localdomain blueman-mechanism[1832]: Skipping plugin RfKill: Hardware kill switch not found
dec 21 11:50:56 localhost.localdomain blueman-mechanism[1832]: loading Ppp
dec 21 11:50:56 localhost.localdomain blueman-mechanism[1832]: loading Rfcomm
dec 21 11:50:56 localhost.localdomain blueman-mechanism[1832]: loading Network
``

leigh123linux, 2015-12-21:
@infirit TBH I don't give a crap is blueman is broken in the gnome workstation media, cinnamon, mate and xfce were my intended targets.

spinosae, 2015-12-21:
@infirit You don't have blueman-applet or blueman.desktop in your log. Other than that our logs look quite similar. Gnome-bluetooth is painful to use as it doesn't create audio profiles for bluetooth headset; as a result connected devices don't show up in sound setting. I didn't know about lacking of support for gnome, it did work initially though. Maybe I'll try xfce sometime, or stick with my fragile workaround for now. Thanks for your support and please close this issue if you feel it is not concern of the project.

infirit, 2015-12-21:
@spinosae 
>create audio profiles for bluetooth headset

Changing them is done under the sound preference but you sometimes have to reconnect to make it show up.

@leigh123linux, ok good to know.

closed 2015-12-21 by infirit

hackswitch, 2020-07-04:
OK Same problem here, ish. 
running 20.04 KDE Plasma Lubuntu.
Blueman and all tools installed; 
XO Speaker pairs bluetooth (Serial) but cannot play audio. 
Had her working once. 
output. 

$: sudo apt-get update
Hit:1 http://linux.teamviewer.com/deb stable InRelease
Hit:2 https://brave-browser-apt-release.s3.brave.com stable InRelease                  
Hit:3 http://us.archive.ubuntu.com/ubuntu focal InRelease                              
Hit:4 https://apt.repos.intel.com/mkl all InRelease                                    
Hit:5 http://us.archive.ubuntu.com/ubuntu focal-updates InRelease   
Hit:6 http://ppa.launchpad.net/bdbcat/opencpn/ubuntu focal InRelease
Hit:7 http://us.archive.ubuntu.com/ubuntu focal-backports InRelease      
Hit:8 http://us.archive.ubuntu.com/ubuntu focal-security InRelease       
Ign:9 http://ppa.launchpad.net/bluetooth/bluez/ubuntu focal InRelease
Hit:10 http://ppa.launchpad.net/iconnor/zoneminder-1.34/ubuntu focal InRelease
Hit:11 http://ppa.launchpad.net/kubuntu-ppa/backports/ubuntu focal InRelease
Ign:12 http://ppa.launchpad.net/webupd8team/indicator-kdeconnect/ubuntu focal InRelease
Hit:13 http://ppa.launchpad.net/xrdesktop/xrdesktop/ubuntu focal InRelease
Err:14 http://ppa.launchpad.net/bluetooth/bluez/ubuntu focal Release
  404  Not Found [IP: 91.189.95.83 80]
Err:15 http://ppa.launchpad.net/webupd8team/indicator-kdeconnect/ubuntu focal Release
  404  Not Found [IP: 91.189.95.83 80]
Reading package lists... Done
E: The repository 'http://ppa.launchpad.net/bluetooth/bluez/ubuntu focal Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: The repository 'http://ppa.launchpad.net/webupd8team/indicator-kdeconnect/ubuntu focal Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
o***@o***-NUC7i5BNH:~$ sudo apt-get upgrade 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  libllvm9 libllvm9:i386
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
o@o-NUC7i5BNH:~$ sudo apt install pulseaudio pulseaudio-utils pavucontrol pulseaudio-module-bluetooth
Reading package lists... Done
Building dependency tree       
Reading state information... Done
pavucontrol is already the newest version (4.0-1build1).
pulseaudio is already the newest version (1:13.99.1-1ubuntu3.3).
pulseaudio-module-bluetooth is already the newest version (1:13.99.1-1ubuntu3.3).
pulseaudio-utils is already the newest version (1:13.99.1-1ubuntu3.3).
The following packages were automatically installed and are no longer required:
  libllvm9 libllvm9:i386
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
o***@o**-NUC7i5BNH:~$ sudo service bluetooth restart
o***@o***-NUC7i5BNH:~$ journalctl -f
-- Logs begin at Sat 2020-04-11 04:51:47 CDT. --
Jul 04 13:30:33 o***-NUC7i5BNH blueman-applet[200634]: blueman-manager version 2.1.2 starting
Jul 04 13:30:33 o***-NUC7i5BNH dbus-daemon[1367]: [system] Successfully activated service 'org.freedesktop.hostname1'
Jul 04 13:30:33 o***-NUC7i5BNH systemd[1]: Started Hostname Service.
Jul 04 13:30:33 o***-NUC7i5BNH dbus-daemon[1367]: [system] Activating via systemd: service name='org.blueman.Mechanism' unit='blueman-mechanism.service' requested by ':1.76' (uid=1000 pid=3155 comm="/usr/bin/python3 /usr/bin/blueman-applet " label="unconfined")
Jul 04 13:30:33 o***-NUC7i5BNH systemd[1]: Starting Bluetooth management mechanism...
Jul 04 13:30:33 o***-NUC7i5BNH blueman-mechanism[203823]: Unable to init server: Could not connect: Connection refused
Jul 04 13:30:34 o***-NUC7i5BNH blueman-mechanism[203823]: Unable to init server: Could not connect: Connection refused
Jul 04 13:30:34 o****-NUC7i5BNH blueman-mechani[203823]: gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed
Jul 04 13:30:34 o***-NUC7i5BNH dbus-daemon[1367]: [system] Successfully activated service 'org.blueman.Mechanism'
Jul 04 13:30:34 o***-NUC7i5BNH systemd[1]: Started Bluetooth management mechanism.
Jul 04 13:31:03 o***-NUC7i5BNH systemd[1]: systemd-hostnamed.service: Succeeded.
Jul 04 13:31:04 o***-NUC7i5BNH systemd[1]: blueman-mechanism.service: Succeeded.
^C
O*****@***-NUC7i5BNH:~$ journalctl -b | grep blueman
[... 47 lines cut ...]
Jul 04 13:29:57 o-NUC7i5BNH systemd[1]: blueman-mechanism.service: Succeeded.
Jul 04 13:30:32 o-NUC7i5BNH blueman-applet[3155]: blueman-applet 13.30.32 WARNING  DiscvManager:109 update_menuitems: warning: Adapter is None
Jul 04 13:30:32 o**-NUC7i5BNH blueman-applet[200634]: UnicodeDecodeError: 'utf-8' codec can't decode byte 0x90 in position 0: invalid start byte
Jul 04 13:30:32 o**-NUC7i5BNH blueman-applet[200634]: UnicodeDecodeError: 'utf-8' codec can't decode byte 0xbe in position 4: invalid start byte
Jul 04 13:30:32 o**-NUC7i5BNH blueman-applet[200634]: UnicodeDecodeError: 'utf-8' codec can't decode byte 0xa6 in position 2: invalid start byte
[... cut ...]
Jul 04 13:30:33 o**-NUC7i5BNH blueman-mechanism[203823]: Unable to init server: Could not connect: Connection refused
Jul 04 13:30:34 o**-NUC7i5BNH blueman-mechanism[203823]: Unable to init server: Could not connect: Connection refused
Jul 04 13:30:34 o**-NUC7i5BNH blueman-mechani[203823]: gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed
Jul 04 13:30:34 o**-NUC7i5BNH dbus-daemon[1367]: [system] Successfully activated service 'org.blueman.Mechanism'
Jul 04 13:31:04 o**-NUC7i5BNH systemd[1]: blueman-mechanism.service: Succeeded.

cowsay | Thanks for checking this out!