blueman-project/blueman #421: Samsung Wireless Audio - Connection Failed

sgaucoin, 2015-12-29:
Device is found. Can pair and trust but audio sink connects and disconnects immediately.

infirit, 2015-12-29:
Outside of the profile switching blueman does not do anything special and should be similar to using ``bluetoothctl``. Can you follow the steps described in __debugging__ section on the following page, https://github.com/blueman-project/blueman/wiki/Troubleshooting and paste them to paste site or a github gist?

sgaucoin, 2015-12-30:
thanks for looking into this...here the gist link

https://gist.github.com/sgaucoin/4d03e865c3c4d421a255

cschramm, 2016-01-03:
Can't see anything wrong except you're using a very old version of blueman. I'd suggest to try a newer one and / or bluetoothctl.

sgaucoin, 2016-01-03:
Thanks. The bluez' version is the package that comes with the recent stable
version of Linux Mint. Ubuntu (Mint's base) has been extremely slow on
upgrading (i.e. I think over 4 years) to the newer version. Will attempt to
compile/install bluez 5.37.

infirit, 2016-01-03:
Upgrading BlueZ is a PITA as you will likely also need a newer version of pulseaudio. Better is to just try blueman 2.0.X from a more recent ubuntu release and see how it goes. I think wily should do but as a gentoo guy I am not too familiar with binary distro's.

sgaucoin, 2016-01-03:
PITA is an understatement....spent most of the morning
compiling/making/breaking to no avail. Since this is a relatively new ditro
installation, I may just nuke and pave and install something more bluez 5
friendly. I'm thinking Ubuntu Wily but any suggestions are welcome. I'm not
too familiar with linux distros.

cschramm, 2016-01-17:
You can just download a recent package e.g. [here](http://packages.ubuntu.com/xenial/blueman#pdownload) and install it with e.g. `dpkg`. The dependencies should work in Linux Mint 17.3 (you may need to run `aptitude safe-upgrade` after `dpkg` to install them).

You can also try to connect with `hcitool` and see what the result is there.

sgaucoin, 2016-01-17:
Thanks for the reply. Tried multiple times installing more recent bluetooth
packages but nothing worked with Mint 17.3. Since it was a clean install, I
just decided to try other packages. I've installed Ubuntu Wily 15.10 which
has the latest bluez package and everything works fine.

Thanks again.

Stephane

infirit, 2016-01-17:
Ok, thanks, I'll close it then.

closed 2016-01-17 by infirit