blueman-project/blueman #422: bluaman cannot connect device after some time

akostadinov, 2016-01-05:
Hello, I see a number of related issues: #240, #189, #254. Basically after some time, when trying to connect to keyboard, I'm getting (in blueman-applet):
```
Connection Failed: Connection ":1.39" is not allowed to add more match rules (increase limits in configuration file if required)
```
I am wondering if this can be fixed. What I see in log is:
```
Jan  5 12:52:09 localhost dbus[1204]: [system] Activating service name='org.bluema
n.Mechanism' (using servicehelper)
Jan  5 12:52:09 localhost org.blueman.Mechanism: Unable to init server: Could not 
connect: Connection refused
Jan  5 12:52:09 localhost org.blueman.Mechanism: Unable to init server: Could not 
connect: Connection refused
Jan  5 12:52:09 localhost blueman-mechanism: Starting blueman-mechanism
Jan  5 12:52:09 localhost dbus[1204]: [system] Successfully activated service 'org
.blueman.Mechanism'
Jan  5 12:52:09 localhost blueman-mechanism: loading Ppp
Jan  5 12:52:09 localhost blueman-mechanism: loading Rfcomm
Jan  5 12:52:09 localhost blueman-mechanism: loading RfKill
Jan  5 12:52:09 localhost blueman-mechanism: loading Network
Jan  5 12:52:09 localhost org.blueman.Mechanism: (blueman-mechanism:15537): Gtk-CR
ITICAL **: gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' faile
d
Jan  5 12:52:39 localhost blueman-mechanism: Exiting
```

It initially works for 2-3 days (perhaps 5-6 reconnections per day). I've also filed a fedora bug but I doubt the package maintainer would be able to help.
https://bugzilla.redhat.com/show_bug.cgi?id=1295724
--
blueman: 2.0
BlueZ: 5.36
Distribution: 
Desktop: XFCE

infirit, 2016-01-05:
@akostadinov, does killing and restarting blueman work?

@cschramm, I have been thinking of turning the dbus wrapper modules like Device and Adapter into singletons so we only ever use a single (python) object for each Device, Adapter etc. Currently we always create new object every time we call the class. Do you think this would help with this?

akostadinov, 2016-01-05:
@infirit, I think so, because running `blueman-report` I read that it terminates applet and now I tried disconnect/connect a few times and it is working again.

cschramm, 2016-01-06:
Yes, that should work. However, I guess the issue might not be present in current master anyway.

infirit, 2016-01-06:
@akostadinov, if you are up for it you could build current master with the rpm spec file from [here](http://pkgs.fedoraproject.org/cgit/rpms/blueman.git/tree/?h=f23) and then update the commit at the top to bb52fff62abf33a786f1c923df46b5171aa3354a.

akostadinov, 2016-01-06:
@infirit, sorry, you mean to cherry-pick bb52fff on top of master or build bb52fff itself? I can try in the following days but please confirm how to use that commit.

infirit, 2016-01-06:
@akostadinov, no just build a new rpm. I used the below to get a working package of the latest git.

 * run ``dnf install @development-tools fedora-packager rpmdevtools`` to download the neccesary tools
 * run ``rpmdev-setuptree`` to create the dir structure in your home. The dirs will be in ``~/rpmbuild``.
 * download [this](http://pkgs.fedoraproject.org/cgit/rpms/blueman.git/plain/blueman.spec?h=f23) spec file and place it in ``~/rpmbuild/SPECS/``
 * download [this](http://pkgs.fedoraproject.org/cgit/rpms/blueman.git/plain/fix_bluetoothd_path.patch?h=f23) patch and place it in ``~/rpmbuild/SOURCES/``
 * Modify the spec file and update the line starting with ``%global gitcommit`` so it has commit ``bb52fff62abf33a786f1c923df46b5171aa3354a``
 * Also bump the release number (from 0.6 to 0.7) in the same spec file.
 * Now install blueman build dependancies with ``sudo dnf builddep SPECS/blueman.spec``
 * Now we build the rpm with ``rpmbuild -ba SPECS/blueman.spec``
 * If all goes well you now have a fresh blueman package in ``~/rpmbuild/RPMS``.
 * Install it with ``sudo dnf install RPMS/x86_64/blueman-2.1-0.6.gitbb52fff.fc23.x86_64.rpm``.

__Do note that I build this on a 64bit virtualbox vm so the name and file location may differ on your machine__.

Now restart your session or reboot you should have the latest git version of blueman installed.

akostadinov, 2016-01-06:
Thank you for detailed instructions. Will do and report back.

infirit, 2016-01-06:
Sigh noticed I forgot one step, to download the source from github with wget. I updated the steps to now include this, it is after downloading the patch.

akostadinov, 2016-01-07:
FYI didn't have any time yesterday but just compiled and installed per your extremely detailed instructions (must put it in a wiki somewhere). I also did `sudo systemctl restart blueman-mechanism.service` and restarted applet. Now time to wait!

akostadinov, 2016-01-09:
Hello, so far I'm always able to reconnect with bluemap-applet. I'll report again after some more days pass. I see though some error messages on my terminal where I started blueman-applet. Here I'm pasting them because I see some handler related:
```
$ blueman-manager version 2.0 starting
                                                                 /usr/lib64/python3.4/site-packages/gi/overrides/GObject.py:541: Warning: gsignal.c:2634: instance '0x55f6cedd1f80' has no handler with id '1217'
                                               return func(*args, **kwargs)

$ 
$ Traceback (most recent call last):
                                                                 File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/SerialManager.py", line 60, in _on_device_property_changed
                         self.terminate_all_scripts(Device(path).Address)
                                                                           File "/usr/lib/python3.4/site-packages/blueman/main/Device.py", line 41, in __init__
                                                                                 self.Properties = self.Device.get_properties()
                                               File "/usr/lib/python3.4/site-packages/blueman/bluez/errors.py", line 148, in warp
                                                   raise parse_dbus_error(exception)
    File "/usr/lib/python3.4/site-packages/blueman/bluez/errors.py", line 146, in warp
        return func(*args, **kwargs)
                                      File "/usr/lib/python3.4/site-packages/blueman/bluez/PropertiesBase.py", line 51, in get_properties
                                                           return self.__properties_interface.GetAll(self._interface_name)
                                          File "/usr/lib64/python3.4/site-packages/dbus/proxies.py", line 70, in __call__
                                           return self._proxy_method(*args, **keywords)
       File "/usr/lib64/python3.4/site-packages/dbus/proxies.py", line 145, in __call__
         **keywords)
                      File "/usr/lib64/python3.4/site-packages/dbus/connection.py", line 651, in call_blocking
                                message, timeout)
                                                 dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownObject: Method "GetAll" with signature "s" on interface "org.freedesktop.DBus.Properties" doesn't exist

                                                     Traceback (most recent call last):
       File "/usr/lib/python3.4/site-packages/blueman/bluez/errors.py", line 146, in warp
           return func(*args, **kwargs)
                                         File "/usr/lib/python3.4/site-packages/blueman/bluez/NetworkServer.py", line 17, in register
                                                       self._interface.Register(uuid, bridge)
             File "/usr/lib64/python3.4/site-packages/dbus/proxies.py", line 70, in __call__
              return self._proxy_method(*args, **keywords)
                                                            File "/usr/lib64/python3.4/site-packages/dbus/proxies.py", line 145, in __call__
                                                              **keywords)
                                                                           File "/usr/lib64/python3.4/site-packages/dbus/connection.py", line 651, in call_blocking
                                                                                     message, timeout)
                    dbus.exceptions.DBusException: org.bluez.Error.AlreadyExists: Already Exists

              During handling of the above exception, another exception occurred:

                                                                                 Traceback (most recent call last):
                                   File "/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py", line 212, in Run
                                            ret = getattr(inst, function)(*args, **kwargs)
          File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/Networking.py", line 54, in on_adapter_added
                                      self.update_status()
                                                            File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/Networking.py", line 57, in update_status
                                                                                     self.set_nap(self.Config["nap-enable"])
                                            File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/Networking.py", line 70, in set_nap
                                                               s.register("nap", "pan1")
        File "/usr/lib/python3.4/site-packages/blueman/bluez/errors.py", line 148, in warp
            raise parse_dbus_error(exception)
                                             blueman.bluez.errors.DBusAlreadyExistsError: Already Exists
                      blueman-manager version 2.0 starting
                                                          /usr/lib64/python3.4/site-packages/gi/overrides/GObject.py:541: Warning: gsignal.c:2634: instance '0x55ad20169aa0' has no handler with id '1169'
                                        return func(*args, **kwargs)
                                                                    blueman-manager version 2.0 starting
[... cut ...]
                                                    raise parse_dbus_error(exception)
   blueman.bluez.errors.DBusAlreadyExistsError: Already Exists
                                                              blueman-manager version 2.0 starting
                /usr/lib64/python3.4/site-packages/gi/overrides/GObject.py:541: Warning: gsignal.c:2634: instance '0x562185a51aa0' has no handler with id '1166'
                                                                                return func(*args, **kwargs)
```

infirit, 2016-01-10:
The error come from 7f7c6a502aeff5bea9267f0c03ee5737b42bbf99, I'll have a look and fix it.

infirit, 2016-01-10:
>The error come from 7f7c6a5, I'll have a look and fix it.

Fixed in b9c06dafdeab230e6b3220dd3472004a9d3eeaa7

akostadinov, 2016-01-10:
Today I've got again `Connection Failed: Connection ":1.274" is not allowed to add more match rules (increase limits in configuration file if required)` :/

infirit, 2016-01-11:
Ok, I have to delve into the mystical world python class and object creation. Hope I do not go mad :smiling_imp:

infirit, 2016-01-11:
Wait, something is not right in ``blueman.main.Device``, it seem to go nuts.

@cschramm, do I remember correctly that at some point we should get rid of ``blueman.main.Device``?

infirit, 2016-01-11:
I reproduced this and here is [my full log](https://gist.github.com/infirit/93dbb8d55b460810c1e7). The interesting bit is below and from then on things are broken.

```python
_________
__init__ (/usr/lib/python3.4/site-packages/blueman/main/Device.py:23)
caching initial properties 
Traceback (most recent call last):
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/SerialManager.py", line 60, in _on_device_property_changed
    self.terminate_all_scripts(Device(path).Address)
  File "/usr/lib/python3.4/site-packages/blueman/main/Device.py", line 47, in __init__
    self._manager = Manager()
  File "/usr/lib/python3.4/site-packages/blueman/bluez/errors.py", line 148, in warp
    raise parse_dbus_error(exception)
  File "/usr/lib/python3.4/site-packages/blueman/bluez/errors.py", line 146, in warp
    return func(*args, **kwargs)
  File "/usr/lib/python3.4/site-packages/blueman/bluez/Manager.py", line 27, in __init__
    self._handle_signal(self._on_interfaces_added, 'InterfacesAdded')
  File "/usr/lib/python3.4/site-packages/blueman/bluez/Base.py", line 33, in _handle_signal
    self.__bus.add_signal_receiver(*args, path_keyword=path_keyword)
  File "/usr/lib64/python3.4/site-packages/dbus/bus.py", line 160, in add_signal_receiver
    self.add_match_string(str(match))
  File "/usr/lib64/python3.4/site-packages/dbus/bus.py", line 398, in add_match_string
    BUS_DAEMON_IFACE, 'AddMatch', 's', (rule,))
  File "/usr/lib64/python3.4/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.LimitsExceeded: Connection ":1.49" is not allowed to add more match rules (
```

akostadinov, 2016-02-09:
Hello, I see you guys are not losing your time and have already removed the main device (whatever it was) 3019f0699, also I see some other dbus and NM related clean-ups in commit log. Do you think this issue could be fixed or at least better now? Should I try a new build?

infirit, 2016-02-09:
Not yet. I like to get the GDbus migration done first as it makes the Bluez clases simpler. And while reading up on the singleton concept it sound simple to implement, but getting it right is tricky :smile:

infirit, 2016-02-26:
@akostadinov, https://github.com/infirit/blueman/commit/ffc2269a8e329d253ce26829a14fdd2f2d7ac4db should alleviate the problem and possibly solve it for most people. Can you try it out?

akostadinov, 2016-02-29:
Just compiled and installed. Will report back how's it going.

akostadinov, 2016-03-03:
Using the new version for 2-3 days. I didn't hit the exact issue but this morning, after resume laptop from suspend I found blueman applet missing icon from XFCE panel. I have pasted some part of xsession-errors, maybe not everything related to blueman but I can't figure how to know what exactly is from blueman and what is something else: http://fpaste.org/332745/

infirit, 2016-03-03:
Strange, which commit id did you use and did you add the patch in the spec file?

infirit, 2016-03-03:
It has been running well for me for a while and have not had any issues. So I merged it and if you follow earlier instructions, just replace the commit with ``19d39087557339fb8021e674c8fd511122e49efc``.

akostadinov, 2016-03-03:
I have used ffc2269 and I didn't remove the patch from rpmbuild dir so I guess it was used.

cschramm, 2016-03-03:
Guess something must have been wrong with the setup. xsession-errors refers to `blueman/main/Device.py` which is not part if ffc2269.

infirit, 2016-03-03:
@akostadinov, yeah there was an assumption on my part which caused confusion, sorry about that. I updated the [original instruction](https://github.com/blueman-project/blueman/issues/422#issuecomment-169293906) to reflect the new commit hash. This should also take care of the other error that was in the log.

akostadinov, 2016-03-03:
It is possible that I'm pasting from an old log :/ THere is not time in it. I'll install from master and wipe old log to avoid further confusion, sorry.

akostadinov, 2016-03-09:
FYI this morning I've seen DBus input/output error but a second attempt to connect succeeded. So I guess things are much better now.

akostadinov, 2016-03-15:
Hello again. I think I've now used the new version long enough to conclude the old issue is gone. I'm getting the dbus error above from time to time but after usually one more attempt it succeeds. Sometimes a few attempts to connect are needed though. In any case situation now is a lot better. And original issue is gone as far as I can tell.

btw I don't see anything in ` ~/.xsession-errors` so I can't tell anything more about the above error message.

Thank you a lot for fixing that issue!

infirit, 2016-03-15:
Good to see it works well. The error means that bluez is unable to communicate with the device. It could be that either the adapter and/or the device is not ready yet after resuming.

akostadinov, 2016-03-15:
I am wondering if that's the reason I have to manually conect the device in the first place. When I try to connect the keyboard and watch at blueman applet, the connection bars are appearing and disappearing. And only once I try to connect manually, then it manages to connect. Only very rarely the connection is established by itself.

This is lenovo t530 intel wireless. While on another machine a $5 dongle worked flawlessly. I'm not sure blueman can help with that.

Thank you again for spending time to resolve the issue.

infirit, 2016-03-15:
It could be that the PowerManager plugin and killswitch is causing issues here. Can you disable the PowerManager plugin and see if this makes a difference?

akostadinov, 2016-03-16:
@infirit, behaves like that regardless of killswitch and power manager plug-ins.

akostadinov, 2016-06-12:
After a lot of time I in fact got the issue back. Here's the backtrace. This time I don't see `main/Device.py` and I have installed blueman-2.1-0.7.git19d3908.fc23.x86_64
```
AttributeError: 'NoneType' object has no attribute 'get_object_path'
Traceback (most recent call last):
  File "/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py", line 21
2, in Run
    ret = getattr(inst, function)(*args, **kwargs)
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/ShowConnected.py
", line 111, in on_adapter_added
    self.enumerate_connections()
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/ShowConnected.py
", line 52, in enumerate_connections
    devices = adapter.list_devices()
  File "/usr/lib/python3.4/site-packages/blueman/bluez/Adapter.py", line 35, in 
list_devices
    return [Device(device) for device in devices]
  File "/usr/lib/python3.4/site-packages/blueman/bluez/Adapter.py", line 35, in 
<listcomp>
    return [Device(device) for device in devices]
  File "/usr/lib/python3.4/site-packages/blueman/bluez/Base.py", line 49, in __n
ew__
    instance._init(*args, **kwargs)
  File "/usr/lib/python3.4/site-packages/blueman/bluez/Device.py", line 14, in _
init
    super(Device, self)._init(interface_name=self._interface_name, obj_path=obj_path)
  File "/usr/lib/python3.4/site-packages/blueman/bluez/PropertiesBase.py", line 31, in _init
    path_keyword='path')
  File "/usr/lib/python3.4/site-packages/blueman/bluez/Base.py", line 111, in _handle_signal
    self.__bus.add_signal_receiver(*args, path_keyword=path_keyword)
  File "/usr/lib64/python3.4/site-packages/dbus/bus.py", line 157, in add_signal_receiver
    watch = self.watch_name_owner(bus_name, callback)
  File "/usr/lib64/python3.4/site-packages/dbus/bus.py", line 373, in watch_name_owner
    return NameOwnerWatch(self, bus_name, callback)
  File "/usr/lib64/python3.4/site-packages/dbus/bus.py", line 80, in __init__
    arg0=bus_name)
  File "/usr/lib64/python3.4/site-packages/dbus/bus.py", line 160, in add_signal_receiver
    self.add_match_string(str(match))
  File "/usr/lib64/python3.4/site-packages/dbus/bus.py", line 398, in add_match_string
    BUS_DAEMON_IFACE, 'AddMatch', 's', (rule,))
  File "/usr/lib64/python3.4/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.LimitsExceeded: Connection ":1.41" is not allowed to add more match rules (increase limits in configuration file if required)
Traceback (most recent call last):
  File "/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py", line 212, in Run
    ret = getattr(inst, function)(*args, **kwargs)
  File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/GameControllerWakelock.py", line 35, in on_device_property_changed
    klass = bluez.Device(path).get_properties()["Class"] & 0x1fff
  File "/usr/lib/python3.4/site-packages/blueman/bluez/Base.py", line 49, in __new__
    instance._init(*args, **kwargs)
  File "/usr/lib/python3.4/site-packages/blueman/bluez/Device.py", line 14, in _init
    super(Device, self)._init(interface_name=self._interface_name, obj_path=obj_path)
  File "/usr/lib/python3.4/site-packages/blueman/bluez/PropertiesBase.py", line 31, in _init
    path_keyword='path')
  File "/usr/lib/python3.4/site-packages/blueman/bluez/Base.py", line 111, in _handle_signal
    self.__bus.add_signal_receiver(*args, path_keyword=path_keyword)
  File "/usr/lib64/python3.4/site-packages/dbus/bus.py", line 157, in add_signal_receiver
    watch = self.watch_name_owner(bus_name, callback)
  File "/usr/lib64/python3.4/site-packages/dbus/bus.py", line 373, in watch_name_owner
    return NameOwnerWatch(self, bus_name, callback)
Object::connect: No such signal org::freedesktop::UPower::Device::Changed()
```

Not as big issue as before, because it took weeks. Maybe I should open another bug?

infirit, 2016-06-13:
>Maybe I should open another bug

No, it is the same problem, just harder to reproduce now, I'll reopen for now so I can poke at this some more :smile:. 

However, master now uses GDBus so it would be best to build that and post the traceback of it.

akostadinov, 2016-06-16:
built off fc570cf , btw the patch file is now incorrect as new filename it should patch is `apps/blueman-report.in`

infirit, 2016-06-17:
Thx for sticking with this :+1:. The patch is not relevant for this so you could comment it out. But you can edit the patch and add the ``.in`` to the lines with the filename. When we release 2.1 I am sure the maintainer will update the patch.

akostadinov, 2016-07-19:
FYI in addition to `apps/blueman-report.in` for compiling f810c73 on fedora 24 I had to change `SPECS/blueman.spec` from `%global _py3_ver 3.4` to `%global _py3_ver 3.5`. Also had to install `python3-devel` package.

Have not seen the issue since I wrote last time but I'm changing hardware now which makes it probably less likely to hit it again. Will report if I do though.

While on it, I saw some compile warnings:
```
_blueman.c: In function '__pyx_pf_8_blueman_9conn_info_4get_rssi':
_blueman.c:3243:58: warning: pointer targets in passing argument 2 of 'connection_get_rssi' differ in signedness [-Wpointer-sign]
   __pyx_v_res = connection_get_rssi((&__pyx_v_self->ci), (&__pyx_v_rssi));
                                                          ^
In file included from _blueman.c:245:0:
libblueman.h:25:5: note: expected 'int8_t * {aka signed char *}' but argument is of type 'char *'
 int connection_get_rssi(struct conn_info_handles *ci, int8_t *ret_rssi);
     ^~~~~~~~~~~~~~~~~~~
_blueman.c: In function '__pyx_pf_8_blueman_9conn_info_8get_tpl':
_blueman.c:3523:57: warning: pointer targets in passing argument 2 of 'connection_get_tpl' differ in signedness [-Wpointer-sign]
   __pyx_v_res = connection_get_tpl((&__pyx_v_self->ci), (&__pyx_v_tpl), __pyx_t_1);
                                                         ^
In file included from _blueman.c:245:0:
libblueman.h:27:5: note: expected 'int8_t * {aka signed char *}' but argument is of type 'char *'
 int connection_get_tpl(struct conn_info_handles *ci, int8_t *ret_tpl, uint8_t type);
     ^~~~~~~~~~~~~~~~~~
modem-prober.c: In function 'do_probe':
modem-prober.c:512:3: warning: 'fd' may be used uninitialized in this function [-Wmaybe-uninitialized]
   close(fd);
   ^~~~~~~~~
```

netikras, 2022-05-30:
> [cschramm](https://github.com/cschramm) added this to the [2.1](https://github.com/blueman-project/blueman/milestone/2) milestone [on 6 Oct 2016](https://github.com/blueman-project/blueman/issues/422#event-814031792)

Does this mean versions 2.1 onwards should no longer have this problem?

LinuxMint 19.1 repositories are using 2.0.5 and, like in the case of @RecNes , I see this popping up rn.

cschramm, 2022-05-31:
@netikras: 2.1.alpha1 and later do not run into that issue, yes.