blueman-project/blueman #423: Blueman inclusion in openSUSE

posophe, 2016-01-06:
Hi !

I'm working on openSUSE packaging actually, to provide it as default bluetooth manager in Mate. The Security just gave me a negative answer (I quote) :
"Still the same issues with blueman. Every DBUS function running as root,
allowed to be called by anyone without any authorization.

CVE-2015-8612 as an recent example.

Cant be whitelisted."

Can anyone tell me if it is possibly fixable on upstream ?

Regards.
Benjamin

cschramm, 2016-01-06:
No it's not. blueman does some things that require root privileges (network, PPP, rfcomm, rfkill). Those tasks are done by blueman-mechanism which provides an API to anybody on the system as obviously anybody could potentially use blueman. If that's not the case on openSUSE and there is only one potential user, you could install respective dbus rules, however that does not make much sense both from a functional and a security viewpoint.

infirit, 2016-01-07:
Do you know why policykit is not used for this? Policykit is designed to do privileged operations like enabling port forwarding.

cschramm, 2016-01-07:
It actually is. The DHCP part of blueman-mechanism checks the policykit authentication. The policy file also includes two other actions, but it looks like they are not used anymore. We could add the check to all API methods. However the only improvements I see would be that inactive sessions would not be allowed to use the methods by default and distributions could easily overwrite the rules. I don't think that's what openSUSE security "asks" for as it is not such a big improvement.

infirit, 2016-01-07:
Sorry, I meant why do we not use polkit as the normal user to do the privileged actions. Or put it another way, why do we not run mechanism directly as the user (not as root) and if needed it will prompt for credentials.

cschramm, 2016-01-07:
Ah, I see. It would be pretty annoying to have to do the authorization on every start / action / connection, wouldn't it? 😐

infirit, 2016-01-07:
Well, yes and no :smile:. There are options for distro's to create rules that allow this to be done without password for user in group like admin or wheel. I use the below for MATE. I think this a large part of the Suse folks object to

````
/* Allow users in wheel group to change various MATE settings requiring root priviliges */
polkit.addRule(function(action, subject) {
    if (action.id == "org.mate.settingsdaemon.datetimemechanism.settimezone" ||
        action.id == "org.mate.settingsdaemon.datetimemechanism.settime" ||
        action.id == "org.mate.settingsdaemon.datetimemechanism.configurehwclock" ||
        action.id == "org.mate.cpufreqselector" &&
        subject.isInGroup("wheel")) {
        return polkit.Result.YES;
    }
});
````

infirit, 2016-01-07:
quick experiment, with the below patch I get a permission denied when doing privileged operations. I tested this on a Fedora23 vm as I broke my box again.

```diff
From f078cedc551043c4d6f480e24e2d561f88886acc Mon Sep 17 00:00:00 2001
From: infirit <infirit@gmail.com>
Date: Thu, 7 Jan 2016 12:37:33 +0100
Subject: [PATCH] Fix policy

---
 data/configs/org.blueman.Mechanism.conf | 4 ++++
 1 file changed, 4 insertions(+)

diff --git a/data/configs/org.blueman.Mechanism.conf b/data/configs/org.blueman.Mechanism.conf
index f2e9ccc..b4b5719 100644
--- a/data/configs/org.blueman.Mechanism.conf
+++ b/data/configs/org.blueman.Mechanism.conf
@@ -6,8 +6,12 @@
 <busconfig>
 	<policy user="root">
 		<allow own="org.blueman.Mechanism"/>
+		<allow send_destination="org.blueman.Mechanism"/>
+		<allow receive_sender="org.blueman.Mechanism"/>
 	</policy>
   	<policy context="default">
     		<allow send_destination="org.blueman.Mechanism"/>
+		<allow receive_sender="org.blueman.Mechanism"/>
   	</policy>
 </busconfig>
-- 
2.7.0
```

And adding these rules makes them go away.

```
/* Allow users in wheel group to run blueman network config without authentication */
polkit.addRule(function(action, subject) {
    if ((action.id == "org.blueman.network.setup" ||
         action.id == "org.blueman.bluez.config" || 
         action.id == "org.blueman.dhcp.client") &&
        subject.isInGroup("wheel")) {
        return polkit.Result.YES;
    }
});
```

posophe, 2016-01-07:
Thank you very much for all your efforts, very glad to see that :). 
@infirit : indeed, if you implement the polkit checking process in blueman, I'll ask for blueman to be added in openSUSE allowed to be launched without password program list.

infirit, 2016-08-13:
The PR got merged and a patch against 2.0 was provided. Anything else holding this up for OpenSuse?

posophe, 2016-08-13:
It should be fine. I didn't take the time to work on blueman, too busy the last times

infirit, 2016-08-13:
No worries, just wondering if we needed to keep this open. Let's close this when it actually lands :smile:

closed 2016-10-05 by cschramm