blueman-project/blueman #425: RFC: Drop Python2

infirit, 2016-01-11:
This allows us to remove some ugly code like below to check for strings.
```python
if hasattr(instance, "format") and hasattr(instance, "upper"):
```

AFAIK all ditro's now have a recent enough python 3 version.

yars068, 2016-01-12:
Please no! :) Slackware currently will keep python2.

infirit, 2016-01-13:
:sob:, also not in the next release @yars068 ?

yars068, 2016-01-16:
The next release still not ready, currently it is a 14.2 beta. But, maybe Patrick will migrate to Python3, or do Python2/Python3 mix, I don't know. This time, we still have Python2... :-(

cschramm, 2016-01-17:
I think specific handlings for python 2 (__builtins__ and hasattr) account for less than 10 lines of code. Not a big thing.

infirit, 2016-01-17:
It is not just the code. It is the maintenance of keeping things working in both versions. For example I saw you GVariant map in the gdbus branch that maps the python types to variant signatutres which  was missing the ``unicode`` type so it would have failed running on python2.

But it is not a big problem for now but I will pester ppl again after blueman 2.1 :smiling_imp:

closed 2016-01-17 by infirit