blueman-project/blueman #429: BUG/RFE: plugin SerialManager: on-connect script: invocation always fails

Lanchon, 2016-01-12:
hi,

blueman: 1.99~alpha2-1~trusty~ppa4
BlueZ: 4.101-0ubuntu13.1

the SerialManager plugin supports running a script when an rfcomm connection is established. this is totally borked, and the script is never invoked. popup says:

    Serial port connection script failed
    There was a problem launching script [xxx]
    argument 2 to map() must support iteration

problem lies in [this line](https://github.com/blueman-project/blueman/blob/6b6a361931bb513644feff7d4be660d8224d1d50/blueman/plugins/applet/SerialManager.py#L106)

but please, please, *pretty please...* :)

since clearly nobody is using this functionality -given that it doesn't work at all-, instead of fixing it, could you please implement it better? instead of passing all parameters with an imposed order, please allow named parameter expansion in the script specification. that way we don't have to create silly scripts for every use. (see the last sample [here](https://pythonadventures.wordpress.com/2011/04/04/new-string-formatting-syntax/).)

thanks so much
and thanks for blueman!

infirit, 2016-01-13:
>instead of passing all parameters with an imposed order, please allow named parameter expansion in the script specification.

I am not sure I understand what you are suggesting. The script that is executed can be of any type Perl, Ruby or even a C program. And the script is launched just like you would do from a terminal.

To the particular problem, I don't have a serial device but I'll' attempt to fix it at least for you until @cschramm ok's it. Can you apply the below and let me know what the output is.

```diff
diff --git a/blueman/plugins/applet/SerialManager.py b/blueman/plugins/applet/SerialManager.py
index 18cc1dd..e8345bc 100644
--- a/blueman/plugins/applet/SerialManager.py
+++ b/blueman/plugins/applet/SerialManager.py
@@ -99,6 +99,9 @@ class SerialManager(AppletPlugin):
         GObject.child_watch_add(process.pid, self.on_script_closed, (address, node))
 
     def call_script(self, address, name, sv_name, uuid16, node):
+        dprint("** DEBUG **")
+        dprint(type(uuid16))
+        dprint(address, name, sv_name, uuid16, node)
         c = self.get_option("script")
         if c and c != "":
             args = c.split(" ")
```

infirit, 2016-01-13:
Ok, remembered my cordless home phone exposes a serial service. It seems we now get an int instead of a iterable so the below should work for you. __I do still like to see the output__ :smile:.

```diff
diff --git a/blueman/plugins/applet/SerialManager.py b/blueman/plugins/applet/SerialManager.py
index 18cc1dd..d82e7ac 100644
--- a/blueman/plugins/applet/SerialManager.py
+++ b/blueman/plugins/applet/SerialManager.py
@@ -103,7 +103,7 @@ class SerialManager(AppletPlugin):
         if c and c != "":
             args = c.split(" ")
             try:
-                args += [address, name, sv_name, ",".join(map(lambda x: hex(x), uuid16)), node]
+                args += [address, name, sv_name, "%s" % hex(uuid16), node]
                 dprint(args)
                 p = Popen(args, preexec_fn=lambda: os.setpgid(0, 0))
```

infirit, 2016-01-13:
So, it's pretty broken alright :smile_cat:. I just pushed af2956af4e2a71178c29535fcbdec046acb90c06 and 294a308bf603500b24780ca9515fe6669bd31df8 to fix other problems which you will need to apply on top of the above. This till leaves a problem with disconnection failing in dbus.

Lanchon, 2016-01-13:
>I am not sure I understand what you are suggesting. The script that is executed can be of any type Perl, Ruby or even a C program. And the script is launched just like you would do from a terminal.

this is what i mean: i don't know python but it looks like you invoke the named script with these parameters: address, name, sv_name, uuid16, node.

now imagine that i want to automatically open "gtkterm" whenever an rfcomm device is connected. right now i need to create an intermediate script with this content:

    gtkterm --port "$5"

givie it a silly name, and enter the path in the script field of the serial plugin. this is a total drag!

it would be much better if i could just enter this into the script field of the serial plugin:

    gtkterm --port "{node}"

and have the plugin expand the parameters, then executing the resulting command.

Lanchon, 2016-01-13:
thanks! don't worry about an immediate solution for me, i can wait for the next release. :)

infirit, 2016-01-13:
Ok, I see now what you mean. Let me first make it work and then @cschramm and I can discuss if this makes sense. I think it does but I do not have a lot of blueman serial experience.

infirit, 2016-01-17:
Ok, hope to have resolved all the issues now and things appear to work ok for my cordless home phone. See #431 for this fixes.

closed 2016-10-05 by cschramm

Lanchon, 2016-10-13:
thanks! and sorry i didn't reply. i was working on a prototype robot at the time that i don't have access to anymore so i couldn't really check the fixes you made. thanks again!