blueman-project/blueman #430: bluetooth device manager preferences, connect, and setup buttons not working. Pair works, but pair dialogue doesn't pop up, so pairing is always unsuccessful.  Have updated and uninstalled and reinstalled

ericltongfb, 2016-01-12:


infirit, 2016-01-12:
Have a look at https://github.com/blueman-project/blueman/wiki/Troubleshooting and report back,

Do make sure to provide the debug from before you attempt till after.

ericltongfb, 2016-01-12:
I get this output:
 sudo blueman-applet report
Loading configuration plugins
blueman-applet version 1.99.alpha1 starting
Using GConf config backend
_________
Load (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:54)
['NMDUNSupport', 'NMPANSupport', 'Headset', 'AuthAgent', 'Menu', 'SerialManager', 'NetUsage', 'StatusIcon', 'DBusService', 'AppIndicator', 'PowerManager', 'StandardItems', 'RecentConns', 'ShowConnected', 'KillSwitch', 'DhcpClient', 'TransferService', 'DiscvManager', 'Networking', 'ExitItem', 'PPPSupport'] 
Using GConf config backend
_________
get_interface_version (/usr/lib/python2.7/dist-packages/blueman/bluez/BlueZInterface.py:8)
Detected BlueZ 5 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
Using GConf config backend
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
Traceback (most recent call last):
  File "/usr/bin/blueman-applet", line 110, in <module>
    BluemanApplet()
  File "/usr/bin/blueman-applet", line 53, in __init__
    self.Plugins.Load()
  File "/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py", line 115, in Load
    if (cls.__autoload__ or cls.__name__ in c) and not (cls.__unloadable__ and "!" + cls.__name__ in c):
TypeError: argument of type 'NoneType' is not iterable

infirit, 2016-01-12:
Alpha1 is very old and it would be better if you upgrade to the latest stable (2.0.3), which distro and version are you on?

The problem is likely fixed by https://github.com/blueman-project/blueman/commit/8ce4f61ed160fdd940d8330ec4fd05e977b59ba2 which is in version alpha2 and up.

ericltongfb, 2016-01-13:
thanks for the help, I used sudo apt-get dist-upgrade, and when I use blueman, now the bluetooth icon pops up in my toolbar, but still can't use the aforementioned buttons

sudo blueman-applet report
Loading configuration plugins
blueman-applet version 1.99.alpha1 starting
Stale PID, overwriting
Using GConf config backend
_________
Load (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:54)
['NMDUNSupport', 'NMPANSupport', 'Headset', 'AuthAgent', 'Menu', 'SerialManager', 'NetUsage', 'StatusIcon', 'DBusService', 'AppIndicator', 'PowerManager', 'StandardItems', 'RecentConns', 'ShowConnected', 'KillSwitch', 'DhcpClient', 'TransferService', 'DiscvManager', 'Networking', 'ExitItem', 'PPPSupport'] 
Using GConf config backend
_________
get_interface_version (/usr/lib/python2.7/dist-packages/blueman/bluez/BlueZInterface.py:8)
Detected BlueZ 5 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
Using GConf config backend
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Headset.Headset'> 
Using GConf config backend
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Networking.Networking'> 
Using GConf config backend
_________
load_nap_settings (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/Networking.py:30)
Loading NAP settings 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.TransferService.TransferService'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'> 
Using GConf config backend
_________
on_load (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/KillSwitch.py:25)
Using the new killswitch system 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.DhcpClient.DhcpClient'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.SerialManager.SerialManager'> 
Using GConf config backend
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.RecentConns.RecentConns'> 
Using GConf config backend
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
Not loading PPPSupport because it's conflict has higher priority 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.StandardItems.StandardItems'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'> 
Using GConf config backend
_________
update_menuitems (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/DiscvManager.py:113)
warning: Adapter is None 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.AppIndicator.AppIndicator'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
Not loading NMPANSupport because it's conflict has higher priority 
_________
update_state (/usr/lib/python2.7/dist-packages/blueman/main/KillSwitchNG.py:162)
State: True 
_________
on_switch_added (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/KillSwitch.py:45)
killswitch registered 0 
_________
on_obex_owner_changed (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/TransferService.py:70)
obex owner changed:  
_________
on_dbus_name_owner_change (/usr/bin/blueman-applet:85)
org.bluez owner changed to :1.3 
_________
set_nap (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/Networking.py:61)
set nap False 
_________
register_agent (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/AuthAgent.py:56)
Registering agent 
_________
initialize (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/RecentConns.py:138)
rebuilding menu 
_________
on_obex_owner_changed (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/TransferService.py:70)
obex owner changed: :1.4 
Using GConf config backend
_________
enumerate_connections (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/ShowConnected.py:45)
Found 0 existing connections 
_________
err (/usr/lib/python2.7/dist-packages/blueman/ods/OdsManager.py:116)
Couldn't create opp server (DBusException(dbus.String(u'Could not create server socket'),),) 
_________
err (/usr/lib/python2.7/dist-packages/blueman/ods/OdsManager.py:116)
Couldn't create ftp server (DBusException(dbus.String(u'Could not create server socket'),),)

infirit, 2016-01-13:
Again, this alpha version had many bugs wrt Bluez5 and I highly recommend to upgrade to the latest stable in the 2.0.X series. If you are on debian (still waiting to hear your distro) try the package from stretch.

What is the output of ``bluetoothctl list``?

infirit, 2016-01-13:
>try the package from stretch.

Sorry, I did not know 2.0.3 was in jessie backports, please use backports version of blueman.

cschramm, 2016-01-17:
The output of blueman-manager when clicking on the buttons would be interesting.

closed 2016-04-18 by cschramm