blueman-project/blueman #432: Bluez 5 has no audio api

infirit, 2016-01-15:
It looks like this was removed but we still have a Headset proxy and plugin. Should these be removed?

cschramm, 2016-01-17:
In fact there is no Headset1 interface. No idea if there is any way to implement that AnswerRequested handling for BlueZ 5. We should probably just drop the plugin, yes.

infirit, 2016-01-17:
I don't have a headset so I am not sure what interfaces we get from Bluez. But if there is no property to check against there is I think little we can do in blueman.

MatrixMike, 2016-01-18:
I have watching this thread with interest. I have had a surge of
interest in Bluetooth as I am working on a simple BT Android App. So
'collecting' BT devices and examining their properties. I am using
Blueman 1.23 with some success to connect to RNBT device for Arduino
purposes also.   Thanks to you guys for doing this work - not sure when
and how I can help yet - but I appreciate your work. Maybe I could test
a few things via the gui...  ?

infirit, 2016-01-18:
@MatrixMike 1.23 is BlueZ 4 only and old :smile:. Current release is 2.0 and the last version to support BlueZ 4. Current master which will become 2.1 will be for BlueZ 5 only.

Testing master (or 2.0) is always good if you are willing and let us know if we broke anything :smiley:. 

If you have a headset it would be nice if you could post the introspection under __BlueZ 5__. It will give me hopefully enough info (interfaces with methods and properties) to see if we can salvage this plugin. Example invocation below.

```
gdbus introspect --system --dest "org.bluez" --object-path "/org/bluez/hci0/dev_00_07_61_48_7A_2C"
```

infirit, 2016-01-27:
Opened PR #442 to remove the headset interface, service and plugin from blueman.

infirit, 2016-01-27:
It has been removed now, closing.

closed 2016-01-27 by infirit