blueman-project/blueman #433: BUG: serial connection is not properly torn down if the connection to the device is lost

Lanchon, 2016-01-17:
hi,

blueman: 1.99~alpha2-1~trusty~ppa4
BlueZ: 4.101-0ubuntu13.1

Serial connection is not properly torn down if the connection to the device is lost.

Steps to reproduce:

1. Connect to bluetooth serial port. An device is created, say /dev/rfcomm0.
2. Turn off device (eg: remove battery) without properly shutting down the bluetooth connection.
3. Note that /dev/rfcomm0 continues to exist.
4. Turn device on again, and again connect to its serial port. An device is created, say /dev/rfcomm1.

Workaround:

After loosing connection, kill root process "blueman-rfcomm-watcher". Next connection will be assigned to /dev/rfcomm0 again.

infirit, 2016-01-17:
I think you are seeing the same problems as mentioned in #429. I have pushed two fixes already and have a pull request you can follow here https://github.com/blueman-project/blueman/pull/431.

infirit, 2016-07-15:
The PR got merged so closing. If this still is an issue please let us know :smile:

closed 2016-07-15 by infirit