blueman-project/blueman #435: pyenv maybe doesn't like usage of "env python" in blueman

rworkman, 2016-01-18:
We got a report of problems with pyenv due to blueman's usage of "#!/usr/bin/env python" instead of "#!/usr/bin/python" - I'm not suggesting that this *is* a bug with blueman; I'm more interested in thoughts and how to address it with the reporter: http://www.linuxquestions.org/questions/slackware-14/current-blueman-using-env-sourced-python-4175564239/

infirit, 2016-01-18:
I am not sure we want to support switching version on the fly for blueman as we build a c module. Switching within the same major would probably be ok but I am pretty sure things go badly when switching major version.

infirit, 2016-01-18:
There is no abi guarantee for python2 and a limited one for python3 so I would be very careful with switching version.

What most distro's do it rewrite the shbang so it works for them. Gentoo ppl of-course take this to the extreme and basically solved this problem already through rewriting shbang [1], python-exec wrapper and building for any version installed or needed/wanted.

[1] https://gitweb.gentoo.org/repo/gentoo.git/tree/eclass/python-utils-r1.eclass#n1059

zakame, 2016-01-19:
I'm the one who reported my problem on the Slackware forum; let me post in more detail here.

I noticed this first when Slackware-current updated blueman to the latest current stable (2.0.3) where in my Fluxbox startup I had `/usr/bin/blueman-applet &`, which failed.  Upon running on an xterm, I saw

```
θ61° [zakame@drift:~] % /usr/bin/blueman-applet
Traceback (most recent call last):
  File "/usr/bin/blueman-applet", line 12, in <module>
    import dbus
ImportError: No module named 'dbus'
```

So naturally I checked the script, and saw it had a `#!/usr/bin/env python` shebang.  Running with system python set explicitly seems to work:

```
θ80° [zakame@drift:~] % /usr/bin/python /usr/bin/blueman-applet
/usr/bin/blueman-applet:20: PyGIWarning: Notify was imported without specifying a version first. Use gi.require_version
('Notify', '0.7') before import to ensure that the right version gets loaded.                                         
  from gi.repository import Notify
blueman-applet version 2.0.3 starting
_________
Load (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:60)
['NetUsage', 'DBusService', 'Networking', 'RecentConns', 'DiscvManager', 'Headset', 'ExitItem', 'GameControllerWakelock
', 'PowerManager', 'PPPSupport', 'StatusIcon', 'NMDUNSupport', 'Menu', 'ShowConnected', 'AuthAgent', 'StandardItems', '
SerialManager', 'TransferService', 'NMPANSupport', 'KillSwitch', 'DhcpClient']                                        
_________
get_interface_version (/usr/lib64/python2.7/site-packages/blueman/bluez/BlueZInterface.py:13)
Detected BlueZ 5 
/usr/lib64/python2.7/site-packages/blueman/plugins/applet/GameControllerWakelock.py:11: PyGIWarning: GdkX11 was importe
d without specifying a version first. Use gi.require_version('GdkX11', '3.0') before import to ensure that the right ve
rsion gets loaded.                                                                                                    
  from gi.repository import Gdk, GdkX11
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Headset.Headset'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Networking.Networking'> 
_________
load_nap_settings (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/Networking.py:36)
Loading NAP settings 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.SerialManager.SerialManager'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ExitItem.ExitItem'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'> 
_________
__load_plugin (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:133)
[... 131 lines cut ...]
_________
enumerate_connections (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/ShowConnected.py:50)
Found 0 existing connections 
_________
initialize (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/RecentConns.py:144)
rebuilding menu 
_________
UpdatePowerState (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/PowerManager.py:159)
off False 
foff False 
on True 
current state True 
new state True 
```

However, upon trying to click on the Bluetooth systray icon (or equivalently, right-clicking it to get the context menu and click "Devices") to open the blueman-manager fails:

```
Traceback (most recent call last):
  File "/usr/bin/blueman-manager", line 11, in <module>
    import dbus
ImportError: No module named 'dbus'
```

Backtracing this leads to `blueman.Functions#launch`, which in turn uses `Gio.AppInfo#launch`.  I could no longer trace beyond that, but I suspect it uses an `os.execv()` to actually launch the scripts, which might explain the above (e.g. it launches the script with my pyenv-enabled user environment.)

zakame, 2016-01-19:
My workaround for now so far is to wrap my `blueman-applet` call in a script that explicitly turns off pyenv:

```
#!/bin/sh
pyenv shell system
/usr/bin/blueman-applet
```

cschramm, 2016-01-19:
As @infirit describes for Gentoo, the shebangs also get replaced in deb packages by debhelper. After switching to Python 3 it now sets `#!/usr/bin/python3`. I thought this is common practice between all packaging systems.

We introduced `/usr/bin/env` as it seemed like the better default, see #187.

infirit, 2016-02-10:
I think we can summarize this thread with:

 * ``/usr/bin/env python`` is more portable than hardcoding the paths.
 * Distro's typically modify the shebang to suit their needs.
 * We do not support switching python version on the fly as we require binary modules

If I am wrong please do correct me otherwise I suggest we close this :smile:.

closed 2016-02-10 by cschramm

rworkman, 2016-02-10:
Yep, I think that's a decent summary - I'll recommend appropriate changes to our build. Thanks!