blueman-project/blueman #436: blueman-sendto:  error handler fails

infirit, 2016-01-19:
I thought it was something I broke with the device branch, it wasn't and it happens in master. Documenting to not forget about it/

Not sure what is going on, I start a transfer from the command line and then select a file and after a bit of time the transfer failed and the error_handler is called, which fails with the below.

```
_on_properties_changed (/usr/lib64/python3.4/site-packages/blueman/bluez/obex/Transfer.py:26)
/org/bluez/obex/client/session4/transfer4 Status error 
TypeError: on_transfer_error() missing 1 required positional argument: 'msg'
```

cschramm, 2016-01-20:
I guess removing one of the of the `None` arguments in `Sender.on_transfer_failed` fixes this.

infirit, 2016-01-20:
The thing is that other errors do show up properly, for example when my phone is turned off. It may be the phone doing something we do not expect or wrong. I'll go back to 2.0 and try to confirm this is a recent problem.

infirit, 2016-01-20:
Ok, traced it to the removal of BlueZ4 in c57bd1ecf6f340698066e40ced5894910c6d469e. From then on I get no error, before it tells me something went wrong. Did not investigate further so tbc.

infirit, 2016-01-20:
More info, we never get to ``on_transfer_failed`` in blueman-sendto. I see that we emit the ``error`` signal on the Transfer object which is connected to Sender.on_transfer_error. When I look at the bluez4 removal commit I see we used to add None to the signal but we don't any-more and fail.

I'll push the below which adds a default to the msg keyword in Sender.on_transfer_error.

```diff
diff --git a/apps/blueman-sendto b/apps/blueman-sendto
index dcb8e08..b93f751 100755
--- a/apps/blueman-sendto
+++ b/apps/blueman-sendto
@@ -223,7 +223,7 @@ class Sender(Gtk.Dialog):
         if self.object_push:
             self.object_push.send_file(file_path)
 
-    def on_transfer_error(self, _transfer, msg):
+    def on_transfer_error(self, _transfer, msg=None):
         if not self.error_dialog:
             self.speed.reset()
             d = Gtk.MessageDialog(self, type=Gtk.MessageType.ERROR)
```

commit bbd6eb15 (2016-01-20): blueman-sendto: The error signal does not provide a msg

Before we added None to the error signal in the Transfer object. This was
removed when we dropped Bluez4 support.

fixes #436
closed 2016-01-20 by infirit