blueman-project/blueman #441: Still connected signals

infirit, 2016-01-24:
With the removal of ``SignalTracker`` we have places were we keep objected connected and preventing them from being garbage collected and most likely this is also why ``__del__`` is not called. An example is the TransferService plugin which does not unregister the agent on unload properly. Which leads to the following on load. I'll send a PR for this later this week but ..

There are probably plenty of other places where we run into problems so I like to keep this issue open until we have hunted down every single place we do not disconnect.

```python
_________
Unload (/usr/lib64/python3.4/site-packages/blueman/main/PluginManager.py:183)
Unloading TransferService 
_________
__load_plugin (/usr/lib64/python3.4/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.TransferService.TransferService'> 
_________
_on_obex_owner_changed (/usr/lib64/python3.4/site-packages/blueman/plugins/applet/TransferService.py:164)
obex owner changed: :1.40 
Traceback (most recent call last):
  File "/usr/lib64/python3.4/site-packages/dbus/connection.py", line 604, in msg_reply_handler
    reply_handler(*message.get_args_list(**get_args_opts))
  File "/usr/lib64/python3.4/site-packages/blueman/plugins/applet/TransferService.py", line 169, in _on_obex_owner_changed
    self._agent = _Agent(self._applet)
  File "/usr/lib64/python3.4/site-packages/blueman/plugins/applet/TransferService.py", line 29, in __init__
    self._agent = obex.Agent(self._agent_path)
  File "/usr/lib64/python3.4/site-packages/blueman/bluez/obex/Agent.py", line 31, in __init__
    dbus.service.Object.__init__(self, dbus.SessionBus(), agent_path)
  File "/usr/lib64/python3.4/site-packages/dbus/service.py", line 485, in __init__
    self.add_to_connection(conn, object_path)
  File "/usr/lib64/python3.4/site-packages/dbus/service.py", line 576, in add_to_connection
    self._fallback)
KeyError: "Can't register the object-path handler for '/org/blueman/obex_agent': there is already a handler"
```

infirit, 2016-01-25:
Updated the title, I created https://github.com/infirit/blueman/commit/f2ac6bd9bff2b9ac9d7d53040210a965be1fa577 to track the signals and disconnect. Also added an explicit register/unregister to _Agent and added in various places. I now see ``on_registered`` and ``on_unregistered`` from AgentManager in the logs but it looks like it fails to unregister. Then when we register again we error ofcourse. AFAICS we call the UnRegisterAgent properly in AgentManager so I am confused why it doesn't...

cschramm, 2016-01-25:
The error does not have something to do with BlueZ' RegisterAgent / UnregisterAgent, but refers to registering the dbus object. I'm not sure if dbus-python should unregister it automatically e.g. when the object gets deallocated, but there is an explicit `remove_from_connection` connection we could use to clean up.

infirit, 2016-01-27:
Hmm, missed that :smile:, I'll see if using ``remove_from_connection`` works.

infirit, 2016-01-27:
Does not seem to work here. We could just add ``__unloadable__ = False`` :wink:

cschramm, 2016-01-27:
I drafted a patch for another idea to avoid the issue: https://github.com/blueman-project/blueman/commit/419e834e96ca6e29431dfb35fed4178148892a04

Not tested, but should help.

infirit, 2016-01-27:
Same problem unfortunately :cry:

cschramm, 2016-01-27:
Wut?! Cannot be the same traceback, can it?

infirit, 2016-01-27:
Here is the log from enabling -> disabling -> enabling, traceback looks identical.

```python
_________
UpdatePowerState (/usr/lib64/python3.4/site-packages/blueman/plugins/applet/PowerManager.py:160)
off False 
foff False 
on True 
current state True 
new state True 
_________
check (/usr/lib64/python3.4/site-packages/blueman/plugins/applet/PowerManager.py:120)
callbacks done 
_________
set_adapter_state (/usr/lib64/python3.4/site-packages/blueman/plugins/applet/PowerManager.py:91)
True 
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
_________
__load_plugin (/usr/lib64/python3.4/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.TransferService.TransferService'> 
_________
_on_obex_owner_changed (/usr/lib64/python3.4/site-packages/blueman/plugins/applet/TransferService.py:170)
obex owner changed: :1.40 
_________
on_registered (/usr/lib64/python3.4/site-packages/blueman/bluez/obex/AgentManager.py:15)
/org/blueman/obex_agent 
_________
Unload (/usr/lib64/python3.4/site-packages/blueman/main/PluginManager.py:183)
Unloading TransferService 
_________
on_unregistered (/usr/lib64/python3.4/site-packages/blueman/bluez/obex/AgentManager.py:24)
/org/blueman/obex_agent 
_________
__load_plugin (/usr/lib64/python3.4/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.TransferService.TransferService'> 
_________
_on_obex_owner_changed (/usr/lib64/python3.4/site-packages/blueman/plugins/applet/TransferService.py:170)
obex owner changed: :1.40 
Traceback (most recent call last):
  File "/usr/lib64/python3.4/site-packages/dbus/connection.py", line 604, in msg_reply_handler
    reply_handler(*message.get_args_list(**get_args_opts))
  File "/usr/lib64/python3.4/site-packages/blueman/plugins/applet/TransferService.py", line 178, in _on_obex_owner_changed
    self._agent = _Agent(self._applet)
  File "/usr/lib64/python3.4/site-packages/blueman/plugins/applet/TransferService.py", line 29, in __init__
    self._agent = obex.Agent(self._agent_path)
  File "/usr/lib64/python3.4/site-packages/blueman/bluez/obex/Agent.py", line 31, in __init__
    dbus.service.Object.__init__(self, dbus.SessionBus(), agent_path)
  File "/usr/lib64/python3.4/site-packages/dbus/service.py", line 485, in __init__
    self.add_to_connection(conn, object_path)
  File "/usr/lib64/python3.4/site-packages/dbus/service.py", line 576, in add_to_connection
    self._fallback)
KeyError: "Can't register the object-path handler for '/org/blueman/obex_agent': there is already a handler"
```

cschramm, 2016-01-28:
OK, I missed that the plugin system creates new instances for some reason, so TransferService._agent needs to be a class variable.

https://github.com/blueman-project/blueman/commit/abfbd113c766ea2ffe44dce7f42376b56bce50c7 works for me. :smiley:

infirit, 2016-01-29:
Yeah, all ok now :smile:

commit abfbd113 (2016-01-28): [TransferService] Keep agent dbus object alive

...just register and unregister it with BlueZ. Should fix #441
commit 5ced737c (2016-01-29): [TransferService] Keep agent dbus object alive

...just register and unregister it with BlueZ. Fixes #441
closed 2016-01-29 by cschramm