blueman-project/blueman #443: blueman error

nimmuk09, 2016-01-29:
Hi,

I have an erro when starting blueman

> Resource Not Ready
> Traceback (most recent call last):
>   File "/usr/lib/python3.5/site-packages/blueman/bluez/errors.py", line 146, in warp
>     return func(*args, **kwargs)
>   File "/usr/lib/python3.5/site-packages/blueman/bluez/Adapter.py", line 134, in start_discovery
>     self.get_interface().StartDiscovery()
>   File "/usr/lib/python3.5/site-packages/dbus/proxies.py", line 145, in __call__
>     **keywords)
>   File "/usr/lib/python3.5/site-packages/dbus/connection.py", line 651, in call_blocking
>     message, timeout)
> dbus.exceptions.DBusException: org.bluez.Error.NotReady: Resource Not Ready
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/usr/bin/blueman-manager", line 210, in inquiry
>     self.List.DiscoverDevices()
>   File "/usr/lib/python3.5/site-packages/blueman/gui/DeviceList.py", line 399, in DiscoverDevices
>     self.Adapter.start_discovery()
>   File "/usr/lib/python3.5/site-packages/blueman/bluez/errors.py", line 148, in warp
>     raise parse_dbus_error(exception)
> blueman.bluez.errors.DBusNotReadyError: Resource No

bluetooth is on

> [stoop@TiFS ~]$ bluetooth
> bluetooth = on

[stoop@TiFS ~]$ sudo systemctl status bluetooth

> [sudo] password for stoop: 
> ● bluetooth.service - Bluetooth service
>    Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; vendor preset: disabled)
>    Active: active (running) since Fri 2016-01-29 08:00:04 PHT; 35min ago
>      Docs: man:bluetoothd(8)
>  Main PID: 754 (bluetoothd)
>    Status: "Running"
>     Tasks: 1 (limit: 512)
>    CGroup: /system.slice/bluetooth.service
>            └─754 /usr/lib/bluetooth/bluetoothd
> 
> Jan 29 08:04:04 TiFS bluetoothd[754]: Failed to set mode: Failed (0x03)
> Jan 29 08:09:08 TiFS bluetoothd[754]: Failed to set mode: Failed (0x03)
> Jan 29 08:16:03 TiFS bluetoothd[754]: Failed to set mode: Failed (0x03)
> Jan 29 08:18:05 TiFS bluetoothd[754]: Failed to set mode: Failed (0x03)
> Jan 29 08:18:21 TiFS bluetoothd[754]: Failed to set mode: Failed (0x03)
> Jan 29 08:20:08 TiFS bluetoothd[754]: Failed to set mode: Failed (0x03)
> Jan 29 08:21:09 TiFS bluetoothd[754]: Failed to set mode: Failed (0x03)
> Jan 29 08:21:20 TiFS bluetoothd[754]: Failed to set mode: Failed (0x03)
> Jan 29 08:22:31 TiFS bluetoothd[754]: Failed to set mode: Failed (0x03)
> Jan 29 08:32:50 TiFS bluetoothd[754]: Failed to set mode: Failed (0x03)
> 

i have tried purging and reinstalling everything related to bluetooth (as suggested in simmilar issue posted years ago), still no luck.   I think this happen after i installed bluediving.

cschramm, 2016-01-29:
The BlueZ error means that the adapter is not powered. blueman should handle that, however I expect it to be an old version of blueman.

https://github.com/blueman-project/blueman/blob/master/CONTRIBUTING.md

infirit, 2016-01-29:
@cschramm would the same happen in "Airplane mode"? If so that may be a thing to check @nimmuk09

cschramm, 2016-01-30:
I don't know of any airplane mode handling in BlueZ. It probably either powers the adapters off or makes them unavailable at all, so yes, it may be the reason that the adapter is not powered, but still blueman should catch that.

nimmuk09, 2016-01-31:
hi @cschramm, here are my blueman info. 

> blueman: 2.0.2
> BlueZ: 5.37
> Distribution: 
> Desktop: X-Cinnamon

@infirit, thanks. airplane mode is off.

cschramm, 2016-01-31:
First of all, 2.0.2 has a severe privilege escalation issue and you should upgrade.

I'm not sure how blueman gets to the point where the exception is raised. Please provide the full output from blueman-applet (https://github.com/blueman-project/blueman/wiki/Troubleshooting).

nimmuk09, 2016-01-31:
@cschramm here's the [out put](http://pastebin.com/sKEhqWuW) after running blueman-applet

blueman will run with a dialog box saying that "bluetooth is disabled".  when i clicked the "enable" button.  this [error](http://pastebin.com/THXXGNM8) apears in a dialog box. i will update to the latest version using tar ball. aur repo wasn't update.

nimmuk09, 2016-01-31:
hi @cschramm i reinstalled blueman using fresh the updated zip file.  [here's ](http://pastebin.com/mr3cfD3j) my error. i tried using sudo, same output.

cschramm, 2016-02-01:
The blueman package is up to date: https://www.archlinux.org/packages/?q=blueman

The log looks like blueman fails to activate bluetooth due to a hard kill switch and it seems like our

cschramm, 2016-04-18:
The glitch in `check_bluetooth_status` is fixed in #498.

closed 2016-04-18 by cschramm