blueman-project/blueman #444: Trying to register NAP on second adapter added

infirit, 2016-01-29:
Documenting so we don't forget (and my head is not willing to work well due to a cold :mask:).

Looks like we are trying to enable when it already was with the fist adapter.

```python
Run (/usr/lib64/python3.4/site-packages/blueman/main/PluginManager.py:208)
Function on_adapter_added on Networking Failed 
Traceback (most recent call last):
  File "/usr/lib64/python3.4/site-packages/blueman/bluez/errors.py", line 146, in warp
    return func(*args, **kwargs)
  File "/usr/lib64/python3.4/site-packages/blueman/bluez/NetworkServer.py", line 17, in register
    self._interface.Register(uuid, bridge)
  File "/usr/lib64/python3.4/site-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib64/python3.4/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib64/python3.4/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.bluez.Error.AlreadyExists: Already Exists

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib64/python3.4/site-packages/blueman/main/PluginManager.py", line 212, in Run
    ret = getattr(inst, function)(*args, **kwargs)
  File "/usr/lib64/python3.4/site-packages/blueman/plugins/applet/Networking.py", line 54, in on_adapter_added
    self.update_status()
  File "/usr/lib64/python3.4/site-packages/blueman/plugins/applet/Networking.py", line 57, in update_status
    self.set_nap(self.Config["nap-enable"])
  File "/usr/lib64/python3.4/site-packages/blueman/plugins/applet/Networking.py", line 70, in set_nap
    s.register("nap", "pan1")
  File "/usr/lib64/python3.4/site-packages/blueman/bluez/errors.py", line 148, in warp
    raise parse_dbus_error(exception)
blueman.bluez.errors.DBusAlreadyExistsError: Already Exists
```

infirit, 2016-01-29:
The problem is that we loop over all adapters and try to register without checking if this was already done. Below should work but feel free to suggest a better solutions :smile:.

```diff
diff --git a/blueman/plugins/applet/Networking.py b/blueman/plugins/applet/Networking.py
index c18d2f7..31a0295 100644
--- a/blueman/plugins/applet/Networking.py
+++ b/blueman/plugins/applet/Networking.py
@@ -20,6 +20,8 @@ class Networking(AppletPlugin):
     _signal = None
 
     def on_load(self, applet):
+        self._registered = {}
+
         self.Applet = applet
 
         self.Config = Config("org.blueman.network")
@@ -65,8 +67,16 @@ class Networking(AppletPlugin):
         if self.Applet.Manager != None:
             adapters = self.Applet.Manager.list_adapters()
             for adapter in adapters:
-                s = NetworkServer(adapter.get_object_path())
+                object_path = adapter.get_object_path()
+
+                registered = self._registered.setdefault(object_path, False)
+                if registered:
+                    continue
+
+                s = NetworkServer(get_object_path)
                 if on:
                     s.register("nap", "pan1")
+                    self._registered[object_path] = True
                 else:
                     s.unregister("nap")
+                    self._registerd[object_path] = False
```

cschramm, 2016-01-30:
The patch looks like it's not about multiple adapters, but about registering NAP on the same adapter multiple times, right?

infirit, 2016-01-30:
Yes, the problem only shows up when we insert a second adapter.
-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

infirit, 2016-01-30:
Ugh, my patch had a pretty big fault causing it to never unregister. The, imo, correct patch is now in #445.

closed 2016-01-31 by infirit