blueman-project/blueman #446: Doesnt connect A2DP withe Bluetooth Headset on Ubuntu Gnome 15.10

raulgrangeiro, 2016-01-31:
I can not connect my Bluetooth Headset with A2DP connection on Ubuntu Gnome 15.10 x64.

infirit, 2016-01-31:
15.10 now comes with bluez5 I believe and things changed a bit. If I remember correctly there is now a generic connect which includes audio. To help with this do have a look at [1] and provide the logs when trying to connect.

[1] https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging

cschramm, 2016-01-31:
Do you have pulseaudio-module-bluetooth to pick up the device?

@infirit: Stop criticizing people for using blueman (under gnome)! :laughing:

infirit, 2016-01-31:
>Stop criticizing people for using blueman (under gnome)

It is not that I want them to stop using blueman but rather surprised people are and I wonder why :smile:

raulgrangeiro, 2016-01-31:
Hi guys @infirit  @cschramm 

> 5.10 now comes with bluez5 I believe and things changed a bit. If I remember correctly there is now a generic connect which includes audio. To help with this do have a look at [1] and provide the logs when trying to connect.
> 
> edit: Why do you use blueman under gnome instead of what comes with gnome?
> 
> [1] https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging
> 

I have did what you asked me for and this is the log of blueman-applet and the log is attached this message. I use blueman because some people told me to use it for trying to fix this bug that happens independent of blueman. This information is also useful:

blueman: 2.0
BlueZ: 5.35
Distribution: Ubuntu Gnome 15.10 x64
Desktop: GNOME

> Do you have pulseaudio-module-bluetooth to pick up the device?

Yes, I have this installed.

Could you help me to solve this problem?? I really need this bluetooth headset working with A2DP.

I am waiting for answer. God bless you!

[log.txt](https://github.com/blueman-project/blueman/files/111479/log.txt)

cschramm, 2016-02-01:
So you cannot discover the device? I guess it's just not (in) discoverable (mode). Did you get the device working with any other system or application?

cschramm, 2016-02-01:
Btw: Your version of blueman has a grave security issue. Unfortunately Ubuntu Security folks do not seem to read security announcements for the software they take from Debian, so only the xenial version is fine... :unamused: I just reported it to them explicitly.

raulgrangeiro, 2016-02-01:
> So you cannot discover the device? I guess it's just not (in) discoverable (mode). Did you get the device working with any other system or application?

@cschramm  No, I can connect, and it was connected in that time, but I can't use A2DP just "hands free" mode ( I forgot the name of this kind of connection).

> Btw: Your version of blueman has a grave security issue. Unfortunately Ubuntu Security folks do not seem to read security announcements for the software they take from Debian, so only the xenial version is fine... :unamused: I just reported it to them explicitly.

What kind of issue has this version?? What can I do for fixing the issue of connection with my bluetooth headset?

infirit, 2016-02-01:
>trying to fix this bug that happens independent of blueman.

What bug?

>I can connect, and it was connected in that time

I asked for the log when connecting so we can see if there are errors. Can you disconnect and provide the log when connecting. Also of use may be the log from blueman-manager.

closed 2016-04-18 by cschramm