blueman-project/blueman #447: blueman-applet fails to display icon under Wayland

kimdouglasmason, 2016-01-31:
### Environment
blueman: 2.0  # The installed package is 2.1-0.3.git7a2e20e.fc23
BlueZ: 5.36
Distribution: Fedora 23 Workstation x86-64
Desktop environment: Gnome 3.18.2

### Description
When logging into a Wayland session, the blueman icon does not display. (Note: When logging into a Gnome X session, blueman works perfectly.) After performing a 'killall blueman-applet', and then running 'blueman-applet', the following output appears (see gist below).

https://gist.github.com/kimdouglasmason/e3bcabdb0e39881389f2

The following output in particular looks suspicious:

    __load_plugin (/usr/lib/python3.4/site-packages/blueman/main/PluginManager.py:133)
    loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'> 
    Traceback (most recent call last):
      File "/usr/lib/python3.4/site-packages/blueman/plugins/BasePlugin.py", line 54, in _load
        self.on_load(parent)
      File "/usr/lib/python3.4/site-packages/blueman/plugins/applet/GameControllerWakelock.py", line 25, in on_load
        self.root_window_id = "0x%x" % Gdk.Screen.get_default().get_root_window().get_xid()
    AttributeError: 'gtk.gdk.WaylandWindow' object has no attribute 'get_xid'

kimdouglasmason, 2016-01-31:
I have also restarted blueman-applet under X11, and captured the output in the following gist:

https://gist.github.com/kimdouglasmason/a25f1316859192c33e46

infirit, 2016-01-31:
The problem is not Wayland but gnome 3 not having a traditional notification area where the blueman icon should be. Why use blueman under gnome3 anyway?

There is a problem with the wakelock plugin which we should look into though.
-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

cschramm, 2016-01-31:
We need to detect wayland and throw an ImportError in GameControllerWakelock as it's X11 only.

cschramm, 2016-01-31:
#448

commit 1ef2ecb8 (2016-01-31): [GameControllerWakelock] Do not load in non-X11 environments

Fixes #447
commit 0b2131ba (2016-01-31): [GameControllerWakelock] Do not load in non-X11 environments

Fixes #447
closed 2016-01-31 by cschramm