blueman-project/blueman #450: No pin from the Device being Pared

gdwinslow, 2016-02-03:
I am running: Raspberry PI 2
Bluetooth Dongle: Bus 001 Device 008: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
OS: Raspberry Jessie

Installed all three Parts: bluetooth bluez blueman
bluetoothctr ver: 5.23
bluemen applet: 1.99.alpha1
Also added the pi user to the bluetooth group

Normal Operation to pair a Keyboard:
1. Search to get address.
2. Pair address
3. the Keyboard responds with pin the user is going to type into it, displayed on the Computer System.
4. The User types the pin on the keyboard and hits enter.
5. If all is well: the keyboard is paired and the keyboard stops trying to pair.
6. The Computer system Connects to the keyboard.
7. The computer system Marks the keyboard as trusted.

The Bluetooth Manager never displayes the pin the keyboard supplys.
the bluetoothctr dos display the the pin the keyboard supplys.
Looks like the Hook the Bluetooth Manager is supposed to setup is not geting setup.  If you need more data please let me now

cschramm, 2016-02-03:
Please provide the output from blueman-applet, see https://github.com/blueman-project/blueman/wiki/Troubleshooting.

infirit, 2016-02-03:
If I remember correctly a more recent and stable version is in jessie backports. You may want to pick that one up as it has lots of fixes.

infirit, 2016-02-03:
>version is in jessie backports

[I remembered incorrectly unfortunately](https://packages.debian.org/search?suite=jessie-backports&searchon=all&keywords=blueman). Would be nice if we can point people to a non alpha version.

gdwinslow, 2016-02-03:
Can you give me and url for non-Alpha version?

gdwinslow, 2016-02-03:
Hear you go:

 

pi@raspberrypi:~ $ sudo killall blueman-applet

pi@raspberrypi:~ $ sudo blueman-applet

 

** (blueman-applet:2229): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files

Loading configuration plugins

blueman-applet version 1.99.alpha1 starting

Using GConf config backend

_________

Load (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:54)

['PowerManager', 'StandardItems', 'Headset', 'StatusIcon', 'RecentConns', 'AuthAgent', 'Networking', 'DiscvManager', 'TransferService', 'PPPSupport', 'ShowConnected', 'NMPANSupport', 'DBusService', 'AppIndicator', 'NetUsage', 'DhcpClient', 'KillSwitch', 'Menu', 'ExitItem', 'SerialManager', 'NMDUNSupport'] 

Using GConf config backend

_________

get_interface_version (/usr/lib/python2.7/dist-packages/blueman/bluez/BlueZInterface.py:8)

Detected BlueZ 5 

_________

__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)

loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 

Using GConf config backend

_________

__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)

loading <class 'blueman.plugins.applet.DBusService.DBusService'> 

Traceback (most recent call last):

  File "/usr/bin/blueman-applet", line 110, in <module>

    BluemanApplet()

  File "/usr/bin/blueman-applet", line 53, in __init__

    self.Plugins.Load()

  File "/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py", line 115, in Load

    if (cls.__autoload__ or cls.__name__ in c) and not (cls.__unloadable__ and "!" + cls.__name__ in c):

TypeError: argument of type 'NoneType' is not iterable

 

Is it posable to get non alfa version?

cschramm, 2016-02-04:
In the log it looks like the applet does not even start...

A recent version is in stretch: You can just download and install it: https://packages.debian.org/stretch/blueman#pdownload (might need to run aptitude safe-upgrade afterward to get the dependencies in if there are warnings).

gdwinslow, 2016-02-04:
Ok, took a While.  System is still intact, but it is still loading the blueman-applet version 1.99.alpha1 on the Desktop

gdwinslow, 2016-02-12:
Ok, Started over from scratch (Re-imaged SD-Card).  Ran the blueman-applet version 1.99.alpha1 in the Python Debugger.  Hear is the results:

Python 2.7.9 (default, Mar  8 2015, 00:52:26) 
[GCC 4.9.2] on linux2
Type "copyright", "credits" or "license()" for more information.
>>> ================================ RESTART ================================
>>> 
Loading configuration plugins
blueman-applet version 1.99.alpha1 starting
Using GConf config backend
Using GConf config backend
Using GConf config backend
Using GConf config backend
Using GConf config backend
Using GConf config backend
Using GConf config backend
Using GConf config backend
Using GConf config backend
Using GConf config backend
ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 230, in maybe_handle_message
    self._handler(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/blueman/bluez/PropertiesBlueZInterface.py", line 51, in wrapper
    handler(name, value, **kwargs)
  File "/usr/lib/python2.7/dist-packages/blueman/plugins/applet/SerialManager.py", line 48, in on_device_property_changed
    d = Device(path)
  File "/usr/lib/python2.7/dist-packages/blueman/main/Device.py", line 39, in __init__
    self.Properties = self.Device.get_properties()
  File "/usr/lib/python2.7/dist-packages/blueman/bluez/errors.py", line 143, in warp
    raise parse_dbus_error(exception)
DBusException: org.freedesktop.DBus.Error.UnknownObject: Method "GetAll" with signature "s" on interface "org.freedesktop.DBus.Properties" doesn't exist

after that i built on the Raspberry PI the 'bluez-5.37', but it had the same issue.

Hop this helps

cschramm, 2016-02-14:
A jessie-backports build of version 2.0.3 is coming soon.

cschramm, 2016-03-04:
Should finally be available now.

gdwinslow, 2016-03-08:
success with version: 2.0.3
Built it on the Device, with clean install of raspbion Jessie 16-02-09

Bluetooth Dongle: Bus 001 Device 005: ID 0a5c:2101 Broadcom Corp. BCM2045 Bluetooth
OS: Raspberry Jessie
Installed all three Parts: bluetooth bluez blueman
bluetoothctr ver: 5.23
bluemen applet: 2.0.3
bluetooth support: 5.23
bluetooth tools deamon: 5.23
bluetooth printer driver CUPS: 5.23
bluez fermware: 1.2
bluez OBEX deamon: 5.23
bluez tools: 5.23
btscanner: 5.23
libbluetooth3: 5.23

bluetooth-dev: 5.23
libnet bluetooth perl: 5.23
python bluez: 0.20
Added the pi user to the bluetooth group

Some of this are no doubt for Compiling the 2.0.3 blueman, and others are for QT5.6

Oww, don't now if this is designed behaver or not.  Wall pairing a HID device (Keyboard) it dos not connect the device after pairing.  I have to go to the Devices menu in the Bluetooth manager to connect.

I will continue to run some test, but this issue is closed.

Hop this help. thanks

closed 2016-03-08 by gdwinslow