blueman-project/blueman #451: Blueman Assistant: no adapter found

francesco1119, 2016-02-03:
On Debian Jessie the Bluetooth Manager seems installed correctly.
However if I press on setup new device it says: `Blueman Assistant: no adapter found`.
My bluetooth is an Atheros AR9485 and the [driver](https://packages.debian.org/jessie/kernel/firmware-atheros) is correctly installed.

What to do?

yars068, 2016-02-03:
First, please read [Troubleshooting article](https://github.com/blueman-project/blueman/wiki/Troubleshooting).

francesco1119, 2016-02-22:
blueman: 1.99.alpha1
BlueZ: Version: 5.23-2+b1
Distribution: Debian Jessie
Desktop environment: Cinnamon

I have no idea how to test my bluetooth with "bluez-test-adapater list" (BlueZ 4) or "bluetoothctl list" (BlueZ 5) as **there is no documentation out there!**

...and I'm not the only one who find it hard to debug:

http://danielhnyk.cz/bluetooth-linux-indecipherable-mystery/

infirit, 2016-02-22:
>I have no idea how to test my bluetooth with

You have to run them from a terminal emulator window, in your case it will be ``bluetoothctl list``. I updated the wiki so it hopefully is more clear now.

closed 2016-04-18 by cschramm

1nf1del, 2019-05-15:
bluetoothctl list shows Nothing

don-dev, 2020-01-06:
Seems old, but for context,... came here from System76\pop_OS upgrade where they suggest installing blueman to diagnose. Bluetooth Manager shows no adapters and list command returns 0 out