blueman-project/blueman #453: We do not handle authentication failed

infirit, 2016-02-08:
I initially thought this was due to my gdbus work (confused the hell out of me) but we have the same problem in master. I have not checked the 2-0-stable branch so we may need to fix it there as well.

Documenting so we do not forget :smile:.

```python
_________
RequestConfirmation (/usr/lib64/python3.4/site-packages/blueman/main/applet/BluezAgent.py:201)
Agent.RequestConfirmation 
_________
DBusException(dbus.String('Authentication Failed'),)
Cancel (/usr/lib64/python3.4/site-packages/blueman/main/applet/BluezAgent.py:153)
Agent.Cancel 
'org.bluez.Error.AuthenticationFailed: Authentication Failed'
['org.bluez.Error.AuthenticationFailed:', 'Authentication Failed']
Traceback (most recent call last):
  File "/usr/lib64/python3.4/site-packages/dbus/connection.py", line 607, in msg_reply_handler
    *message.get_args_list()))
  File "/usr/lib64/python3.4/site-packages/blueman/bluez/Device.py", line 26, in err
    self._call('Pair', reply_handler=ok, error_handler=err)
blueman.bluez.errors.DBusAuthenticationFailedError: Authentication Failed
```

cschramm, 2016-02-09:
In `Blueman.bond` (blueman-manager) `pair` is used without arguments (handlers) which leads to an asynchronous call (as the `bluez.Device` methods define an `error_handler` wrapper unconditionally) where a positive result is ignored and a negative result raises an exception. This is also present in 2-0-stable, yes.

First of all I think we should change the behavior of the methods to either require an `error_handler` from the caller or do a synchronous call if none is provided (so that the caller may catch an exception).

Second, `Blueman.bond` needs to handle exceptions either using an `error_handler` or catching the exception if the call is made synchronous.

As the (re-)introduction of actual synchronous calls would still allow the caller to just forget about the error handling, I think we should instead require an `error_handler` to be set.

infirit, 2016-02-09:
Had a quick look and pair was made async and then used with handlers in assistant but not in other places. I'll have a look later and come up with something.

infirit, 2016-02-09:
While looking at this I found out why we did not close the notification and fixed it in 045454d0f05d81e7953744624eb9352f7bbaef5c. I think we can just handle the exception by ignoring it in blueman-manager as BluezAgent will handle (ie close the notification window) the rejection.

Does that sound reasonable?

infirit, 2016-02-09:
Ok, here is a proof of concept patch (if ok will split in two commits). Should we handle it like this in all function and clases or just for Pair?

```diff
diff --git a/apps/blueman-manager b/apps/blueman-manager
index e285fd4..c548a36 100755
--- a/apps/blueman-manager
+++ b/apps/blueman-manager
@@ -35,6 +35,7 @@ from blueman.gui.MessageArea import MessageArea
 from blueman.main.PluginManager import PluginManager
 import blueman.plugins.manager
 from blueman.plugins.ManagerPlugin import ManagerPlugin
+from blueman.bluez.errors import DBusAuthenticationFailedError
 
 DBusGMainLoop(set_as_default=True)
 
@@ -197,7 +198,8 @@ class Blueman(Gtk.Window):
         launch(command, None, False, "blueman", _("Bluetooth Assistant"))
 
     def bond(self, device):
-        self.List.Adapter.find_device(device['Address']).pair()
+        try: self.List.Adapter.find_device(device['Address']).pair()
+        except DBusAuthenticationFailedError: pass
 
     def adapter_properties(self):
         adapter = self.List.Adapter
diff --git a/blueman/bluez/Device.py b/blueman/bluez/Device.py
index 79de480..69830bf 100644
--- a/blueman/bluez/Device.py
+++ b/blueman/bluez/Device.py
@@ -25,7 +25,10 @@ class Device(PropertiesBase):
             else:
                 raise exception
 
-        self._interface.Pair(reply_handler=ok, error_handler=err)
+        if reply_handler or error_handler:
+            self._interface.Pair(reply_handler=ok, error_handler=err)
+        else:
+            self._interface.Pair()
 
     @raise_dbus_error
     def connect(self, reply_handler=None, error_handler=None):
```

infirit, 2016-02-10:
#455

PR #455 (2016-02-12): blueman-manager: Fix authentication failed handling
This picks a commit that changes the handling of bluez methods from the gdbus migration. It is useful as it allows for more things to be pushed down into ``blueman.bluez.Base`` and simplify things.

Just like in my gdbus branch we now check if we either have an error or reply handler and make the call asynchronous.

Then we show a notification when we fail to pair in blueman-manager. Please let me know if we need to specify what error we got in the notification.

This fixes #453
closed 2016-02-12 by infirit