blueman-project/blueman #454: DeviceMenu: Services not available until paired but still available to connect to

infirit, 2016-02-08:
For example, the audio and network interfaces for a mobile phone are not exposed until we paired. We should probably raise some sort of exception or return something for the modules in ``blueman/services`` so the DeviceMenu can handle it. My suggestion would be to still show them but the user can not select them, the device exposes them in the uuids.

cschramm, 2016-10-05:
Is this about the dbus interfaces? So it's probably BlueZ 4 related and those obsolete, right?

infirit, 2016-10-06:
It is about services like pan which i can try to enable but unless the device is paired they will fail. Either we should inutiate a pair before connecting the service or grey the menu item so they can't be enabled (perhaps with tooltip to pair first).

-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

cschramm, 2016-10-06:
Ah, right, `Network1` is still there...

cschramm, 2016-10-06:
Ref #208

cschramm, 2017-11-28:
Still relevant?

infirit, 2017-11-28:
Don't think so, I'll add the to my poke list for this weekend :-)

infirit, 2017-12-02:
This is handled now, closing.

closed 2017-12-02 by infirit