blueman-project/blueman #457: Cannot create a Network Access Point

flyingpengwin, 2016-02-11:
Hello all I'm trying to get my raspberry pi 2 to act as a NAP via bluetooth but, no matter what I try I get an error. If I go through the bluetooth icon go to Local Services -> Network  and try to enable NAP I get the same error no matter what combinination of PAN / DUN support. The error is "Failed to apply network settings TypeError: tuple indices must be integers, not dbus.String" If I go through the Bluetooth Devices window, right click on device I want to connect too after I pair and trust it I get a popup window from linux saying "Trying to obtain IP on bnep0 Please wait..." then "Failed to obtain an IP address on bnep0".

I do not see a bnep0 in my ifconfig... 

If anyone could help me I would really appriciate it! 

Thanks, 
flyingpengwin

cschramm, 2016-02-11:
What version?

flyingpengwin, 2016-02-11:
I did an apt-get install bluez and blueman today so it should be the most recent version they have available... Version 1.99.alpha1
Bluez version 5.23-2+b1

cschramm, 2016-02-11:
I guess the issue is fixed in a more recent version. You can get 2.0.3 from stretch: https://packages.debian.org/stretch/blueman#pdownload

flyingpengwin, 2016-02-12:
I tried getting the version you suggested but it required me to install a bunch of other stretch repositories (so I did). Once I got everything installed blueman no longer launches or atleast, the gui doesn't popup on the screen...

cschramm, 2016-02-12:
> it required me to install a bunch of other stretch repositories

Really? I thought the dependencies should all be available in jessie as well.

> the gui doesn't popup on the screen...

Hmm... Start blueman-applet in a terminal and check the output. https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging

flyingpengwin, 2016-02-12:
After reboot, kill all blueman-applet finds no processes.
Running blueman-applet from terminal outputs:
Traceback (most recent call last):
   File "/usr/bin/blueman-applet", line 12, in <module>
      import dbus
   File "/usr/lib/python3/dist-packages/dbus/__init__.py", line 82, in module <module>
       import dbus.types as types
   File "/usr/lib/python3/dist-packages/dbus/types.py", line 6, in <module>
      from _dbus_bindings import (
ImportError: No module named '_dbus_bindings'

flyingpengwin, 2016-02-12:
There was a double _ in front of and behind "init" .py but github bolded init instead. Also, killall had a space added that I didn't notice. I don't see an option to edit the post from my phone... Sorry.  Thank you so much for your help!

flyingpengwin, 2016-02-12:
Sorry for posting so much but I wanted to clarify that the stretch dependencies that were required revolved mostly around python3_3.5 and python3.5 which are not a stable release yet.

cschramm, 2016-02-13:
Uh, I see. No idea why it depends on 3.5.

Your dbus package seems to have some issue.

I'll prepare a version for jessie-backports.

flyingpengwin, 2016-02-13:
Thank you so much, please let me know if there is anything else you need from me.

flyingpengwin, 2016-02-18:
Hey Chris, I was just wondering if there was any update on this?

infirit, 2016-02-18:
When it show up [here](https://packages.debian.org/search?suite=jessie-backports&searchon=names&keywords=blueman) you can get it from jessie backports.

cschramm, 2016-02-19:
My sponsor did not upload yet. I'll query him again.

flyingpengwin, 2016-02-24:
I'm really sorry to be a bother but it still doesn't seem to be available according to infirit's link.

cschramm, 2016-03-04:
Should finally be available now.

The-Nik, 2016-03-06:
Backports works now but the error still appears :-(

My settings: 
blueman 2.0.3
Services: NAP
DHCP Server Type: dnsmasq (clean installation, no configuration)
IP Adress: ex. 172.16.2.100
PAN Support: Blueman (dhclient)
DUN Support: blueman

After a click on "Apply", this error is shown:
Failed to apply network settings
org.freedesktop.DBus.Python.TypeError: Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python3/dist-packages/blueman/main/DbusService.py", line 36, in wrapper
    return method(*args[1:], **kwargs)
  File "/usr/lib/python3/dist-packages/blueman/plugins/mechanism/Network.py", line 58, in EnableNetwork
    nc.set_dhcp_handler(DHCPDHANDLERS[dhcp_handler])
TypeError: tuple indices must be integers, not dbus.String

Is there any introduction how to initiate a PAN with blueman?

Thanks, Niklas

cschramm, 2016-03-07:
Urgs, our security fix in 2.0.3 broke that. (@infirit)

@The-Nik: For a quick fix you just need to remove the trailing comma in /usr/lib/python3/dist-packages/blueman/plugins/mechanism/Network.py, line 14.

The-Nik, 2016-03-07:
@cschramm: Now everything works fine.

Other bluetooth-question: Do you know, how to connect to a Network Access Point (via PAN) on boot? Currently I have to connect via the bluemann ui. But I need to have a command line command to connect the PANU to the network on boot.

Any ideas? I didn't find a understandable introduction for bluez oder blueman.

cschramm, 2016-03-07:
You can use the BlueZ dbus API. Something like `dbus-send --system --dest=org.bluez <object_path> org.bluez.Network1.Connect string:<uuid>` should work.

closed 2016-03-07 by cschramm

infirit, 2016-03-09:
>Urgs, our security fix in 2.0.3 broke that. (@infirit)

:sob:

flyingpengwin, 2016-03-11:
Hello all sorry for the delay but I've been very sick lately and wasn't able to test this until now. After removing the comma I get to the point where my phone is paired and trusted by when I try to create the network access point it says "Connection Failed The name org.freedesktop.NetworkManager was not pro ided by any .service files" Raspbian (the raspberry pi's main distribution) does not use Network-manager to handle net work connections. After I installed network-manager and try to create  NAP with the phone again the error changes to "Connection Failed:Network Manager Failed to activate the connection" I've tried commenting out the lines in /etc/network/interfaces so that there isn't a conflict with network-manager but I still get that error. Any suggestions? Thanks!

cschramm, 2016-03-11:
You don't need NetworkManager, see blueman _Local Services_ dialog.

The-Nik, 2016-03-11:
I needed the NetworkManager for creating a PAN although I select "blueman" in each list at Local Services. After I installed the NetworkManager everything works fine.

Maybe try:
```
sudo killall blueman-applet
sudo killall blueman-manager
sudo services NetworkManager restart
sudo blueman-applet
```

Then you see probably with 

`sudo service NetworkManager status `

and in the console of "sudo blueman-applet" more information to solve the error.

flyingpengwin, 2016-03-12:
@cschramm @The-Nik Thank you for your help. I forgot that I needed to go into the Local Services and enable NAP, dnsmasq, PAN Support:Blueman DUN Support: Blueman

I still get the "Trying to obtain an IP address on bnep0 Please wait..." Then almost immediately by "Failed to obtain an IP address on bnep0"

GitHubMe78, 2016-04-28:
Hallo Christopher,
I've made the line 14 typo correction and it works like a charm.
But now I'd like to start the access point in command line.
I tried 
dbus-send --system --type=method_call --dest=org.bluez /org/bluez/hci0/dev_XX_XX_XX_XX_XX_XX org.bluez.Network1.Connect string:'nap'
no error message but no pan1 interface created (ifconfig).
How can I manage this?
Vielen Dank für Ihre Hilfe
Alain

cschramm, 2016-04-29:
`nap` is not a UUID. Add `--print-reply` to see the reply, otherwise you will not notice errors.

GitHubMe78, 2016-04-29:
I changed _nap_ for the Controller UUID: Generic Access Profile (_show command_ in bluetoothctl).
here is the result:
**Error org.freedesktop.DBus.Error.UnknownObject: Method "Connect" with signature "s" on interface "org.bluez.Network1" doesn't exist**

GitHubMe78, 2016-04-29:
I may not use the right params:

Device BC:47:B2:A4:50:B0
        Name: Samsung Galaxy S
        Alias: Samsung Galaxy S
        Class: 0x5a020c
        Icon: phone
        Paired: yes
        Trusted: yes
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
        UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
        UUID: NAP                       (00001116-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
        UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
        UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        Modalias: bluetooth:v001Dp1200d1436
when i use 
dbus-send --system --type=method_call --print-reply --dest=org.bluez /org/bluez/hci0/dev_BC_47_B2_A4_50_B0 org.bluez.Network1.Connect string:00001116-0000-1000-8000-00805f9b34fb
i get
method return sender=:1.7 -> dest=:1.40 reply_serial=2
   string "bnep0"
no new interface created

cschramm, 2016-04-30:
I cannot identify which parts of your text are output and which are not, but it looks like BlueZ tells you you should have bnep0 now, so you managed to send the same request as blueman does.

GitHubMe78, 2016-04-30:
i get this answer
**method return sender=:1.7 -> dest=:1.14 reply_serial=2
   string "bnep0"**

but no pan1 or bnp0 interface is created and I can't connect any device to the server.
But if i launch lxde, without any intervention, pan1 interface is created and devices can be connected to the internet through bluetooth interface.
I couldn't catch the commands sent while opening lxde in order to send them manually.
Do I need to do something else after the _dbus-send --system.._ command?

GitHubMe78, 2016-04-30:
It looks like the device is connected and disconnected immediately as i can see

GitHubMe78, 2016-04-30:
Christopher,
this _dbus-send --system --type=method_call --print-reply --dest=org.bluez /org/bluez/hci0/dev_BC_47_B2_A4_50_B0 org.bluez.Network1.Connect string:00001116-0000-1000-8000-00805f9b34fb_ works in the wrong way: I want to share a BT connection **FROM** the server **TO** any device.

cschramm, 2016-04-30:
Well, you picked up the wrong question / solution then. :smile: 
@The-Nik asked for a way to connect to the NAP service of a remote device automatically, i.e. without using the blueman GUI. That's done with that dbus message. blueman does not do anything else under the hood.

If your question is how to set up a local NAP service automatically, the answer is `blueman-applet`. If you set it up once, the Networking plugin will re-apply the settings automatically on load.

If your question is how to set it up without blueman, it's completely out of scope and gets quite complicated. You can see what blueman does in blueman.main.NetConf.apply_settings. It's basically: Create a bridge. Enable IPv4 forwarding. Configure the interface. Set up NAT in iptables. Start a DHCP server if set.

GitHubMe78, 2016-04-30:
Sorry for that.
So what I'd like to do is:
- Set the parameters using the blueman GUI (it works fine);
- Apply this settings without using the blueman GUI;
- Potentially edit the parameters (where are they stored?).

cschramm, 2016-05-01:
> Apply this settings without using the blueman GUI;

As I said, `blueman-applet`. No need to do anything else than starting it.

> Potentially edit the parameters (where are they stored?).

It's stored in binary (python pickle) format in `/var/lib/blueman/network.state`.

You can use Python's NetConf class somehow like this:

````
from blueman.main.NetConf import NetConf
nc = NetConf.get_default()
nc.apply_settings()
nc.ip4_address = 'foobar'
nc.apply_settings()
nc.store()
````

GitHubMe78, 2016-05-01:
Ok thanks.
when i launch 
`root@booth:/usr/bin# ./blueman-applet
`
i get
```
Unable to init server: Could not connect: Connection refused
Unable to init server: Could not connect: Connection refused

(blueman-applet:1109): Gtk-CRITICAL **: gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(blueman-applet:1109): Gtk-CRITICAL **: gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed
Traceback (most recent call last):
  File "./blueman-applet", line 120, in <module>
    BluemanApplet()
  File "./blueman-applet", line 51, in __init__
    setup_icon_path()
  File "/usr/lib/python3/dist-packages/blueman/Functions.py", line 175, in setup_icon_path
    ic.prepend_search_path(ICON_PATH)
AttributeError: 'NoneType' object has no attribute 'prepend_search_path'
root@booth:/usr/bin#
```
although the pan is automaticly set at GUI start

infirit, 2016-05-01:
Sounds like you are suffering from https://github.com/blueman-project/blueman/issues/504#issuecomment-214265872

GitHubMe78, 2016-05-02:
Well, maybe it's the headless use of `blueman-applet`.
I tried in vain to launch `main.NetConf.apply_settings()` this way:
```
#!/usr/bin/env python
# -*- coding: utf-8 -*-
#
#  Napstart.py
#  
#

import sys

if not ('/usr/lib/python3/dist-packages/' in sys.path):
	sys.path.append('/usr/lib/python3/dist-packages/') 

import blueman
from blueman.main.NetConf import *

apply_settings()
```
I get
```
root@booth:~# ./Napstart.py
Unable to init server: Could not connect: Connection refused
Unable to init server: Could not connect: Connection refused
Traceback (most recent call last):
  File "./Napstart.py", line 14, in <module>
    from blueman.main.NetConf import *
  File "/usr/lib/python3/dist-packages/blueman/main/NetConf.py", line 15, in <module>
    from _blueman import create_bridge, destroy_bridge, BridgeException
ImportError: No module named _blueman
root@Photobooth:~# ./Napstart.py
Unable to init server: Could not connect: Connection refused
Unable to init server: Could not connect: Connection refused
Traceback (most recent call last):
  File "./Napstart.py", line 15, in <module>
    from blueman.main.NetConf import *
  File "/usr/lib/python3/dist-packages/blueman/main/NetConf.py", line 15, in <module>
    from _blueman import create_bridge, destroy_bridge, BridgeException
ImportError: No module named _blueman
root@booth:~#

```

cschramm, 2016-05-02:
`if not ('/usr/lib/python3/dist-packages/' in sys.path):`

Something broken in your python setup?

GitHubMe78, 2016-05-02:
I don't think so but I had to use the jessie-backports  repo to install blueman.
So the sys.path is
`'/root', '/usr/lib/python2.7', '/usr/lib/python2.7/plat-arm-linux-gnueabihf', '/usr/lib/python2.7/lib-tk', '/usr/lib/python2.7/lib-old', '/usr/lib/python2.7/lib-dynload', '/usr/local/lib/python2.7/dist-packages', '/usr/lib/python2.7/dist-packages', '/usr/lib/python2.7/dist-packages/gtk-2.0', '/usr/lib/pymodules/python2.7'
`
But anyway I've done a remove/install of python3. No change

cschramm, 2016-05-02:
The jessie-backports package uses python3. You need to use that as well to access modules of the package.

GitHubMe78, 2016-05-02:
ok.
now sys.path is
`'/root', '/usr/lib/python2.7', '/usr/lib/python2.7/plat-arm-linux-gnueabihf', '/usr/lib/python2.7/lib-tk', '/usr/lib/python2.7/lib-old', '/usr/lib/python2.7/lib-dynload', '/usr/local/lib/python2.7/dist-packages', '/usr/lib/python2.7/dist-packages', '/usr/lib/python2.7/dist-packages/gtk-2.0', '/usr/lib/pymodules/python2.7', '/usr/lib/python3/', '/usr/lib/python3/dist-packages/'`
result is the same:
```
Unable to init server: Could not connect: Connection refused Unable to init server: 
Could not connect: Connection refused
Traceback (most recent call last):
  File "./Napstart.py", line 16, in <module>
    import blueman.main.NetConf
  File "/usr/lib/python3/dist-packages/blueman/main/NetConf.py", line 15, in <module>
    from _blueman import create_bridge, destroy_bridge, BridgeException
ImportError: No module named _blueman
```
what is this module `_blueman` ?

cschramm, 2016-05-02:
You're still using python(2). `_blueman` is a native module typically installed in `/usr/lib/python3/dist-packages` which you will never be able to load with Python 2 as it's compiled for 3.4.

GitHubMe78, 2016-05-02:
OK
So with this the code:
```
#!/usr/bin/env python3
# -*- coding: utf-8 -*-
#
#  Napstart.py
#  
#

import sys

if not ('/usr/lib/python3/,' in sys.path):
	sys.path.append('/usr/lib/python3/') 
if not ('/usr/lib/python3/dist-packages/' in sys.path):
	sys.path.append('/usr/lib/python3/dist-packages/') 

from blueman.main.NetConf import NetConf

nc = NetConf.get_default()
nc.apply_settings()
#nc.ip4_address = 'foobar'
#nc.apply_settings()
nc.store()
```
.I get
```
root@booth:~# ./Napstart.py
Unable to init server: Could not connect: Connection refused
Unable to init server: Could not connect: Connection refused
iptables: No chain/target/match by that name.
iptables: Bad rule (does a matching rule exist in that chain?).
iptables: Bad rule (does a matching rule exist in that chain?).
iptables: Bad rule (does a matching rule exist in that chain?).
_________
add_ipt_rule (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:236)
/sbin/iptables -t nat -A POSTROUTING -s 192.168.42.1/255.255.255.0 -j MASQUERADE
Return code 0
_________
add_ipt_rule (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:236)
/sbin/iptables -t filter -A FORWARD -i pan1 -j ACCEPT
Return code 0
_________
add_ipt_rule (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:236)
/sbin/iptables -t filter -A FORWARD -o pan1 -j ACCEPT
Return code 0
_________
add_ipt_rule (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:236)
/sbin/iptables -t filter -A FORWARD -i pan1 -j ACCEPT
Return code 0
_________
do_apply (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:32)
['/usr/sbin/dnsmasq', '--pid-file=/var/run/dnsmasq.pan1.pid', '--bind-interfaces', '--dhcp-range=192.168.42.2,192.168.42.254,60m', '--except-interface=lo', '--interface=pan1', '--dhcp-option=option:router,192.168.42.1']
_________
do_apply (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:32)
Dnsmasq started correctly
_________
do_apply (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:32)
pid 1170

root@booth:~#
```
This works ...
except for the bridging!
with
```
nc = NetConf.get_default()
nc.apply_settings()
nc.ip4_address = 'foobar'
nc.apply_settings()
nc.store()

```
result is
```

root@booth:~# ./Napstart.py
Unable to init server: Could not connect: Connection refused
Unable to init server: Could not connect: Connection refused
iptables: No chain/target/match by that name.
iptables: Bad rule (does a matching rule exist in that chain?).
iptables: Bad rule (does a matching rule exist in that chain?).
iptables: Bad rule (does a matching rule exist in that chain?).
_________
add_ipt_rule (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:236)
/sbin/iptables -t nat -A POSTROUTING -s 192.168.42.1/255.255.255.0 -j MASQUERADE
Return code 0
_________
add_ipt_rule (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:236)
/sbin/iptables -t filter -A FORWARD -i pan1 -j ACCEPT
Return code 0
_________
add_ipt_rule (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:236)
/sbin/iptables -t filter -A FORWARD -o pan1 -j ACCEPT
Return code 0
_________
add_ipt_rule (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:236)
/sbin/iptables -t filter -A FORWARD -i pan1 -j ACCEPT
Return code 0
_________
do_apply (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:32)
['/usr/sbin/dnsmasq', '--pid-file=/var/run/dnsmasq.pan1.pid', '--bind-interfaces', '--dhcp-range=192.168.42.2,192.168.42.254,60m', '--except-interface=lo', '--interface=pan1', '--dhcp-option=option:router,192.168.42.1']
_________
do_apply (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:32)
Dnsmasq started correctly
_________
do_apply (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:32)
pid 1165
Traceback (most recent call last):
  File "./Napstart.py", line 21, in <module>
    nc.apply_settings()
  File "/usr/lib/python3/dist-packages/blueman/main/NetConf.py", line 283, in apply_settings
    ip_str = inet_ntoa(self.ip4_address)
TypeError: 'str' does not support the buffer interface
root@booth:~#

```
and no bridging

cschramm, 2016-05-02:
I guess you will need to use `inet_aton` on the IP address (and `foobar` is obviously no IP address :wink:).

Regarding the bridge, you can run that part insolated with
````
from _blueman import create_bridge
create_bridge('pan1')
````

The only possible reason that it's not getting created and `apply_settings` does not crash should be that the interface already exists.

GitHubMe78, 2016-05-02:
so the code is now
```

#!/usr/bin/env python3
# -*- coding: utf-8 -*-
#
#  Napstart.py
#  
#

import sys

if not ('/usr/lib/python3/,' in sys.path):
	sys.path.append('/usr/lib/python3/') 
if not ('/usr/lib/python3/dist-packages/' in sys.path):
	sys.path.append('/usr/lib/python3/dist-packages/') 

from blueman.main.NetConf import NetConf
from _blueman import create_bridge, get_net_address,get_net_interfaces
from socket import inet_aton

nc = NetConf.get_default()
nc.apply_settings()
ip = inet_aton(get_net_address("wlan0"))
nc.ip4_address = ip
nc.apply_settings()
create_bridge("pan1")

```
and the result

```

root@booth:~# ./Napstart.py
Unable to init server: Could not connect: Connection refused
Unable to init server: Could not connect: Connection refused
iptables: No chain/target/match by that name.
iptables: Bad rule (does a matching rule exist in that chain?).
iptables: Bad rule (does a matching rule exist in that chain?).
iptables: Bad rule (does a matching rule exist in that chain?).
_________
add_ipt_rule (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:236)
/sbin/iptables -t nat -A POSTROUTING -s 192.168.42.1/255.255.255.0 -j MASQUERADE
Return code 0
_________
add_ipt_rule (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:236)
/sbin/iptables -t filter -A FORWARD -i pan1 -j ACCEPT
Return code 0
_________
add_ipt_rule (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:236)
/sbin/iptables -t filter -A FORWARD -o pan1 -j ACCEPT
Return code 0
_________
add_ipt_rule (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:236)
/sbin/iptables -t filter -A FORWARD -i pan1 -j ACCEPT
Return code 0
_________
do_apply (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:32)
['/usr/sbin/dnsmasq', '--pid-file=/var/run/dnsmasq.pan1.pid', '--bind-interfaces', '--dhcp-range=192.168.42.2,192.168.42.254,60m', '--except-interface=lo', '--interface=pan1', '--dhcp-option=option:router,192.168.42.1']
_________
do_apply (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:32)
Dnsmasq started correctly
_________
do_apply (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:32)
pid 876
Traceback (most recent call last):
  File "./Napstart.py", line 27, in <module>
    create_bridge("pan1")
  File "_blueman.pyx", line 246, in _blueman.create_bridge (_blueman.c:2570)
_blueman.BridgeException
root@booth:~# 

```

cschramm, 2016-05-03:
The exception has an errno property. You can catch and print it:
````python
try:
  create_bridge("pan1")
except Exception as e:
  print(e.errno)
````

I expect 17 which is EEXIST and the only error that's ignored in apply_settings.

GitHubMe78, 2016-05-03:
Yes you are right:

```

#!/usr/bin/env python3
# -*- coding: utf-8 -*-
#
#  Napstart.py
#  
#

import sys

if not ('/usr/lib/python3/,' in sys.path):
	sys.path.append('/usr/lib/python3/') 
if not ('/usr/lib/python3/dist-packages/' in sys.path):
	sys.path.append('/usr/lib/python3/dist-packages/') 

from blueman.main.NetConf import NetConf
from _blueman import create_bridge, get_net_address,get_net_interfaces
from socket import inet_aton

nc = NetConf.get_default()
nc.apply_settings()
ip = inet_aton(get_net_address("wlan0"))
nc.ip4_address = ip
nc.apply_settings()
try:
  create_bridge("pan1")
except Exception as e:
  print(e.errno)

```
result

```
root@booth:~# python3 Napstart.py
Unable to init server: Could not connect: Connection refused
Unable to init server: Could not connect: Connection refused
_________
[1madd_ipt_rule[0m (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:236)
/sbin/iptables -t nat -A POSTROUTING -s 192.168.4.66/255.255.255.0 -j MASQUERADE 
Return code 0
_________
[1madd_ipt_rule[0m (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:236)
/sbin/iptables -t filter -A FORWARD -i pan1 -j ACCEPT 
Return code 0
_________
[1madd_ipt_rule[0m (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:236)
/sbin/iptables -t filter -A FORWARD -o pan1 -j ACCEPT 
Return code 0
_________
[1madd_ipt_rule[0m (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:236)
/sbin/iptables -t filter -A FORWARD -i pan1 -j ACCEPT 
Return code 0
_________
[1mdo_apply[0m (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:32)
['/usr/sbin/dnsmasq', '--pid-file=/var/run/dnsmasq.pan1.pid', '--bind-interfaces', '--dhcp-range=192.168.4.67,192.168.4.254,60m', '--except-interface=lo', '--interface=pan1', '--dhcp-option=option:router,192.168.4.66'] 
_________
[1mdo_apply[0m (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:32)
Dnsmasq started correctly 
_________
[1mdo_apply[0m (/usr/lib/python3/dist-packages/blueman/main/NetConf.py:32)
pid 867 
17
root@booth:~# 

```
interfaces:
```

root@booth:~# ifconfig
eth0      Link encap:Ethernet  HWaddr b8:27:eb:91:46:09
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:136 errors:0 dropped:0 overruns:0 frame:0
          TX packets:136 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:11472 (11.2 KiB)  TX bytes:11472 (11.2 KiB)

pan1      Link encap:Ethernet  HWaddr 0e:fc:87:1b:d5:ee
          inet addr:192.168.42.1  Bcast:192.168.42.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:31 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:5277 (5.1 KiB)

wlan0     Link encap:Ethernet  HWaddr b8:27:eb:c4:13:5c
          inet addr:192.168.4.66  Bcast:192.168.4.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:296 errors:0 dropped:66 overruns:0 frame:0
          TX packets:198 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:48282 (47.1 KiB)  TX bytes:37634 (36.7 KiB)

root@booth:~#

```
But bridge isn't set.

GitHubMe78, 2016-05-03:
One weird thing is if i launch 

```
root@booth:~# tightvncserver

New 'X' desktop is Photobooth:1

Starting applications specified in /root/.vnc/xstartup
Log file is /root/.vnc/Photobooth:1.log

root@booth:~#
```
Pan works perfectly with the parameters set.
I couldn't see what script does the job!

HockeyKS, 2016-11-03:
Hi,
I am trying to build a NAP between a raspberry pi 3 and a Lego Mindstorms EV3 via bluetooth. When I try to setup a Network via blueman the following error message occures:

"Failed to apply network settings TypeError: tuple indices must be integers, not dbus.String" 

I tried to update and downgrade blueman but neither works. As soon as I run "blueman-applet" the following error pops up:

pi@raspberrypi:~ $ blueman-applet

** (blueman-applet:1828): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
Loading configuration plugins
blueman-applet version 1.99.alpha1 starting
There is an instance already running

Anyone who knows how to fix the error?

Thank you very much for your help

infirit, 2016-11-03:
@HockeyKS You will need an updated version of blueman. The latest stable version 2.0.4 should have all the necessary fixes. That version is in [debian backports](https://backports.debian.org/).

HockeyKS, 2016-11-07:
Hi @infirit, thank you very much for your quick response. I tried to follow the instructions in debian backports, but as soon as I run "sudo apt-get -t jessie-backports install "package"" the following error occures:

Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package package

Do you know what could be the problem? Thanks a lot!

HockeyKS, 2016-11-07:
Hi @infirit, thank you very much for your quick response. I installed the latest stable version of blueman from debian backports.

But there occured a few new problems:

org.freedesktop.DBus.Python.TypeError: Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python3/dist-packages/blueman/main/DbusService.py", line 36, in wrapper
    return method(*args[1:], **kwargs)
  File "/usr/lib/python3/dist-packages/blueman/plugins/mechanism/Network.py", line 58, in EnableNetwork
    nc.set_dhcp_handler(DHCPDHANDLERS[dhcp_handler])
TypeError: tuple indices must be integers, not dbus.String

I would be very grateful if you could help me, thanks a lot!

infirit, 2016-11-07:
@HockeyKS, that was fixed with https://github.com/blueman-project/blueman/commit/dabdd5db0e23a2507b41d9f8f9320165048adf57 and is in blueman 2.0.4. Which version of blueman and distro are you on now?

HockeyKS, 2016-11-08:
@infirit  my distro ist jessie (8) and my blueman is 2.0.3. As soon as I try to update blueman with backports the system tells me that it is already on the newest version.

Reading package lists... Done
Building dependency tree       
Reading state information... Done
blueman is already the newest version.
The following package was automatically installed and is no longer required:
  gir1.2-gconf-2.0
Use 'apt-get autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 109 not upgraded.

infirit, 2016-11-08:
The latest version in jessie backports should have the fix so I am confused why you see it. Can you open ``/usr/lib/python3/dist-packages/blueman/plugins/mechanism/Network.py`` and look for the below 
, it should not have a trailing comma at the end of ``"DhcpdHandler": DhcpdHandler}``

```
DHCPDHANDLERS = {"DnsMasqHandler": DnsMasqHandler,
                 "DhcpdHandler": DhcpdHandler}
```

Also to be on the safe side remove ``/var/lib/blueman/network.state`` (as root).

HockeyKS, 2016-11-09:
@infirit ok I did what you said, the Integer error disappeared but there occured another one.

org.freedesktop.DBus.Python.Exception: Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python3/dist-packages/blueman/main/DbusService.py", line 36, in wrapper
    return method(*args[1:], **kwargs)
  File "/usr/lib/python3/dist-packages/blueman/plugins/mechanism/Network.py", line 59, in EnableNetwork
    nc.apply_settings()
  File "/usr/lib/python3/dist-packages/blueman/main/NetConf.py", line 304, in apply_settings
    self.dhcp_handler.do_apply()
  File "/usr/lib/python3/dist-packages/blueman/main/NetConf.py", line 60, in do_apply
    raise Exception("dnsmasq failed to start. Check the system log for errors")
Exception: dnsmasq failed to start. Check the system log for errors

Do I have to change something at the dnsmasq?

infirit, 2016-11-09:
>I did what you said, the Integer error disappeared

Did you have to remove the comma or did deleting the state file solve it?

>Do I have to change something at the dnsmasq

The error tells you what to do, check your system logs and see why it failed to start. I guess jessie now uses `journalctl`` to see the logs.

HockeyKS, 2016-11-09:
@infirit 
1.Yeah I had to remove the comma from the file.
2. I run the journalctl and filtered the error messages:

**Nov 09 14:02:10 raspberrypi dnsmasq[1507]: failed to create listening socket for 169.254.215.180: Address already in use**
Nov 09 14:02:10 raspberrypi dnsmasq[1507]: FAILED to start up
Nov 09 14:02:40 raspberrypi blueman-mechanism[1500]: Exiting
Nov 09 14:02:40 raspberrypi sudo[1418]: pam_unix(sudo:session): session closed for user root
Nov 09 14:02:52 raspberrypi sudo[1527]: pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/bin/journalctl
Nov 09 14:02:52 raspberrypi sudo[1527]: pam_unix(sudo:session): session opened for user root by (uid=0)
Nov 09 14:04:31 raspberrypi dbus[446]: [system] Activating service name='org.blueman.Mechanism' (using servicehelper)
Nov 09 14:04:32 raspberrypi org.blueman.Mechanism[446]: Unable to init server: Could not connect: Connection refused
Nov 09 14:04:32 raspberrypi org.blueman.Mechanism[446]: Unable to init server: Could not connect: Connection refused
Nov 09 14:04:32 raspberrypi blueman-mechanism[1548]: Starting blueman-mechanism
Nov 09 14:04:32 raspberrypi dbus[446]: [system] Successfully activated service 'org.blueman.Mechanism'
Nov 09 14:04:32 raspberrypi org.blueman.Mechanism[446]: (blueman-mechanism:1548): Gtk-CRITICAL **: gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed
Nov 09 14:04:32 raspberrypi blueman-mechanism[1548]: loading Network
Nov 09 14:04:32 raspberrypi blueman-mechanism[1548]: loading Ppp
Nov 09 14:04:32 raspberrypi blueman-mechanism[1548]: loading RfKill
Nov 09 14:04:32 raspberrypi blueman-mechanism[1548]: loading Rfcomm
Nov 09 14:04:32 raspberrypi blueman-mechanism[1548]: Return code 0
Nov 09 14:04:32 raspberrypi blueman-mechanism[1548]: Return code 0
Nov 09 14:04:32 raspberrypi blueman-mechanism[1548]: Return code 0
Nov 09 14:04:32 raspberrypi blueman-mechanism[1548]: Return code 0
**Nov 09 14:04:32 raspberrypi org.blueman.Mechanism[446]: dnsmasq: failed to create listening socket for 169.254.215.180: Address already in use
Nov 09 14:04:32 raspberrypi dnsmasq[1555]: failed to create listening socket for 169.254.215.180: Address already in use**
Nov 09 14:04:32 raspberrypi dnsmasq[1555]: FAILED to start up
Nov 09 14:05:02 raspberrypi blueman-mechanism[1548]: Exiting
Nov 09 14:05:04 raspberrypi sudo[1527]: pam_unix(sudo:session): session closed for user root
Nov 09 14:05:06 raspberrypi sudo[1559]: pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/bin/journalctl
Nov 09 14:05:06 raspberrypi sudo[1559]: pam_unix(sudo:session): session opened for user root by (uid=0)

I am pretty new to raspi and I do not understand why the IP Address should be in use already!?

infirit, 2016-11-09:
>Yeah I had to remove the comma from the file.

Hmm, I thought this was fixed already in backports, /me nudges @cschramm :smile: 

>why the IP Address should be in use already

Probably by another dnsmasq process, kill them all with ``killall dnsmasq`` and try again.

[2 more replies]