blueman-project/blueman #460: Trying to get properties from removed device

infirit, 2016-02-13:
What happens,

 * Discover at least __two__ new devices and select one of them. It may also happen with a single removed device but I have not yet been able to reproduce that.
 * Bluez tries expired them because they were not paired or connect we properly notice and remove them from the list. 
 * We respond selection changed of the first one
 * Selected changes to the second removed device
 * We try to lookup uuids to update the send file button and fail

It looks like this is what the ``Valid`` property was for on ``blueman.main.Device`` and now that is gone we run into trouble.

There are a couple of options to fix this.

 1. Re-introduce the ``Valid`` property and set it to False when a device is removed. Then add the necessary checks in the right places.
 2. Add an error_handler (and new eerror DBusObjectUnknown) to PropertiesBase and handle the case and return None (or something else).
 3. Something I did not think off :smile:.

```python
_________
on_device_selected (/usr/lib64/python3.4/site-packages/blueman/gui/manager/ManagerToolbar.py:84)
*** <ManagerDeviceList object at 0x7f843b9b3ea0 (blueman+gui+manager+ManagerDeviceList+ManagerDeviceList at 0x244c430)> <Device object at 0x7f842dd22828 (blueman+bluez+Device+Device at 0x21786a0)> <GtkTreeIter at 0x2181140> on_selection_changed *** 
Traceback (most recent call last):
  File "/usr/lib64/python3.4/site-packages/blueman/gui/manager/ManagerToolbar.py", line 117, in on_device_selected
    self.update_send(device)
  File "/usr/lib64/python3.4/site-packages/blueman/gui/manager/ManagerToolbar.py", line 122, in update_send
    for uuid in device['UUIDs']:
  File "/usr/lib64/python3.4/site-packages/blueman/bluez/PropertiesBase.py", line 61, in __getitem__
    return self.get(key)
  File "/usr/lib64/python3.4/site-packages/blueman/bluez/PropertiesBase.py", line 43, in get
    raise e
  File "/usr/lib64/python3.4/site-packages/blueman/bluez/PropertiesBase.py", line 38, in get
    prop = self.__properties_interface.Get(self._interface_name, name)
  File "/usr/lib64/python3.4/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib64/python3.4/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownObject: Method "Get" with signature "ss" on interface "org.freedesktop.DBus.Properties" doesn't exist
```

cschramm, 2016-02-14:
I guess the most elegant solution would be to not trigger device-selected in DeviceList.on_selection_changed if the device is invalid. A property does not help there as it might not yet have been set. We just need to check if the object exists.

Unfortunately I do not know an elegant way to check that (with dbus-python). One way would be introspection: `'org.bluez.Device1' in dbus.Interface(dbus.SystemBus().get_object('org.bluez', dev.get_object_path()), 'org.freedesktop.DBus.Introspectable').Introspect()`.

infirit, 2016-02-14:
Let me first go through DeviceList and try to reduce the functions, calling functions, calling functions to do the same thing spaghetti. Cleaning and reworking things as I go.

infirit, 2016-02-15:
Just ran the 2-0-stable branch and we have similar issues there. It is not only master and has perhaps been going on for a while already. Still something to fix though but it looks like it is not recent.

```
__init__ (/usr/lib64/python3.4/site-packages/blueman/main/Device.py:26)
caching initial properties 
ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
  File "/usr/lib64/python3.4/site-packages/dbus/connection.py", line 230, in maybe_handle_message
    self._handler(*args, **kwargs)
  File "/usr/lib64/python3.4/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 55, in wrapper
    handler(name, value, **kwargs)
  File "/usr/lib64/python3.4/site-packages/blueman/plugins/applet/GameControllerWakelock.py", line 33, in on_device_property_changed
    klass = Device(path).get_properties()["Class"] & 0x1fff
  File "/usr/lib64/python3.4/site-packages/blueman/main/Device.py", line 47, in __init__
    self.Properties = self.Device.get_properties()
  File "/usr/lib64/python3.4/site-packages/blueman/bluez/errors.py", line 148, in warp
    raise parse_dbus_error(exception)
  File "/usr/lib64/python3.4/site-packages/blueman/bluez/errors.py", line 146, in warp
    return func(*args, **kwargs)
  File "/usr/lib64/python3.4/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 34, in get_properties
    return self.__properties_interface.GetAll(self.get_interface_name())
  File "/usr/lib64/python3.4/site-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib64/python3.4/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib64/python3.4/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownObject: Method "GetAll" with signature "s" on interface "org.freedesktop.DBus.Properties" doesn't exist

_________
__del__ (/usr/lib64/python3.4/site-packages/blueman/main/Device.py:72)
deleting device None 
_________
Destroy (/usr/lib64/python3.4/site-packages/blueman/main/Device.py:94)
invalidating device None 
_________
__init__ (/usr/lib64/python3.4/site-packages/blueman/main/Device.py:26)
caching initial properties 
ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
  File "/usr/lib64/python3.4/site-packages/dbus/connection.py", line 230, in maybe_handle_message
    self._handler(*args, **kwargs)
  File "/usr/lib64/python3.4/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 55, in wrapper
    handler(name, value, **kwargs)
  File "/usr/lib64/python3.4/site-packages/blueman/plugins/applet/SerialManager.py", line 59, in on_device_property_changed
    d = Device(path)
  File "/usr/lib64/python3.4/site-packages/blueman/main/Device.py", line 47, in __init__
    self.Properties = self.Device.get_properties()
  File "/usr/lib64/python3.4/site-packages/blueman/bluez/errors.py", line 148, in warp
    raise parse_dbus_error(exception)
  File "/usr/lib64/python3.4/site-packages/blueman/bluez/errors.py", line 146, in warp
    return func(*args, **kwargs)
  File "/usr/lib64/python3.4/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 34, in get_properties
    return self.__properties_interface.GetAll(self.get_interface_name())
  File "/usr/lib64/python3.4/site-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib64/python3.4/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib64/python3.4/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownObject: Method "GetAll" with signature "s" on interface "org.freedesktop.DBus.Properties" doesn't exist
```

infirit, 2016-02-16:
Another way to solve this is to add colum(s) to the GtkListStore that hold the data we need to drive the UI. For example a column which holds an ``obex_capable`` boolean or list of uuids we care about. Then we retrieve from the store instead of relying on the dbus proxy. I am assuming the uuid's do not change over time which, afaik, they do not.

While we are in there anyway, add a timestamp when it was added so we can use the build-in sorting feature (instead of recreating the whole thing in a different order).

infirit, 2016-07-15:
There shouldn't be any issues with removed devices anymore. Closing for now but may re-open later if something comes up.

closed 2016-07-15 by infirit