blueman-project/blueman #465: With multiple adapters DeviceList only connects one on startup

infirit, 2016-02-18:
To make sure we don't forget. In function ``SetAdapter`` in class ``DeviceList`` on first run we only ever connect one adapter if there are multiple. I am not sure what the design is supposed to be but I see the following scenarios.

 1. Connect only the selected adapter and emit the ``adapter-property-changes`` signal. In that case we never disconnect and end up sending signals for the wrong adapter and worse, self.Adapter may not belong to the changed property value.
 2. Connect all the adapters and emit ``adapter-property-changes`` signal for all adapters. In this case we only ever connect one of the many on startup.

Looking through the code my guess right now is the first.

Personally I think a ``DeviceList`` should not be sending signals like this as this is the task of the Adapter object. The reason is simple, we emit a changed signal when we set a property so each class that wants to see these changes can connect to the signal and handle them. Opinions and disagreement welcome :smile:.

Currently there are no places this makes a difference as these property changes are not used situations which adapter is used matters. Regardless this needs to be fixed before 2.1.

infirit, 2016-02-18:
At the very least we should make sure we send the correct adapter object, patch below.

```diff
diff --git a/blueman/gui/DeviceList.py b/blueman/gui/DeviceList.py
index d18b80f..7a80d79 100644
--- a/blueman/gui/DeviceList.py
+++ b/blueman/gui/DeviceList.py
@@ -111,12 +111,12 @@ class DeviceList(GenericList):
             dev = row["device"]
             self.emit("device-selected", dev, tree_iter)
 
-    def _on_property_changed(self, _adapter, key, value, _path):
+    def _on_property_changed(self, adapter, key, value, _path):
         if key == "Discovering":
             if not value and self.discovering:
                 self.StopDiscovery()
 
-        self.emit("adapter-property-changed", self.Adapter, (key, value))
+        self.emit("adapter-property-changed", adapter, (key, value))
 
     def _on_device_property_changed(self, _device, key, value, path):
         tree_iter = self.find_device_by_path(path)
```

cschramm, 2016-02-19:
I agree that we have a problem there as the signal does not get disconnected if the adapter changes.

The patch seems wrong though. The idea of the list's adapter-property-changed signal is that it's sent only for the currently selected adapter that's relevant for the list's content. Looking  at the cases where the signal is used (a good search term is `List.connect("adapter-property-changed"`) there are things happening if the Name or Alias change which could also be run if the event is actually triggered by a different adapter and there are things happening if the Discovering property changes. The ManagerMenu and ManagerToolbar disable or enable UI elements that start a discovery and the DeviceSelectorWidget resets the progress bar if the discovery ends. These actions must not be taken if it's not about the Discovery property of the primary device.

I think currently you can get issues - say the UI allows you to start a discovery while another one is already running - e.g. in the following way if you have multiple adapters:

* Disconnect the initially selected adapter
* Start a discovery on another remaining one
* Reconnect the initially selected adapter - it will  probably report Discovery = False and the UI will pretend the running discovery stopped

With your patch you can get there much easier:

* Start a discovery
* Connect an additional adapter and select it - same effect then

I think the scenario of connecting a new adapter and selecting it is more likely then disconnecting the primary adapter and reconnecting it, so this would lead to issues easier.

To actually fix the problem, we have to disconnect the signal when changing the Adapter property.

infirit, 2017-08-11:
I think I have a simple solution for this, see #745

closed 2017-08-19 by infirit