blueman-project/blueman #467: ShowConnected (tray icon) shows more devices connected than there are

infirit, 2016-02-21:
When I plug in a new usb adapter and the number of connection increases while no new devices are connected there. With two devices connected and inserting a second adapter the number of connected devices goes to four. Connecting and disconnecting while the second adapter is already inserted does seem to increment and decrement properly.

Just noticed and documenting so we don't forget :smile:.

cschramm, 2016-02-21:
When an adapter gets added or removed, the device with the Connected property set to True are counted to get the number. When a device gets connected or disconnected (the Connected property changes), the number gets increased or decreased.

Did I get it right that you get 2 connections too much per adapter? Which version?

infirit, 2016-02-21:
I have one usb adapter and one mouse connected to it. The number of connections properly shows as as tooltip as one (1).

Insert a second usb adapter with no devices connected to it. The number of connections increases to two (2).

Remove the adapter the number of connections goes back to one (1).

Now with two usb adapters connect a new device on the same adapter as the other device and the number of connections increases to three (3), disconnect and it goes back to two (2).

Now remove the second adapter and it goes to two (2) which is correct.

Now insert the second adapter with two (2) devices connected on the first and we get a total of four (4) connections.

infirit, 2016-02-21:
The problem is in ``blueman.bluez.Adapter`` with the list_devices function. We do not filter the objects we get from ``GetManagedObjects`` which returns __all__ objects not just for the adapter we called list_devices on. We should filter any device that does not belong to the adapter.

infirit, 2016-02-21:
Below works however I am not liking the use of ``_Base__obj_path`` but it is the only way to access the adapter's path. And this problem exists in stable as well, at least for the BlueZ5 code.

```diff
diff --git a/blueman/bluez/Adapter.py b/blueman/bluez/Adapter.py
index 8867d42..3473700 100644
--- a/blueman/bluez/Adapter.py
+++ b/blueman/bluez/Adapter.py
@@ -28,7 +28,8 @@ class Adapter(PropertiesBase):
         devices = []
         for path, interfaces in objects.items():
             if 'org.bluez.Device1' in interfaces:
-                devices.append(path)
+                if self._Base__obj_path in path:
+                    devices.append(path)
         return [Device(device) for device in devices]
 
     def start_discovery(self):
```

cschramm, 2016-02-21:
Oh yeah, that's broken You could use self.get_object_path there. It would also be better to check the devices Adapter property.

infirit, 2016-02-21:
>You could use self.get_object_path there

Duh, I typed that many times, why did I not think of that now :smile:.

I'd like to avoid creating unnecessary Device objects at that point so used ``startswtith`` on the string instead.

commit ed9549f1 (2016-02-21): Adapter: Only return devices belonging to the adapter

GetManagedObjects returns all object so we need to filter the objects from other adapters.
fixes #467
commit 27bad4d6 (2016-02-21): Adapter: Only return devices belonging to the adapter

GetManagedObjects returns all object so we need to filter the objects from other adapters.
fixes #467
closed 2016-02-21 by infirit