blueman-project/blueman #468: Jaybird X2 connection problem

Robertgrant0608, 2016-02-23:
Im using a HP Probook 6470b with a Targus 4.0 bluetooth adapter ACB10US1 on Mint 17.1 with Blueman 1.23 and I have gotten Blueman to discover the wireless headset and connect, yet the music does not transfer from the laptop speakers to the headset.  I have tried resetting the headset but that does not make a difference.  Any advice.

cschramm, 2016-02-24:
Do you have pulseaudio-module-bluetooth? Do you see the device in pavucontrol?

infirit, 2016-04-21:
No response, see the section on the [troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting#a2dp-audio-device-paired-but-fails-to-connect). Pay close attention on the ``module-switch-on-connect`` bits. 

Closing for now, feel free to reply with more info and we can re-open.

closed 2016-04-21 by infirit