blueman-project/blueman #471: bluez.Base.__del__ never called

infirit, 2016-02-24:
I stuck a dprint("****") in this function and during operation this is __never__ run so we never remove the signal and I guess we eventually run out of dbus watches.

cschramm, 2016-02-24:
Hrm, right, this can only work if the object is deleted explicitly. It cannot be triggered by garbage collection as the signal receivers hold references to the object. That may change with migration to GLib dbus handling. Did you try it there?

infirit, 2016-02-24:
>object is deleted explicitly

If you mean ``del()``, no that won't and does not work either.

I have been playing around with creating a remote control for av devices, [code here](https://github.com/infirit/blueman/blob/mediaplayer/blueman/plugins/applet/AVRemote.py). SO I have a new bluez class that is only ever used in the new code so can keep track of things.

When I properly connect and disconnect signals, many references are still there, labels hold a reference :confused:.

I'l rebase my gdbus branch and see how things work there.

```python
__init__ (/usr/lib64/python3.4/site-packages/blueman/plugins/applet/AVRemote.py:28)
Added device:  /org/bluez/hci0/dev_6C_83_36_B9_C6_D6/player0 
[{'_artist': <Label object at 0x7fee5a27cdc8 (GtkLabel at 0xdd5f90)>,
  '_manager': <Manager object at 0x7fee5a27ce58 (blueman+bluez+Manager+Manager at 0xa4a500)>,
  '_next': <Button object at 0x7fee5a27cd38 (GtkButton at 0xea7400)>,
  '_pause': <Button object at 0x7fee5a27cca8 (GtkButton at 0xea7740)>,
  '_play': <Button object at 0x7fee5a27cc60 (GtkButton at 0xea78e0)>,
  '_prev': <Button object at 0x7fee5a27cd80 (GtkButton at 0xea7260)>,
  '_remote_control_devices': {},
  '_stop': <Button object at 0x7fee5a27ccf0 (GtkButton at 0xea75a0)>,
  '_track': <Label object at 0x7fee5a27ce10 (GtkLabel at 0xdd5e00)>,
  'active_remote': <AVRemote object at 0x7fee5a27cfc0 (blueman+bluez+AVRemote+AVRemote at 0xa4c4a0)>},
 <frame object at 0xebe678>,
 <bound method AVRemote._on_properties_changed of <AVRemote object at 0x7fee5a27cfc0 (blueman+bluez+AVRemote+AVRemote at 0xa4c4a0)>>]
```

infirit, 2016-02-24:
>I'l rebase my gdbus branch and see how things work there.

Nope same thing, after disconnecting I still get _on_properties_changed from ``PropertiesBase`` and not from my new class.

infirit, 2016-07-15:
So,`` __del__`` is not guaranteed to run (especially in GObject with all the signals) as something may hold a ref to the object. 

So we are best to explicitly clean-up, disconnect signals etc. Or create a new signal tracker :stuck_out_tongue_winking_eye: .

infirit, 2016-09-17:
This is not relevant anymore with the recent changes, singleton and move to GDBus.

closed 2016-09-17 by infirit