blueman-project/blueman #474: Blueman can't detect adapters anymore

Gensa23, 2016-02-28:
blueman: 1.23-git
BlueZ: 4.1
Distribution: mint rosa 
Desktop environment: Mate 

Debug output: http://pastebin.com/bc3Rngxr
when running: 

> bluez-test-adapter list

http://pastebin.com/rxE96nrz
I have to mention that blueman was working when I first installed Mint but due to me applying this tutorial it doesn't work any more.

http://firstdoit.com/quick-tip-bluetooth-a2dp-on-linux-mint-17-qiana-mate/?utm_source=blog.gadr.me

anyhelp will be appreciated. Thanks

infirit, 2016-02-28:
Your adapter is not being recognized by bluez which means there is little we can do in blueman. But..

My best guess is that the kernel module is not loading or not loading properly due to missing firmware. Look at the output of your kernel (dmesg) and look for message related to bluetooth. Use lspci or lsusb to figure out the brand and model which you can use in google to figure out which kernel module is used for your adapter.

Gensa23, 2016-02-28:
I've found my way around it now. But the original problem is still there. I can connect to my bluetooth headset but there is no sound. and it fails every time I try to connect A2DP. 

Is there a solution to this?

infirit, 2016-02-28:
Ok, 1.23 is old and it would be best to get the latest stable 2.0 release. I understand this may not be possible so.. make sure pulseaudio-module-bluetooth is installed. Does the device show up in pavucontrol when it is connected? Does a right click menu in blueman-manager show any options to select a profile?

Gensa23, 2016-02-28:
pulseaudio-module-bluetooth is installed. No it doesn't show up on pavucontrol. right click on the device in blueman gives me the options to (Audio sink, Headset service, input service). Headset option is the only one that words others give me the message: "Operation not supported" for input service 
"stream setup failed" for audio sink.

cschramm, 2016-02-28:
"Stream setup failed" points to some BlueZ 4 configuration issue (`Socket` not enabled in `/etc/bluetooth/audio.conf`), see. e.g. #285.

Gensa23, 2016-02-28:
Thank you all for your support and patience.. I truly appreciate it. I've followed the solution in the link and played a little with the system and now it works perfectly.

closed 2016-02-28 by Gensa23

infirit, 2016-02-28:
excellent :smile: