blueman-project/blueman #475: bluetooth automatically turn on

rrishichauhan, 2016-03-01:
i am using dell vostro 3000 series whenever i closed my lid and again open it. bluetooth automatically turn on .. why please tell me

infirit, 2016-03-01:
In recent versions (2.0) of blueman the powermanager plugin will turn it on. You can check with ``gsettings get org.blueman.plugins.powermanager auto-power-on`` and if true change it with ``gsettings set org.blueman.plugins.powermanager auto-power-on false`` (or use dconf-editor).

cschramm, 2016-03-01:
No need to use gsettings. It's also available through the UI in the plugin settings.

infirit, 2016-03-02:
>UI in the plugin settings

:+1:, how can I forget something like that :confused:

infirit, 2016-04-21:
Closing, if there still is an issue feel free to reply and we can re-open.

closed 2016-04-21 by infirit