blueman-project/blueman #477: 2.0.4

cschramm, 2016-03-07:
Guess it gets time for a 2.0.4 bug fix release. I updated the changelog in 2-0-stable. The issue mentioned in https://github.com/blueman-project/blueman/issues/457#issuecomment-193132014 needs to get fixed. Anything else?

cschramm, 2016-03-09:
I guess #463 should be a candidate as well.

infirit, 2016-03-09:
For #463, does it fix the problem for that user?

It would be nice if we could include #431.

cschramm, 2016-03-09:
#463: In contact, waiting for feedback.
#431: I'll test it.

cschramm, 2016-03-10:
All right, #431 is tested, merged, and cherry-picked into 2-0-stable. I also cherry-picked cec53a9 and c0261e0 and updated the changelog accordingly. Found another issue with serial services (#481), but I think that's out of scope for 2.0.4.

infirit, 2016-03-10:
OK, when I am back home tonight I'll build and make sure I did not add more typos.
-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

infirit, 2016-03-11:
Pushed one more commit for corner cases which I ran into just now with testing dues to the the  issues with NAP in 2.0. It leaves things in an inconsistent state which is pretty likely now for users.

Are you able to use your phone for internet __without__ NetworkManager? I only ever used this with NetworkManager afaicr.

infirit, 2016-03-11:
Other than that NAP appears to be working properly now.

cschramm, 2016-03-13:
I have to check that later. Do you get the exception with or without 660199c5cd7e8369754f250e6c9d765900cf8588?

cschramm, 2016-03-21:
@infirit: NAP works fine for me with both 2.0.3 and 2-0-stable.

All set for 2.0.4 now?

infirit, 2016-03-21:
Yes, I think as long as it works for you it should be fine.

infirit, 2016-04-19:
Its out so close :smile:

closed 2016-04-19 by infirit