blueman-project/blueman #478: File Transfer, No Options

Carl393, 2016-03-08:
Hello for Local File Transfer

I send a file to my blackberry no problems.
however when I attempt to send a file from my blackberry to the debian computer through blueman 1.99 applet.. it says unable to find service.

is it not possible to send a file to the computer? Sorry.. Just can't figure this out.. file transfers work on PC MAC OS etc... maybe I'm not setting something up properly... I have the device connected, discoverable, trusted. etc...

cschramm, 2016-03-08:
You're supposedly using blueman 1.99.alpha1 on supposedly Ubuntu 15.04. Note that the blueman version is quite dated now and in alpha state and contains a severe security issue. Also, if I guessed your system right, [it's EOL](https://lists.ubuntu.com/archives/ubuntu-announce/2016-January/000203.html).

When you try to send a file to the computer only an error message containing "unable to find service record" shows up. Where? Please check the setting for the incoming folder in the local services dialog.

Carl393, 2016-03-08:
Hello thanks for your emailI'm running Debian Jessie 8.3You're correct it's applet version 1.99When sending a file from a blackberry phone via bluetooth (even though paired)it says (on the phone during the file transfer) unable to find service (referring to the computer unable to "accept" a file)This feature has worked with other operating systems.‎ Hence why I'm not sure what's going on.                                                                                                                                     Under blueman applet settings... the file transfer option is grayed out. Would this have something to do with it ? Where would I set settings for the incoming folder.. I am not sure what "local‎ services dialogue is"Very sorry I'm new to the Linux world                                                                                                                                                                                                   Carl                                                                                                                                                                                                                From: Christopher SchrammSent: Tuesday, March 8, 2016 2:26 AMTo: blueman-project/bluemanReply To: blueman-project/bluemanCc: Carl393Subject: Re: [blueman] File Transfer, No Options (#478)You're supposedly using blueman 1.99.alpha1 on supposedly Ubuntu 15.04. Note that the blueman version is quite dated now and in alpha state and contains a severe security issue. Also, if I guessed your system right, it's EOL.

When you try to send a file to the computer only an error message containing "unable to find service record" shows up. Where? Please check the setting for the incoming folder in the local services dialog.

cschramm, 2016-03-08:
If it's grayed out no transfer service is provided by blueman, so you're trying to use a service that's not present (not sure why your phone indicates it is). You need to enable the TranferService plugin, but it will most probably not work as bluez and obex-data-server do not work in jessie. You should better get a recent version of blueman from jessie-backports.

closed 2016-03-08 by cschramm

Carl393, 2016-03-08:
Thank you Chris for your very helpful response. I really appreciate the
time you spent to help answer my question :O)

Carl

cschramm, 2016-03-08:
You're welcome. :smiley: