blueman-project/blueman #479: AppImage for blueman

kskarthik, 2016-03-09:
Appimage helps packing apps which can run on all major Linux distributions. It's based on concept of one __app = one file__ No installation necessary. Just download & run.

If blueman can be packed along with all its dependencies inside an appimage, users can simply download give permission & run the file. Appimage also supports Delta updates.

Popular apps like subsurface, scribus, krita also adopted this method to distribute their apps. It's easy to pack apps using AppImage. No extra knowledge required.

Project repo: https://github.com/probonopd/AppImagekit

The developer @probonopd is happy to help you packing your app :)

cschramm, 2016-03-09:
> blueman can be packed along with all its dependencies inside an appimage

Seriously? Is that how it works? No dependency management? Then it's far from applicable for blueman. Dependencies include stuff like bluez, dbus, or python.

infirit, 2016-03-09:
It is yet another attempt to do this and, as with the others, it will fail in the end.

closed 2016-03-09 by cschramm

kskarthik, 2016-03-10:
well, it supports delta updates too. Regarding the dependencies you should include which aren't provided by the base system

cschramm, 2016-03-10:
> Regarding the dependencies you should include which aren't provided by the base system

Read that on the website, but I don't get the term "base system" there. The only definition that would make sense to me there is the intersection of the stuff that's provided in the default setup of all target systems. I guess that's basically just any version of the Linux kernel and nothing else, which leads to "package everything" and, again, it does not make sense to create an image that includes dbus, bluez, python, etc. - if that would even be possible.

infirit, 2016-03-10:
>base system

Define base system (lsb?) and does it apply across all distributions?

I think the concept is fundamentally flawed on gnu+linux. How are you going to avoid symbol collision with libraries provided by the distributions? What is the solution for security updates in for example openssl? We run into the same problems as with bundled libraries [1], which is exactly what this is just with a new name.

[1] https://www.freebsd.org/doc/en/books/porters-handbook/bundled-libs.html.

infirit, 2016-03-10:
It bundles 3rd party libraries as well, sigh... https://github.com/probonopd/AppImageKit/tree/master/AppDirAssistant.AppDir/usr/lib

probonopd, 2016-03-10:
> Read that on the website, but I don't get the term "base system" there. The only definition that would make sense to me there is the intersection of the stuff that's provided in the default setup of all target systems.

Correct.

> I guess that's basically just any version of the Linux kernel and nothing else, which leads to "package everything"

Not correct. Check out the [example AppImages](https://github.com/probonopd/AppImages), they don't nearly package "everything" but only what cannot be assumed to be part of the targeted systems.

I am not familiar with blueman in particular, but you  are of course free to limit the base systems you are targeting to such systems on which bluez is already present, and throw a GUI error box otherwise.

kskarthik, 2016-03-11:
The main idea is to make blueman as appimage so that we could carry it on usb sticks. Why don't just give a try ?

cschramm, 2016-03-11:
> Why don't just give a try ?

It's too hard work.

> I am not familiar with blueman in particular, but you are of course free to limit the base systems you are targeting to such systems on which bluez is already present, and throw a GUI error box otherwise.

That would mean to assume that say bluez 5, dbus, python 3.2+ are already installed. We would then have to implement checks and error messages for all of those assumptions. As all dependencies are available from the package management on our target systems (they either already were or the maintainers of the respective blueman package added them) this leads us to bundling no dependencies at all and checking about 25 things. The package would also require a specific Python version as it includes a binary module.

If AppImage would have dependencies (in a way like e.g. a .deb package have dependencies) and care about installing them via APT, yum, pacman, or whatever the target system uses or at least checking if they are fulfilled so that we do not have to implement all those checks, it would be useful. (However, it would still be a binary package that in our case e.g. requires a specific Python version which may lead to the need of different images for different target systems which in terms brings us back to system specific .debs, .rpm, .tar.gz, etc...)

cschramm, 2016-03-11:
Also, another point: The website says AppImages run without installation and root privileges. blueman-mechanism requires both.