blueman-project/blueman #481: SerialService.disconnect is broken

cschramm, 2016-03-10:
I don't know what's going on there; guess it has to do with recent changes in BlueZ.

Our `release_rfcomm_device` fails (-15) for me in any case, but causes the device file to get removed anyway. If I disable the call, so that the only thing we do is killing our watcher process, it seems like the device sometimes gets auto-released (which would be a weird change in the behavior of the rfcomm system) but sometimes it stays alive.

infirit, 2016-03-10:
Do you also see the kernel crash? If so I am seeing the same thing.

infirit, 2016-03-10:
So from what I can see it expects an int but we provide a double. However turning it into an int does not make a difference unfortunately :disappointed:.
```
disconnect (/usr/lib64/python3.4/site-packages/blueman/services/meta/SerialService.py:41)
**** 0.0 <class 'dbus.Double'> 
```

cschramm, 2016-03-10:
Nope, no kernel crash.

The double gets cast when provided to `c_release_rfcomm_device`.

closed 2018-02-01 by cschramm