blueman-project/blueman #482: Shebang: blueman uses hard-coded "#!/usr/bin/env python"

MingcongBai, 2016-03-14:
When building Blueman on AOSC OS, a distribution using Python 2.7 as its default Python interpreter, Blueman will refuse to start after specifying `PYTHON=/usr/bin/python3` as an option for `configure`, because the installed programs would still contain `#!/usr/bin/env python` as its Shebang. As reported in the issue below.

https://github.com/AOSC-Dev/aosc-os-abbs/issues/170

Our build manifests can be found here (with workaround for shebang issue).

https://github.com/AOSC-Dev/aosc-os-abbs/tree/staging/extra-utils/blueman

cschramm, 2016-03-14:
See #435. Distributions usually replace the shebang. In Gentoo they have the python_fix_shebang tool, in Debian / Ubuntu debhelper does that job.

While we could set the shebang to $PYTHON during configure / make, I think that would be wrong as what you set with that variable is the python interpreter to be used during build. So it is the one on the build system, while the one on the target system which is set by the shebang could be different.

However, I guess it's a good idea to set either `#!/usr/bin/env python2` or `#!/usr/bin/env python3` depending on the major version of $PYTHON. I'll draft a patch for that.

MingcongBai, 2016-03-14:
> See #435. Distributions usually replace the shebang. In Gentoo they have the python_fix_shebang tool, in Debian / Ubuntu debhelper does that job.

So I did. While it really shouldn't have happened.

> So it is the one on the build system, while the one on the target system which is set by the shebang could be different.

Well that was usually done in GNOME packages and they would work flawlessly as to set Shebang to the `PYTHON` variable specified during build.

> However, I guess it's a good idea to set either #!/usr/bin/env python2 or #!/usr/bin/env python3 depending on the major version of $PYTHON. I'll draft a patch for that.

Yes it is, thanks for the quick response mate XD.

cschramm, 2016-03-14:
> Well that was usually done in GNOME packages and they would work flawlessly as to set Shebang to the PYTHON variable specified during build.

That does not make sense to me. Think about a complex build system that sets $PYTHON to say `/var/buildsystem/python/3.4.4/python` while the target system just has `/usr/bin/python(3(.4))`, so if $PYTHON would be used in the shebang, it would not be found, but `/usr/bin/env python(3)` will.

Patch proposal for major versions is in #483. It should fix your specific case up to the point where you add pyenv to the distribution and have to start replacing Python shebangs anyway.

infirit, 2016-03-14:
>Patch proposal for major versions is in #483. 

Added some comments :smile:.

>It should fix your specific case up to the point where you add pyenv to the distribution or something similar happens and you have to start replacing Python shebangs anyway.

Most distros now have a similar setup so adding the major number should be ok for most of them. However I think this is as far as we can go. Each distro has their own little eccentricities which we can not code for.

cschramm, 2016-03-14:
> Most distros now have a similar setup so adding the major number should be ok for most of them.

If not they do not follow PEP 394, so we're not to blame.

> However I think this is as far as we can go. Each distro has their own little eccentricities which we can not code for.

Totally agree. We cannot add minor versions as there's no convention for them and we cannot set a full path (instead of using env) as we do not know the path on the target system (it's probably even variable in case of some distributions), so if there's stuff like pyenv or multiple minor versions the maintainer has to take care of that.

closed 2016-03-14 by cschramm