blueman-project/blueman #484: troubleshooting debugging help needed

mccart, 2016-03-14:
Hello,

I'm trying to connect Phonack Compilot with my desktop using bluetooth 4.0 LE USB adapter.

Bluetooth pairing by all appearances worked.  Seems I cannot sustain a connection and 
I do not know why.  I included OS info, lsusb, bluetoothctl list, and terminal blueman-applet
output below.  In brief, after pairing and when trying to connect via blueman-manager, I get
_________
fail (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:120)
fail (DBusException(dbus.String(u'No such file or directory'),),) 
/usr/lib/python2.7/dist-packages/gi/overrides/__init__.py:92: Warning: Source ID 756 was not found when attempting to remove it
  return fn(*args, **kwargs)

What else can I do to proceed forward from here? 
Is there any additional debugging to do?
Any additional troubleshooting information?

Thanks much - Michael McCartney

$ hostnamectl
...
  Operating System: Debian GNU/Linux 8 (jessie)
            Kernel: Linux 3.16.0-4-amd64
      Architecture: x86-64

$ lsusb
Bus 006 Device 005: ID 0a5c:21e8 Broadcom Corp. BCM20702A0 Bluetooth 4.0

$ bluetoothctl list
[NEW] Controller 5C:F3:70:76:C1:59 debian [default]
[CHG] Controller 5C:F3:70:76:C1:59 Discovering: yes
[NEW] Device 00:0F:59:1F:B0:F5 Phonak ComPilot
[CHG] Device 00:0F:59:1F:B0:F5 RSSI is nil
[CHG] Controller 5C:F3:70:76:C1:59 Discovering: no
[CHG] Device 00:0F:59:1F:B0:F5 Connected: yes
[CHG] Device 00:0F:59:1F:B0:F5 UUIDs:
	0000110b-0000-1000-8000-00805f9b34fb
	0000110c-0000-1000-8000-00805f9b34fb
	0000110e-0000-1000-8000-00805f9b34fb
	0000111e-0000-1000-8000-00805f9b34fb
	00001131-0000-1000-8000-00805f9b34fb
[CHG] Device 00:0F:59:1F:B0:F5 Paired: yes
[CHG] Device 00:0F:59:1F:B0:F5 Connected: no

$ terminal blueman-applet output...
_________
get_interface_version (/usr/lib/python2.7/dist-packages/blueman/bluez/BlueZInterface.py:8)
Detected BlueZ 5 
Loading configuration plugins
Using GConf config backend
_________
SetAdapter (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:264)
None 
_________
on_adapter_property_changed (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/DiscvManager.py:86)
prop Discovering 1 
_________
on_property_changed (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:148)
adapter propery changed Discovering 1 
_________
on_device_created (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:246)
created /org/bluez/hci0/dev_00_0F_59_1F_B0_F5 
_________
__init__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:16)
caching initial properties 
_________
add_device (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:310)
adding new device 
_________
do_cache (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:489)
[... 136 lines cut ...]
set_op (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:76)
op: regenerating instance <ManagerDeviceMenu object at 0x7f9e1c8044b0 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x1c16dd0)> 
_________
__init__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:16)
caching initial properties 
_________
notify (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/RecentConns.py:239)
<Device object at 0x7f312b0f3f50 (blueman+main+Device+Device at 0x2b01140)> org.bluez.Device1 dbus.Array([], signature=dbus.Signature('s')) 
_________
initialize (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/RecentConns.py:138)
rebuilding menu 
_________
unset_op (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:90)
op: regenerating instance <ManagerDeviceMenu object at 0x7f9e1ea7fe60 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x1c16b50)> 
_________
Generate (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:213)
Phonak ComPilot 
_________
unset_op (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:90)
op: regenerating instance <ManagerDeviceMenu object at 0x7f9e1c8044b0 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x1c16dd0)> 
_________
fail (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:120)
fail (DBusException(dbus.String(u'No such file or directory'),),) 
/usr/lib/python2.7/dist-packages/gi/overrides/__init__.py:92: Warning: Source ID 756 was not found when attempting to remove it
  return fn(*args, **kwargs)

infirit, 2016-03-14:
It would be best to first update to the latest stable version in jessie backports, see http://backports.debian.org/Instructions/. It has lots of fixes over the version you use now.

Question, you are trying to connect with the manager menu and have an entry for "Audio Source" right?

mccart, 2016-03-14:
uhmmm... backports is for when to bring later software versions from testing (strech) or unstable (sid) into stable (jessie) Debian distribution.  thank you for the suggestion but that's not the solution.

Thanks to @cschramm suggestion in issue #468, I was missing bluez-firmware, pavucontrol, pulseaudio-module-bluetooth.  Looking for pulseaudio-module-bluetooth led me to https://wiki.debian.org/BluetoothUser/a2dp.   Just installing blueman is not enough.

I'm able to stream audio now - many thanks and perhaps the Debian BluetoothUser wiki link should be to FAQ for Debian?

mccart, 2016-03-14:
@infirit - after some thought - did you have a specific package in mind specifically for bluetooth?

cschramm, 2016-03-15:
@infirit meant blueman. The version in jessie is an old alpha version and it is advised to use the one from jessie-backports instead.

closed 2016-03-15 by cschramm

infirit, 2016-03-15:
Yes I meant blueman from backport :smile: .

I added something to [troubleshooting](https://github.com/blueman-project/blueman/wiki/Troubleshooting), hopefully it helps users from now on.

mccart, 2016-03-16:
Ok, will attempt to backport blueman - @cschramm, again many thanks.  
@infirit - troubleshooting update looks good!