blueman-project/blueman #490: No bluetooth devce found (ever)

educobian, 2016-03-28:
Since installing Linux Mint 17.3 bluetooth has never worked.
I have installed lates bios (lenovo b50-10. bios v: 2.05) and latest linux kernel (4.5. At least now shutdown works).

`$ rfkill list
0: ideapad_wlan: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: ideapad_bluetooth: Bluetooth
	Soft blocked: no
	Hard blocked: no
2: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
3: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
`

infirit, 2016-03-28:
Please look at [this](https://github.com/blueman-project/blueman/wiki/Troubleshooting#reporting-a-bug--asking-for-support) and provide the output of both the applet and manager. Make sure to paste them to a pastebin or a [gist](https://gist.github.com/).

Also please confirm the version of BlueZ and blueman.

educobian, 2016-03-28:
Hi. Thanks.
Not sure to be able to do as you ask.
blueman version: 1.23.
BlueZ version: I don't know how to look at it but I re-installed it yesterday.
blueman-applet:
    `$ blueman-applet 
Loading configuration plugins
blueman-applet version 1.23 starting
Using gconf config backend
_________
Load (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:54)
['StatusIcon', 'DBusService', 'AuthAgent', 'PulseAudio', 'Headset', 'SerialManager', 'Networking', 'ShowConnected', 'StandardItems', 'DhcpClient', 'NMDUNSupport', 'KillSwitch', 'NMPANSupport', 'RecentConns', 'TransferService', 'PowerManager', 'Menu', 'NetUsage', 'ExitItem', 'PPPSupport', 'AppIndicator', 'DiscvManager'] 
Using gconf config backend
_________
Load (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:54)
Unable to load plugin module AppIndicator
No module named appindicator 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
Using gconf config backend
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Headset.Headset'> 
Using gconf config backend
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Networking.Networking'> 
Using gconf config backend
_________
load_nap_settings (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/Networking.py:30)
Loading NAP settings 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'> 
Using gconf config backend
_________
update_menuitems (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/DiscvManager.py:114)
warning: Adapter is None 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.PulseAudio.PulseAudio'> 
Using gconf config backend
_________
pa_context_event (/usr/lib/python2.7/dist-packages/blueman/main/PulseAudioUtils.py:306)
1 
_________
on_load (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/PulseAudio.py:160)
PulseAudio version: [4, 0, 0] 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'> 
Using gconf config backend
_________
on_load (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/KillSwitch.py:25)
Using the new killswitch system 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
[... 455 lines cut ...]
             physical id: 0.0.0
             bus info: scsi@1:0.0.0
             logical name: /dev/cdrom
             logical name: /dev/sr0
             version: GY01
             capabilities: removable audio cd-r cd-rw dvd dvd-r dvd-ram
             configuration: ansiversion=5 status=nodisc
  *-battery
       description: Lithium Ion Battery
       product: SR Real Battery
       vendor: Intel SR 1
       physical id: 1
       version: Date
       serial: 123456789
       slot: I2C2
       configuration: voltage=3,8V
  *-power UNCLAIMED
       description: OEM_Define1
       product: OEM_Define5
       vendor: OEM_Define2
       physical id: 2
       version: OEM_Define6
       serial: OEM_Define3
       capacity: 75mWh
`

cschramm, 2016-04-18:
So... what's the problem?

educobian, 2016-04-20:
The problem is that no bluetooth device is detected so I cannot use my bluetooth speaker.

Two hardware devices are UNCLAIMED: Smbus and power.

I have tried everything I've seen on internet and nothing works.

Thanks.

infirit, 2016-04-20:
There is nothing out of the ordinary in the output. Please check and post the output after you tried to search for devices.

>Since installing Linux Mint 17.3 bluetooth

What did you use before, also Mint or another distribution?

educobian, 2016-05-17:
Not working but closed anyway.

cschramm, 2016-05-17:
No response from the reporter for four weeks...

infirit, 2016-05-19:
>Not working but closed anyway.

Not working but still waiting for [info](https://github.com/blueman-project/blueman/issues/490#issuecomment-212497712) :wink:

cschramm, 2016-07-06:
I know what follows, but again no response for nearly two months now...

closed 2016-07-06 by cschramm