blueman-project/blueman #492: bluetooth heasphones paired but not connected

KingBubu, 2016-04-09:
Hi all,

I recently bought the Bose® SoundLink® around-ear wireless headphones II and tried to connect them to my ThinkPad T440s.

blueman: v2
BlueZ: v5
Distribution: Ubuntu 15.10
Desktop environment: Unity

<script src="https://gist.github.com/KingBubu/319cf2f81e3b2d64c2f4c952dfdb50ed.js"></script>

I read the debian and archlinux wiki pages and also tried to connect via bluetoothctl. Here is what `systemctl status bluetooth` gives me:

`Unable to register GATT service with handle 0x0001 for device 08:DF:1F:C3:32:02:`

Any help is greatly appreciated.

Thanks!

cschramm, 2016-04-18:
BlueZ' GATT API is an experimental feature. You have to start bluetoothd with `-E`.

KingBubu, 2016-04-26:
I upgraded my system to Ubuntu 16.04 and successfully connected my headphones to it. In Sound Settings I selected the A2DP profile and everything seemed to be perfect. However when there is no music playing or only a little bit I distinctly hear a background noise. Is this normal? 

In /etc/pulse/daemon.conf I uncommented the line "enable-lfe-remixing = yes" in order for my external soundcard to work correctly. Could this be an issue?

cschramm, 2016-04-26:
I have no profound knowledge of pulseaudio, sorry.

closed 2016-04-26 by cschramm